Koenkk/zigbee2mqtt #17352: TS011F_plug_3 and v1 power consume not detected

h3ld3rkid, 2023-04-15:
### What happened?

a few days, after last updates my plugs TS011F left to detect current consume and power...

i try phoscon to eliminate possibilities of hardware malfunction and with phoscon works!

lets try fix?

### Zigbee2MQTT version

1.30.3-1

### Adapter firmware version

 2023.04.0

### Adapter

conbeeII

jocamane, 2023-04-15:
Me too
https://github.com/Koenkk/zigbee2mqtt/issues/17060#issue-1628451770

chatainsim, 2023-05-16:
Is there anything we can do to provide more information for debugging this issue ? Thanks

chatainsim, 2023-06-02:
Latest version seems to resolve this issue for me.
In Settings (Specific) I've put all the precision on 3.
Thanks

closed 2023-07-11 by github-actions[bot]