Koenkk/zigbee2mqtt #17359: [New device support]: Samotech SM308-2CH

jafrouk, 2023-04-16:
### Link

https://www.samotech.co.uk/products/zigbee-switch-sm308-2ch/

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0x040d84fffe43605a","nwkAddr":30244,"manufId":4107,"manufName":"Samotech","powerSource":"Mains (single phase)","modelId":"SM308-2CH","epList":[1,2,11,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"SM308-2CH","manufacturerName":"Samotech","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r3"}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":255}},"genLevelCtrl":{"attributes":{}},"genIdentify":{"attributes":{"identifyTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x385cfbfffee3511a","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x385cfbfffee3511a","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x385cfbfffee3511a","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":7,"inClusterList":[0,1794,2820,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"genOnOffSwitchCfg":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r3","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1801730331},"lastSeen":1681510376036,"defaultSendRequestWhen":"immediate"}

### Comments

I tried following the supporting new devices page, and was able to get Switch 1 working (endpoint 1), but can't figure out how to make switch 2 (endpoint 2) work.  I looked at some similar devices such as the Ubisys S2 but couldn't figure out the syntax (it would be good perhaps to have a simple example of multiple endpoint devices like this in the docs?)

I've included the config i have working for input 1. I have the device wired up ready to test if someone can help!

### External converter

```shell
`
// copy of SM308 defintion with the model details updated. Only works for endpoint 1 (S1, switch 1)
const definition  = {
        zigbeeModel: ['SM308-2CH'],
        model: 'SM308-2CH',
        vendor: 'Samotech',
        description: 'Zigbee 2 channel in wall switch',
        extend: extend.switch(),
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'genIdentify', 'genOnOff']);
        },
};
module.exports = definition;`
```

Koenkk, 2023-04-16:
Try with:

```js
const definition  = {
    zigbeeModel: ['SM308-2CH'],
    model: 'SM308-2CH',
    vendor: 'Samotech',
    description: 'Zigbee 2 channel in wall switch',
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2')],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genBasic', 'genIdentify', 'genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
};
```

jamesefraserbb, 2023-04-17:
Thanks @Koenkk!  That worked perfectly, both circuits are now being switched.

There are some other things the device does which aren't being reflected (below is the list of clusters it has) - but this is definitely covering the basics. I haven't tested that attaching buttons to the s1 and s2 connections and pressing them reports (I don't have a physical button handy to wire up).

Shall I go ahead and submit a PR to add this to the samotech.js file for the next release, even though it's not covering all features?

Input clusters:
• 0x0000: Basic
• 0x0003: Identify
• 0x0004: Groups
• 0x0005: Scenes
• 0x0006: On/off
• 0x0b04: Electrical Measurement
• 0x0b05: Diagnostics
• 0x0702: Simple Metering 

Output Clusters:
• 0x0019: OTA

Koenkk, 2023-04-18:
Seems electrical measurements are also supported, try with:

```js
const definition  = {
    zigbeeModel: ['SM308-2CH'],
    model: 'SM308-2CH',
    vendor: 'Samotech',
    description: 'Zigbee 2 channel in wall switch',
    extend: extend.switch({fromZigbee: [fz.electrical_measurement, fz.metering]}),
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.power(), e.current(), e.voltage(), e.energy()],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genBasic', 'genIdentify', 'genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint1);
        await reporting.activePower(endpoint1);
        await reporting.rmsCurrent(endpoint1);
        await reporting.rmsVoltage(endpoint1);
        await reporting.readMeteringMultiplierDivisor(endpoint1);
        await reporting.currentSummDelivered(endpoint1);
    },
};
```

jamesefraserbb, 2023-04-18:
That didn't work - but I think it was just because the power data is on endpoint 11.  I adjusted that, and also added some other things based on Samotech SM309-S.  Now it's reporting power consumption, I believe accurately.

`const definition  = {
    zigbeeModel: ['SM308-2CH'],
    model: 'SM308-2CH',
    vendor: 'Samotech',
    description: 'Zigbee 2 channel in wall switch',
    extend: extend.switch({fromZigbee: [fz.electrical_measurement, fz.metering]}),
    exposes: [
        e.switch().withEndpoint('l1'), 
        e.switch().withEndpoint('l2'), 
        e.power_on_behavior().withEndpoint('l1'),
        e.power_on_behavior().withEndpoint('l2'),
        e.power(), e.current(), e.voltage(), e.energy()],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        const endpoint2 = device.getEndpoint(2);
        const endpoint11 = device.getEndpoint(11);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genBasic', 'genIdentify', 'genOnOff']);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genBasic', 'genIdentify', 'genOnOff']);
        await reporting.bind(endpoint11, coordinatorEndpoint, ['genOta','haElectricalMeasurement', 'seMetering']);
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint11);
        await reporting.readMeteringMultiplierDivisor(endpoint11);
        await reporting.rmsVoltage(endpoint11, {min: 10, change: 20});
        await reporting.rmsCurrent(endpoint11, {min: 10, change: 10});
        await reporting.activePower(endpoint11, {min: 10, change: 15});
        await reporting.currentSummDelivered(endpoint11, {min: 300});
   
    },
};  `

I can submit a pull request to add this, unless you want to take care of it?

jafrouk, 2023-04-18:
That didn't work - but I think it was just because the power data is on endpoint 11.  I adjusted that, and also added some other things based on Samotech SM309-S.  Now it's reporting power consumption, I believe accurately.

```
const definition  = {
    zigbeeModel: ['SM308-2CH'],
    model: 'SM308-2CH',
    vendor: 'Samotech',
    description: 'Zigbee 2 channel in wall switch',
    extend: extend.switch({fromZigbee: [fz.electrical_measurement, fz.metering]}),
    exposes: [
        e.switch().withEndpoint('l1'), 
        e.switch().withEndpoint('l2'), 
        e.power_on_behavior().withEndpoint('l1'),
        e.power_on_behavior().withEndpoint('l2'),
        e.power(), e.current(), e.voltage(), e.energy()],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        const endpoint2 = device.getEndpoint(2);
        const endpoint11 = device.getEndpoint(11);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genBasic', 'genIdentify', 'genOnOff']);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genBasic', 'genIdentify', 'genOnOff']);
        await reporting.bind(endpoint11, coordinatorEndpoint, ['genOta','haElectricalMeasurement', 'seMetering']);
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint11);
        await reporting.readMeteringMultiplierDivisor(endpoint11);
        await reporting.rmsVoltage(endpoint11, {min: 10, change: 20});
        await reporting.rmsCurrent(endpoint11, {min: 10, change: 10});
        await reporting.activePower(endpoint11, {min: 10, change: 15});
        await reporting.currentSummDelivered(endpoint11, {min: 300});
   
    },
};  
```

I can submit a pull request to add this, unless you want to take care of it?

jafrouk, 2023-05-03:
Hi @Koenkk  - yes, it is reporting voltage, energy and current (sorry, i was using "power" as a general term encapsulating that!)

I submitted the pull request. One question though - I see with the way we have written the definition, there are two new endpoints showing called "l1" and "l2", and they even show up in the binding UI.  Did we need to do that, could we not use the existing endpoint 1 and 2?   It seems like these are duplicate "virtual" endpoints.   

I'm afraid I don't understand it enough to try tweaking it, yet!

Koenkk, 2023-05-03:
These are not new endpoints, we just assign friendly names to the endpoint. I will take a look at the PR!

closed 2023-05-03 by Koenkk