Koenkk/zigbee2mqtt #17361: [New device support]: TUYA device - temperature and humidity sensor - model TS0601 from the company _TZE200_vs0skpuc

kosmaluk, 2023-04-16:
### Link

https://www.houseiq.pl/pl/p/Czujnik-temperatury-wilgotnosci-ZIGBEE-TUYA-SMART/845

### Database entry

Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_vs0skpuc'

### Comments

Hello. I have a new TUYA device - temperature and humidity sensor - model TS0601 from the company
_TZE200_vs0skpuc and it is not support with zigbee2mqtt. I have the EDGE version installed and I did not find this device in the "new devices" section. I am a novice Home Assistant user and I can't manage to create a converter on my own. Please add this device. Thanks a lot

[image]

### Supported color modes

not applicable

### Color temperature range

not applicable

Koenkk, 2023-04-16:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/a24b9ecee5158bf42bb17294a0133451

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

kosmaluk, 2023-04-16:
The converter works. The device is exposed. Thanks a lot

Wysłane z aplikacji Poczta dla systemu Windows 

Od: Koen Kanters
Wysłano: niedziela, 16 kwietnia 2023 20:54
Do: Koenkk/zigbee2mqtt
DW: kosmaluk; Author
Temat: Re: [Koenkk/zigbee2mqtt] [New device support]: TUYA device -temperature and humidity sensor - model TS0601 from the company_TZE200_vs0skpuc (Issue #17361)

Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/a24b9ecee5158bf42bb17294a0133451
• save this as file next to configuration.yaml as ext_converter.js
• add it to configuration.yaml:
external_converters:
  - ext_converter.js
• start z2m, check if issue is fixed

closed 2023-04-17 by Koenkk

True0Man, 2023-12-31:
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/a24b9ecee5158bf42bb17294a0133451
>
> * save this as file next to `configuration.yaml` as [...]

Hi there,

i just used your external converter - but after restart ZIGBEE2MQTT all i see is this
---
[13:49:03] INFO: Preparing to start...
[13:49:03] INFO: Socat not enabled
[13:49:03] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:19
        {modelID: 'TS0601', manufacturerName: '_TZE200_vvmbj46n'},
        ^

SyntaxError: Unexpected token '{'
    at new Script (node:vm:100:7)
    at createScript (node:vm:265:10)
    at Object.runInNewContext (node:vm:306:10)
    at loadModuleFromText (/app/lib/util/utils.ts:154:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:161:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:171:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
----

What should i do?