Koenkk/zigbee2mqtt #17364: [New device support]: TS0202 and manufacturer name '_TZ3040_bb6xaihh

jocamane, 2023-04-16:
### Link

https://www.aliexpress.com/item/1005005099780623.html

### Database entry

{"id":80,"type":"EndDevice","ieeeAddr":"0xa4c13863cb90b4be","nwkAddr":28436,"manufId":4417,"manufName":"_TZ3040_bb6xaihh","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,4,1280,0],"outClusterList":[4,6,4096,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff07afaa","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1681655875502,"defaultSendRequestWhen":"immediate"}

### Comments

don't join:

2023-04-16 15:37:50Zigbee: allowing new devices to join.
Info 2023-04-16 15:37:50MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"288zx-15"}'
Info 2023-04-16 15:37:52Device '0xa4c13863cb90b4be' joined
Info 2023-04-16 15:37:53MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13863cb90b4be","ieee_address":"0xa4c13863cb90b4be"},"type":"device_joined"}'
Info 2023-04-16 15:37:53MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c13863cb90b4be"},"type":"device_connected"}'
Info 2023-04-16 15:37:53MQTT publish: topic 'zigbee2mqtt/0xa4c13863cb90b4be', payload '{"last_seen":"2023-04-16T15:37:53+01:00","linkquality":255}'
Info 2023-04-16 15:37:53Starting interview of '0xa4c13863cb90b4be'
Info 2023-04-16 15:37:53MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13863cb90b4be","ieee_address":"0xa4c13863cb90b4be","status":"started"},"type":"device_interview"}'
Info 2023-04-16 15:37:53MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c13863cb90b4be"},"type":"pairing"}'
Debug 2023-04-16 15:37:53Received Zigbee message from 'Sen Mov Quarto RC ZG', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
Info 2023-04-16 15:37:53MQTT publish: topic 'zigbee2mqtt/Sen Mov Quarto RC ZG', payload '{"battery":100,"battery_low":false,"keep_time":60,"last_seen":"2023-04-16T15:37:53+01:00","linkquality":255,"occupancy":true,"sensitivity":"medium","tamper":false,"voltage":2900}'
Debug 2023-04-16 15:37:55Received Zigbee message from '0xa4c13863cb90b4be', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID null
Debug 2023-04-16 15:37:55Skipping message, definition is undefined and still interviewing
Info 2023-04-16 15:37:55MQTT publish: topic 'zigbee2mqtt/0xa4c13863cb90b4be', payload '{"last_seen":"2023-04-16T15:37:54+01:00","linkquality":255}'
Debug 2023-04-16 15:37:55Received Zigbee message from '0xa4c13863cb90b4be', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID null
Debug 2023-04-16 15:37:55Skipping message, definition is undefined and still interviewing
Info 2023-04-16 15:37:55MQTT publish: topic 'zigbee2mqtt/0xa4c13863cb90b4be', payload '{"last_seen":"2023-04-16T15:37:54+01:00","linkquality":255}'
Debug 2023-04-16 15:37:55Received Zigbee message from '0xa4c13863cb90b4be', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID null
Debug 2023-04-16 15:37:55Skipping message, definition is undefined and still interviewing
Info 2023-04-16 15:37:55MQTT publish: topic 'zigbee2mqtt/0xa4c13863cb90b4be', payload '{"last_seen":"2023-04-16T15:37:55+01:00","linkquality":255}'
Debug 2023-04-16 15:37:55Received Zigbee message from '0xa4c13863cb90b4be', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00212effff07afaa","zoneState":1}' from endpoint 1 with groupID null
Debug 2023-04-16 15:37:55Skipping message, definition is undefined and still interviewing
Info 2023-04-16 15:37:55MQTT publish: topic 'zigbee2mqtt/0xa4c13863cb90b4be', payload '{"last_seen":"2023-04-16T15:37:55+01:00","linkquality":255}'
Info 2023-04-16 15:37:55Successfully interviewed '0xa4c13863cb90b4be', device has successfully been paired
Warning 2023-04-16 15:37:55Device '0xa4c13863cb90b4be' with Zigbee model 'TS0202' and manufacturer name '_TZ3040_bb6xaihh' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-04-16 15:37:56MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c13863cb90b4be","ieee_address":"0xa4c13863cb90b4be","status":"successful","supported":false},"type":"device_interview"}'
Info 2023-04-16 15:37:56MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xa4c13863cb90b4be","supported":false},"type":"pairing"}'
Debug 2023-04-16 15:37:57Received Zigbee message from 'Tomada 7 ZG', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info 2023-04-16 15:37:57MQTT publish: topic 'zigbee2mqtt/Tomada 7 ZG', payload '{"last_seen":"2023-04-16T15:37:57+01:00","linkquality":255,"state":"OFF"}'
Debug 2023-04-16 15:38:00Received Zigbee message from '0xa4c13863cb90b4be', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID null
Warning 2023-04-16 15:38:00Received message from unsupported device with Zigbee model 'TS0202' and manufacturer name '_TZ3040_bb6xaihh'
Warning 2023-04-16 15:38:00Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-04-16 15:38:00MQTT publish: topic 'zigbee2mqtt/0xa4c13863cb90b4be', payload '{"last_seen":"2023-04-16T15:37:59+01:00","linkquality":255}'
Debug 2023-04-16 15:38:00Received Zigbee message from '0xa4c13863cb90b4be', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
Warning 2023-04-16 15:38:00Received message from unsupported device with Zigbee model 'TS0202' and manufacturer name '_TZ3040_bb6xaihh'
Warning 2023-04-16 15:38:00Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-04-16 15:38:00MQTT publish: topic 'zigbee2mqtt/0xa4c13863cb90b4be', payload '{"last_seen":"2023-04-16T15:37:59+01:00","linkquality":255}'
Debug 2023-04-16 15:38:00Received Zigbee message from 'Sen Mov Corredor ZG', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID null
Info 2023-04-16 15:38:00MQTT publish: topic 'zigbee2mqtt/Sen Mov Corredor ZG', payload '{"battery":95,"battery_low":false,"last_seen":"2023-04-16T15:38:00+01:00","linkquality":255,"occupancy":false,"tamper":false,"voltage":2100}'
Debug 2023-04-16 15:38:02Saving state to file /config/zigbee2mqtt/state.json
Debug 2023-04-16 15:38:07Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"288zx-16","value":false}'
Info 2023-04-16 15:38:07Zigbee: disabling joining new devices.

Koenkk, 2023-04-16:
`_TZ3040_bb6xaihh` will be supported in the next z2m release on 1 May.

closed 2023-04-16 by Koenkk

jocamane, 2023-05-01:
Hi @Koenkk 
I don't see this device in the release notes 1.30.4 !!
It's normal ?

Koenkk, 2023-05-02:
@jocamane indeed not reflected in the release notes but it is added.

Anashost, 2023-05-20:
Hey,
I've tested this exact same sensor for couple of days, which doesn't work realibly at all.
Zigbee Manufacturer: _TZ3040_bb6xaihh
Zigbee Model: TS0202 - shows as TuYa TS0202_1 in the dashboard.

[image]

Issues:
- after pairing it, it detects and clears for couple of times (mostly 4 times) then it clears & stops responding, until i either:
     * remove the battery or 
     * re-pair the sensor or
     * wait for some time > 10-15 minutes.

    and so on.
- missing some features like tamper, and keep_time options (30-60-120 seconds).
- shows wrong device image in dashboard.

I have similar ones which are the TuYa IH012-RT01 (it looks exactly like this TS0202_1) works perfectly since over 6 month. So i wonder if this one can be fixed. Really appreciate your work.

logs when pairing the device:
```
Info 2023-05-20 23:42:48Zigbee: allowing new devices to join.
Info 2023-05-20 23:42:48MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"al8wt-8"}'
Info 2023-05-20 23:42:59Device '0xa4c138feb8b9437e' joined
Info 2023-05-20 23:42:59MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138feb8b9437e","ieee_address":"0xa4c138feb8b9437e"},"type":"device_joined"}'
Info 2023-05-20 23:42:59MQTT publish: topic 'homeassistant/binary_sensor/0xa4c138feb8b9437e/occupancy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138feb8b9437e"],"manufacturer":"TuYa","model":"Motion sensor (TS0202_1)","name":"0xa4c138feb8b9437e"},"device_class":"motion","name":"0xa4c138feb8b9437e occupancy","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xa4c138feb8b9437e","unique_id":"0xa4c138feb8b9437e_occupancy_zigbee2mqtt","value_template":"{{ value_json.occupancy }}"}'
Info 2023-05-20 23:42:59MQTT publish: topic 'homeassistant/binary_sensor/0xa4c138feb8b9437e/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138feb8b9437e"],"manufacturer":"TuYa","model":"Motion sensor (TS0202_1)","name":"0xa4c138feb8b9437e"},"device_class":"battery","entity_category":"diagnostic","name":"0xa4c138feb8b9437e battery low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xa4c138feb8b9437e","unique_id":"0xa4c138feb8b9437e_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
Info 2023-05-20 23:42:59MQTT publish: topic 'homeassistant/sensor/0xa4c138feb8b9437e/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138feb8b9437e"],"manufacturer":"TuYa","model":"Motion sensor (TS0202_1)","name":"0xa4c138feb8b9437e"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0xa4c138feb8b9437e linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138feb8b9437e","unique_id":"0xa4c138feb8b9437e_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2023-05-20 23:42:59MQTT publish: topic 'homeassistant/sensor/0xa4c138feb8b9437e/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138feb8b9437e"],"manufacturer":"TuYa","model":"Motion sensor (TS0202_1)","name":"0xa4c138feb8b9437e"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0xa4c138feb8b9437e battery","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138feb8b9437e","unique_id":"0xa4c138feb8b9437e_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info 2023-05-20 23:42:59MQTT publish: topic 'homeassistant/sensor/0xa4c138feb8b9437e/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138feb8b9437e"],"manufacturer":"TuYa","model":"Motion sensor (TS0202_1)","name":"0xa4c138feb8b9437e"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","name":"0xa4c138feb8b9437e voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138feb8b9437e","unique_id":"0xa4c138feb8b9437e_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
Info 2023-05-20 23:42:59Starting interview of '0xa4c138feb8b9437e'
Info 2023-05-20 23:42:59MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138feb8b9437e","ieee_address":"0xa4c138feb8b9437e","status":"started"},"type":"device_interview"}'
Info 2023-05-20 23:42:59MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138feb8b9437e","ieee_address":"0xa4c138feb8b9437e"},"type":"device_announce"}'
Info 2023-05-20 23:43:01Successfully interviewed '0xa4c138feb8b9437e', device has successfully been paired
Info 2023-05-20 23:43:01Device '0xa4c138feb8b9437e' is supported, identified as: TuYa Motion sensor (TS0202_1)
Info 2023-05-20 23:43:01MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Motion sensor","exposes":[{"access":1,"description":"Indicates whether the device detected occupancy","name": [... cut ...]
Info 2023-05-20 23:59:14MQTT publish: topic 'zigbee2mqtt/0xa4c138feb8b9437e', payload '{"battery":null,"battery_low":false,"linkquality":105,"occupancy":false,"tamper":false,"voltage":null}'
Info 2023-05-20 23:59:19MQTT publish: topic 'zigbee2mqtt/0xa4c138feb8b9437e', payload '{"battery":null,"battery_low":false,"linkquality":105,"occupancy":true,"tamper":false,"voltage":null}'
Info 2023-05-21 00:00:49MQTT publish: topic 'zigbee2mqtt/0xa4c138feb8b9437e', payload '{"battery":null,"battery_low":false,"linkquality":105,"occupancy":false,"tamper":false,"voltage":null}'
```

jocamane, 2023-05-24:
Hi @Koenkk

In my case, it respects the time but only when it goes from busy to clear. I put 20s and work. 
to go from clear to busy only after 90 seconds.
detected - wait 20s - clear - 90s - detected
If in 90s exist moviment, the counter reset and is more 90s
I confirm that the image is wrong.
I couldn't get the equipment to stop responding, the delay is those 90 seconds to become available again after becoming clear

Anashost, 2023-05-24:
@jocamane does it stay in "occupied" state if you keep moving? in my case most of the time it doesn't, it almost always go back to clear after about 70-80 seconds regardless of the movement, and then it becomes un-responsive for 10-15 minutes. Until i can use it again.

I have 2 TS0202_1 (TZ3040_bb6xaihh)
Both have same issue. I threw them in the drawer for now, and will follow this issue.

jocamane, 2023-05-24:
Hi @Anashost 

You are right. In my case i put 20s in setting, and after 20s clear regardless of whether or not there is movement.

but if you don't make any movement after 90s it's back to being good to work

I have 2 TS0202_1 (TZ3040_bb6xaihh) too.

I just did the test with a tuya zigbee hub and everything works OK
It works with 60s which should be the default and cannot be changed but work ok

Anashost, 2023-05-24:
This sensor is not usable for me in it's current state, as i eant it to stay "occupied" while ther is movement.

I don't know if this can help determining if its behavior is fixable, here is images of the board:

- Zigbee Model: TS0202
- Zigbee Manufacturer: _TZ3040_bb6xaihh
- Model: TS0202_1

[image]
[image]

Here's also images of the other sensor (looks the same from the outside) that im using since 6 months without any issues:

- Zigbee Model: TS0202
- Zigbee Manufacturer: _TZ3000_6ygjfyll
- Model: IH012-RT01

[image]
[image]

jocamane, 2023-05-27:
Hi @Koenkk

can you do something?

thanks

Koenkk, 2023-05-27:
Can you provide the debug log when moving for 2 minutes in front of the sensor?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

jocamane, 2023-05-27:
The device is '0xa4c138c73d4bfacd'

info  2023-05-27 21:07:12: Successfully changed options
info  2023-05-27 21:07:12: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"4fmar-3"}'
debug 2023-05-27 21:07:16: Received Zigbee message from 'Tomada 10 ZG', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID null
info  2023-05-27 21:07:16: MQTT publish: topic 'zigbee2mqtt/Tomada 10 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":89.06,"indicator_mode":"off/on","last_seen":"2023-05-27T21:07:16+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":228}'
debug 2023-05-27 21:07:25: Received Zigbee message from 'Tomada 8 ZG', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1575]}' from endpoint 1 with groupID null
info  2023-05-27 21:07:25: MQTT publish: topic 'zigbee2mqtt/Tomada 8 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":15.75,"indicator_mode":"off/on","last_seen":"2023-05-27T21:07:25+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":227}'
debug 2023-05-27 21:07:26: Received Zigbee message from 'Tomada 8 ZG', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":227}' from endpoint 1 with groupID null
info  2023-05-27 21:07:26: MQTT publish: topic 'zigbee2mqtt/Tomada 8 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":15.75,"indicator_mode":"off/on","last_seen":"2023-05-27T21:07:26+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":227}'
debug 2023-05-27 21:07:28: Received Zigbee message from 'Int Casa Máq ZG', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":82}' from endpoint 1 with groupID null
info  2023-05-27 21:07:28: MQTT publish: topic 'zigbee2mqtt/Int Casa Máq ZG', payload '{"last_seen":"2023-05-27T21:07:28+01:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
debug 2023-05-27 21:07:29: Received Zigbee message from 'Tomada 4 ZG', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":69}' from endpoint 1 with groupID null
info  2023-05-27 21:07:29: MQTT publish: topic 'zigbee2mqtt/Tomada 4 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":0.28,"indicator_mode":"off/on","last_seen":"2023-05-27T21:07:29+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":226}'
debug 2023-05-27 21:07:31: Received Zigbee message from 'Tomada 8 ZG', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":69}' from endpoint 1 with groupID null
info  2023-05-27 21:07:31: MQTT publish: topic 'zigbee2mqtt/Tomada 8 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":15.75,"indicator_mode":"off/on","last_seen":"2023-05-27T21:07:31+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":227}'
debug 2023-05-27 21:07:38: Received Zigbee message from 'Int WC1 ZG', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
info  2023-05-27 21:07:38: MQTT publish: topic 'zigbee2mqtt/Int WC1 ZG', payload '{"backlight_mode":"medium","last_seen":"2023-05-27T21:07:38+01:00","linkquality":255,"power_on_behavior":"previous","power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
info  2023-05-27 21:07:38: MQTT publish: topic 'zigbee2mqtt/Int WC1 ZG/mode', payload '{"backlight":"medium"}'
debug 2023-05-27 21:07:38: Successfully pinged 'Int WC1 ZG' (attempt 1/2)
debug 2023-05-27 21:07:43: Received Zigbee message from 'Sen Mov Sala ZG', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID null
info  2023-05-27 21:07:43: MQTT publish: topic 'zigbee2mqtt/Sen Mov Sala ZG', payload '{"battery":100,"battery_low":false,"keep_time":60,"last_seen":"2023-05-27T21:07:43+01:00","linkquality":255,"occupancy":true,"sensitivity":"medium","tamper":false,"voltage":2900}'
debug 2023-05-27 21:07:45: Received Zigbee message from 'Sen Mov Corredor ZG', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID null
info  2023-05-27 21:07:45: MQTT publish: topic 'zigbee2mqtt/Sen Mov Corredor ZG', payload '{"battery":92,"battery_low":false,"last_seen":"2023-05-27T21:07:45+01:00","linkquality":255,"occupancy":false,"tamper":false,"voltage":2100}'
[... cut ...]
debug 2023-05-27 21:10:48: Received Zigbee message from 'Int Quintal Fora ZG', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
info  2023-05-27 21:10:48: MQTT publish: topic 'zigbee2mqtt/Int Quintal Fora ZG', payload '{"backlight_mode":null,"last_seen":"2023-05-27T21:10:48+01:00","linkquality":255,"power_on_behavior":"previous","power_on_behavior_left":"previous","state_left":"OFF","state_right":"OFF"}'
info  2023-05-27 21:10:48: MQTT publish: topic 'zigbee2mqtt/Int Quintal Fora ZG/mode', payload '{"backlight":null}'
debug 2023-05-27 21:10:48: Successfully pinged 'Int Quintal Fora ZG' (attempt 1/2)

Koenkk, 2023-05-28:
Thanks, can you now also provide the logs when trigger occupancy and doing nothing for 5 minutes after that?

jocamane, 2023-05-28:
The device name is TEST:

info  2023-05-28 18:40:30: Successfully changed options
info  2023-05-28 18:40:30: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"qjbqj-3"}'
debug 2023-05-28 18:40:34: Received Zigbee message from 'Int Casa Máq ZG', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":82}' from endpoint 1 with groupID null
info  2023-05-28 18:40:34: MQTT publish: topic 'zigbee2mqtt/Int Casa Máq ZG', payload '{"last_seen":"2023-05-28T18:40:34+01:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
debug 2023-05-28 18:40:35: Received Zigbee message from 'Tomada 4 ZG', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,28]}' from endpoint 1 with groupID null
info  2023-05-28 18:40:35: MQTT publish: topic 'zigbee2mqtt/Tomada 4 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":0.28,"indicator_mode":"off/on","last_seen":"2023-05-28T18:40:35+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":233}'
debug 2023-05-28 18:40:37: Received Zigbee message from 'Test', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID null
info  2023-05-28 18:40:37: MQTT publish: topic 'zigbee2mqtt/Test', payload '{"battery":87,"battery_low":false,"last_seen":"2023-05-28T18:40:37+01:00","linkquality":255,"occupancy":true,"tamper":false,"voltage":null}'
debug 2023-05-28 18:40:53: Received Zigbee message from 'Tomada 8 ZG', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":69}' from endpoint 1 with groupID null
info  2023-05-28 18:40:53: MQTT publish: topic 'zigbee2mqtt/Tomada 8 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":15.75,"indicator_mode":"off/on","last_seen":"2023-05-28T18:40:53+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":234}'
debug 2023-05-28 18:40:56: Received Zigbee message from 'Tomada 4 ZG', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":69}' from endpoint 1 with groupID null
info  2023-05-28 18:40:56: MQTT publish: topic 'zigbee2mqtt/Tomada 4 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":0.28,"indicator_mode":"off/on","last_seen":"2023-05-28T18:40:56+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":233}'
debug 2023-05-28 18:41:02: Received Zigbee message from 'Tomada 4 ZG', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":233}' from endpoint 1 with groupID null
info  2023-05-28 18:41:02: MQTT publish: topic 'zigbee2mqtt/Tomada 4 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":0.28,"indicator_mode":"off/on","last_seen":"2023-05-28T18:41:02+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":233}'
debug 2023-05-28 18:41:19: Received Zigbee message from 'Tomada 8 ZG', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":234}' from endpoint 1 with groupID null
info  2023-05-28 18:41:19: MQTT publish: topic 'zigbee2mqtt/Tomada 8 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":15.75,"indicator_mode":"off/on","last_seen":"2023-05-28T18:41:19+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":234}'
debug 2023-05-28 18:41:48: Received Zigbee message from 'Test', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID null
info  2023-05-28 18:41:48: MQTT publish: topic 'zigbee2mqtt/Test', payload '{"battery":87,"battery_low":false,"last_seen":"2023-05-28T18:41:48+01:00","linkquality":255,"occupancy":false,"tamper":false,"voltage":null}'
debug 2023-05-28 18:41:48: Received Zigbee message from 'Tomada 10 ZG', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID null
info  2023-05-28 18:41:48: MQTT publish: topic 'zigbee2mqtt/Tomada 10 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":89.06,"indicator_mode":"off/on","last_seen":"2023-05-28T18:41:48+01:00","linkquality":248,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":234}'
debug 2023-05-28 18:42:02: Received Zigbee message from 'Tomada 4 ZG', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":233}' from endpoint 1 with groupID null
[... cut ...]
debug 2023-05-28 18:45:34: Received Zigbee message from 'Tomada 4 ZG', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,28]}' from endpoint 1 with groupID null
info  2023-05-28 18:45:34: MQTT publish: topic 'zigbee2mqtt/Tomada 4 ZG', payload '{"child_lock":"UNLOCK","current":0,"energy":0.28,"indicator_mode":"off/on","last_seen":"2023-05-28T18:45:34+01:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":231}'

Koenkk, 2023-05-29:
Could you check if the issue is fixed with the following external converter:  https://gist.github.com/Koenkk/36a52ebffb0b3c5f2f08cd68011c0eff

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Anashost, 2023-05-29:
Wow, looks like it's working nicely now.
It detects for 70 seconds, and clears if there's no movement. If there was movement it does holds its occupied state.

Tested two _TZ3040_bb6xaihh sensors, both working, Thanks a lot @Koenkk 
And thanks to you @jocamane  for providing logs.

jocamane, 2023-05-29:
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/36a52ebffb0b3c5f2f08cd68011c0eff
>
> * save this as file next to `configuration.yaml` as [...]

- Correct device image in dashboard
- Work OK after 70s clear-ocupped-clear

5 stars like always

can move on to the next offical version of Z2M

Thanks @Koenkk and @Anashost for the help

Koenkk, 2023-05-30:
Integrated the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Anashost, 2023-06-01:
Hey @Koenkk
I've just updated to v1.31.0-1, which includes the fix for (_TZ3040_bb6xaihh)
but it shows as "NOT supported".

Already deleted the converter file, and removed the mention from zigbee2mqtt configrations.yaml. Any idea whats going on. (as i have a small zigbee network, i have already reinstalled zigbee2mqtt, still same issue).

also device image is wrong wirh red glow on it!

```
Info 2023-06-01 12:33:21Zigbee: allowing new devices to join.
Info 2023-06-01 12:33:21MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"5buwk-1"}'
Info 2023-06-01 12:33:33Device '0xa4c138feb8b9437e' joined
Info 2023-06-01 12:33:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138feb8b9437e","ieee_address":"0xa4c138feb8b9437e"},"type":"device_joined"}'
Info 2023-06-01 12:33:33Starting interview of '0xa4c138feb8b9437e'
Info 2023-06-01 12:33:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138feb8b9437e","ieee_address":"0xa4c138feb8b9437e","status":"started"},"type":"device_interview"}'
Info 2023-06-01 12:33:39Successfully interviewed '0xa4c138feb8b9437e', device has successfully been paired
Warning 2023-06-01 12:33:39Device '0xa4c138feb8b9437e' with Zigbee model 'TS0202' and manufacturer name '_TZ3040_bb6xaihh' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-06-01 12:33:39MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138feb8b9437e","ieee_address":"0xa4c138feb8b9437e","status":"successful","supported":false},"type":"device_interview"}'
Warning 2023-06-01 12:34:45Received message from unsupported device with Zigbee model 'TS0202' and manufacturer name '_TZ3040_bb6xaihh'
Warning 2023-06-01 12:34:45Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

Koenkk, 2023-06-01:
Whoops, I made a mistake, correct it now. You can use the external converter until the next release.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Anashost, 2023-06-01:
It's ok boss, thanks.

chunkysteveo, 2023-06-01:
Same here - the _TZ3040_bb6xaihh was working till the latest version updated. I'll look how to add a convertor file till the next release, eek!

Anashost, 2023-06-01:
@chunkysteveo The converter provided by Koenkk above, works perfectly, and fixes _TZ3040_bb6xaihh previous issues.

chunkysteveo, 2023-06-01:
> @chunkysteveo The converter provided by Koenkk above, works perfectly, and fixes _TZ3040_bb6xaihh previous issues.

haha yeah, I spent about 30 mins trying to work out what how to make a convertor for a Tuya device, and then found that link to the test one - worked in about 5 seconds!! haha, thanks to all!!!

milandzuris, 2023-06-05:
[image]
[image]
I dont understand why when im change CFG in HA make nothing, and in Z2M CFG it works.

fixed77, 2023-11-12:
How i can change keep_time options (30-60-120 seconds)?
default 60s is too long

Nisbo, 2024-01-22:
I am also missing time options (found this thread via google)

vmedunytsia, 2024-02-06:
keep_time options (30-60-120 seconds)?