Koenkk/zigbee2mqtt #17369: [New device support]: Legrand 600087L GreenPower_3 blind control

Pulsahr, 2023-04-16:
### Link

https://www.legrand.fr/pro/catalogue/commande-sans-fils-sans-pile-self-e-dooxie-with-netatmo-pour-volets-roulants-connectes-blanc

### Database entry

{"id":6,"type":"GreenPower","ieeeAddr":"0x0000000000573646","nwkAddr":13894,"manufId":null,"modelId":"GreenPower_3","epList":[242],"endpoints":{"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1681676434155,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,

I tried to follow the new supporting device page but got stuck at keeping error message when clicking on the button.
Here is the said message in the log :

```
No converter available for '600087L' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[138,151,255],"type":"Buffer"}},"commandID":52,"frameCounter":156,"gppGddLink":255,"gppNwkAddr":38794,"options":21664,"payloadSize":0,"srcID":5715526}'
```

For some reason, at a point, the Z2B device web UI took into account what I defined in "description" property, but since then, I tried some modifications and they are simply ignored.

Steps I followed:
1. ✔️  pairing with zigbee2MQTT
2. ✔️  adding device: created /config/zigbee2mqtt/legrand-600087L.js and added external converters as instructed
```
external_converters:
  - legrand-600087L.js 
```
And `log_level: debug` in advanced section.
3. 🔴  edited converter until no more error message

I tried a lot things, starting with copying an existing similar device (`ZLGP15` in [devices/legrand.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/legrand.js])) and editing what should be edited.
I also added manual function into "fromZigbee" (heavily inspired from `fz.legrand_zlgp15` in [/converters/fromZigbee.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js)) because that switch is not yet supported and command IDs (52,53,54) are not available.
I restarted HASS IO each time after an edit, because for some reason just restarting Z2B addon ended up with a 502 in web UI.
I ended up with this external converer in `legrand-600087L.js`, more or less (I edited so much I don't remember).

I'll call it a day and check back tomorrow with renewed energy. Maybe I'll try following the example [here](https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/externalConvertersExample/freepad_ext.js), but I'm not confident in this approach.

Thank you for any help :)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['GreenPower_3'],
        fingerprint: [{modelID: 'GreenPower_3', ieeeAddr: '0x0000000000573646'}],
        model: '600087L',
        vendor: 'Legrand',
        description: 'Wireless and batteryless blind control',
        fromZigbee: [{
        cluster: 'greenPower',
        type: ['commandNotification'],
        convert: (model, msg, publish, options, meta) => {
            const commandID = msg.data.commandID;
            if (hasAlreadyProcessedMessage(msg, model, msg.data.frameCounter, `${msg.device.ieeeAddr}_${commandID}`)) return;
            if (commandID === 224) return; 
            const lookup = {0x34: 'cover_stop', 0x35: 'cover_up', 0x36: 'cover_down'};
            if (!lookup.hasOwnProperty(commandID)) {
                meta.logger.error(`custom from zigbee: missing command '${commandID}'`);
            } else {
                return {action: lookup[commandID]};
            }
        }
        }],
        toZigbee: [],
        exposes: [e.action(['cover_stop', 'cover_up', 'cover_down'])],
    };

module.exports = definition;
```

Koenkk, 2023-04-17:
Use:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['GreenPower_3'],
        fingerprint: [{modelID: 'GreenPower_3', ieeeAddr: '0x0000000000573646'}],
        model: '600087L',
        vendor: 'Legrand',
        description: 'Wireless and batteryless blind control',
        fromZigbee: [{
        cluster: 'greenPower',
        type: ['commandNotification'],
        convert: (model, msg, publish, options, meta) => {
            const commandID = msg.data.commandID;
            if (utils.hasAlreadyProcessedMessage(msg, model, msg.data.frameCounter, `${msg.device.ieeeAddr}_${commandID}`)) return;
            if (commandID === 224) return; 
            const lookup = {0x34: 'cover_stop', 0x35: 'cover_up', 0x36: 'cover_down'};
            if (!lookup.hasOwnProperty(commandID)) {
                meta.logger.error(`custom from zigbee: missing command '${commandID}'`);
            } else {
                return {action: lookup[commandID]};
            }
        }
        }],
        toZigbee: [],
        exposes: [e.action(['cover_stop', 'cover_up', 'cover_down'])],
    };

module.exports = definition;
```

Pulsahr, 2023-04-18:
I tried everything that is in the scope of my understanding, without making it work again.
I also tried after removing the device (force delete), a full reboot (plug power off then on), and inside another browser. 
Note : after plug off/on reboot, the device reappeared in Z2M web UI. So there is a something somewhere that messes with Z2M data.
Did force delete again, pairing, and still stuck at unsupported device warning.
I'll give all useful elements I can think of.

### zigbee2mqtt addon log
Found in HASS UI via settings > addons > Zigbee2MQTT, tab "log"
```
Zigbee2MQTT:debug 2023-04-18 18:31:07: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"1nbg7-1","value":true}'
Zigbee2MQTT:info  2023-04-18 18:31:07: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-04-18 18:31:07: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"1nbg7-1"}'
Zigbee2MQTT:info  2023-04-18 18:31:13: Device '0x0000000000573646' joined
Zigbee2MQTT:info  2023-04-18 18:31:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0000000000573646","ieee_address":"0x0000000000573646"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-04-18 18:31:13: Successfully interviewed '0x0000000000573646', device has successfully been paired
Zigbee2MQTT:warn  2023-04-18 18:31:13: Device '0x0000000000573646' with Zigbee model 'GreenPower_3' and manufacturer name 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:info  2023-04-18 18:31:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x0000000000573646","ieee_address":"0x0000000000573646","status":"successful","supported":false},"type":"device_interview"}'
Zigbee2MQTT:debug 2023-04-18 18:31:13: Received Zigbee message from '0x0000000000573646', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"applicationInfo":4,"deviceID":3,"extendedOptions":242,"gpdClientClusters":{"data":[],"type":"Buffer"},"gpdCommandIdList":{"data":[52,53,54],"type":"Buffer"},"gpdServerClusters":{"data":[],"type":"Buffer"},"keyMic":1828252672,"manufacturerID":0,"modelID":0,"numClientClusters":0,"numGdpCommands":3,"numServerClusters":0,"options":197,"outgoingCounter":338,"securityKey":{"data":[195,217,190,59,85,56,41,118,198,172,39,54,185,234,24,34],"type":"Buffer"}},"commandID":224,"frameCounter":82,"gppGddLink":75,"gppNwkAddr":55693,"options":2048,"payloadSize":32,"srcID":5715526}' from endpoint 242 with groupID 0
Zigbee2MQTT:warn  2023-04-18 18:31:13: Received message from unsupported device with Zigbee model 'GreenPower_3' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2023-04-18 18:31:13: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2023-04-18 18:31:13: Received Zigbee message from '0x0000000000573646', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[141,217,75],"type":"Buffer"}},"commandID":225,"frameCounter":343,"gppGddLink":75,"gppNwkAddr":55693,"options":2336,"payloadSize":0,"srcID":5715526}' from endpoint 242 with groupID 0
Zigbee2MQTT:warn  2023-04-18 18:31:13: Received message from unsupported device with Zigbee model 'GreenPower_3' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2023-04-18 18:31:13: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

### Zigbee2MQTT addon configuration
Found in HASS UI via settings > addons > Zigbee2MQTT, tab "configuration" > edit as YAML
```yaml
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto
  user: mqtt-user
  password: *
serial:
  port: /dev/ttyUSB0
external_converters:
  - legrand-600087L.js
advanced:
  log_level: debug
```

### cat config/zigbee2mqtt/configuration.yaml
```yaml
homeassistant: true
mqtt:
  server: mqtt://core-mosquitto
  user: mqtt-user
  password: *
serial:
  port: /dev/ttyUSB0
frontend:
  port: 8099
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
[... 56 lines cut ...]
    vendor: 'Legrand',
    description: 'Wireless and batteryless blind control',
    fromZigbee: [
        {
            cluster: 'greenPower',
            type: ['commandNotification', 'commandCommisioningNotification'],
            convert: (model, msg, publish, options, meta) => {
                const commandID = msg.data.commandID;
                if (utils.hasAlreadyProcessedMessage(msg, model, msg.data.frameCounter, `${msg.device.ieeeAddr}_${commandID}`)) return;
                if (commandID === 224) return;
                const lookup = {52: 'cover_stop', 53: 'cover_up', 54: 'cover_down'};
                if (!lookup.hasOwnProperty(commandID)) {
                    meta.logger.error(`Legrand GreenPower3 600087L: missing command '${commandID}'`);
                } else {
                    return {action: lookup[commandID]};
                }
            }
        }
    ],
    toZigbee: [],
    exposes: [e.action(['cover_stop', 'cover_up', 'cover_down'])],
};
```

Thank you for any help.

Koenkk, 2023-04-18:
I can integrate the changes if you give me the final converter version.

Pulsahr, 2023-04-18:
> I can integrate the changes if you give me the final converter version.

Thanks for the offer ! It would be a last resort but still better than being stuck.

I don't know how the converter should be, I can't make it work in my network so I'm not sure if it is the correct one.
I however can confirm there is 3 command IDs: 52, 53, and 54 linked respectively to the actions cover_stop, cover_up, cover_down like I wrote in my external converter. I juste can't make it work and I don't know why. Any idea ?

Koenkk, 2023-04-19:
But it works when using the external converter, right?

Pulsahr, 2023-04-19:
That's the catch, it worked at one point (why did I forgot to make a backup then?), then it does not work anymore,
All you see in [this comment](https://github.com/Koenkk/zigbee2mqtt/issues/17369#issuecomment-1513479315) is the current environment, with external converter name set in zigbee2MQTT configuration, and with the content that worked earlier (also quoted in the comment). I don't understand why it does not work anymore.

I even tried to copy the converter into another file and changed the configuration accordingly, no changes.
I'm a web developer and it looks a lot like bad cache problem, but that's not web, I might be completely wrong. I'm honestly lost.

Would it be related to mqtt retained message ? I've read mention of it but did not dig further (did not find a quick purge method).

If I can provide any additional information to simulate my problem (data sent or something) so you can try to reproduce it, just ask, I'll be glad to help.

Koenkk, 2023-04-20:
- Could you explain me what doesn't work anymore?
- Can you verify that the contents of the external converter are correct? (check the z2m data folder)

Koenkk, 2023-04-21:
> configuration (HASS UI > settings > addons > zigbee2mqtt > configuration tab > edit as yaml).

- Can you make a screenshot of this?
- I assume you already restarted the addon after adding the external converter?

Pulsahr, 2023-04-21:
Here is the screenshot: https://i.imgur.com/aSV5jXN.png

And yes, I restarted the addon. 
It didn't work, restarted HASS via UI, didn't work, plug off the RPI from current, plugged in back, still not working.

We can do it! 💪

Koenkk, 2023-04-22:
Can you try to add it to the z2m configuration.yaml directly using the file editor?

Pulsahr, 2023-04-22:
Just did: https://imgur.com/a/sCzqvCE
Nothing changed.
Here is the log.
```
Zigbee2MQTT:debug 2023-04-22 11:35:36: Received Zigbee message from '0x0000000000573880', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[69,95,10],"type":"Buffer"}},"commandID":52,"frameCounter":37,"gppGddLink":10,"gppNwkAddr":24389,"options":21664,"payloadSize":0,"srcID":5716096}' from endpoint 242 with groupID 2948
Zigbee2MQTT:warn  2023-04-22 11:35:36: Received message from unsupported device with Zigbee model 'GreenPower_3' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2023-04-22 11:35:36: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```
Should I delete and re-pair the device?

Here are the 2 files `legrand-600087L.js` and `configuration.yaml` as they are in config/zigbee2mqtt directory, just downloaded via SSH & mobaXterm. I just edited configuration.yaml to remove the password. Didn't open the js in case there is some malformed character I don't see.
[inside-zigbee2mqtt.zip](https://github.com/Koenkk/zigbee2mqtt/files/11301020/inside-zigbee2mqtt.zip)

Koenkk, 2023-04-22:
I cannot explain why it is not loaded, what z2m version are you using?

Pulsahr, 2023-04-22:
Spoiler: Finally made it work! See below.

```
Zigbee2MQTT version
    [1.30.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.30.3) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator type
    zStack3x0

Coordinator revision
    20210708

Coordinator IEEE Address
    0x00124b0026b71fe6

Frontend version
    0.6.127
```

I decided to write a new file from scratch from mobaXterm/SSH and nano. I mean re-writing line by line manually copy pasting just a portion, instead of copy pasting full lines or file in case there is a bad character at the end of a line or something. I made a few mistakes (not on purpose), and z2m start crashed each time logging a parsing error and such. Which means z2m actually reads the file. That's a progress.  
After several trial and error like normal coding process implies, I ended up having a normal bootup and tadaaa it works 🍾   

So first of all, let me copy/paste the code, so I have a working copy somewhere I can refer to.  

WORKING CONVERTER
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  zigbeeModel: ["GreenPower_3"],
  fingerprint: [{modelID: "GreenPower_3", ieeeAddr: /^0x00000000005.....$/}],
  model: "600087L",
  vendor: "Legrand",
  description: "Wireless and batteryless blind control switch",
  fromZigbee: [{
    cluster: "greenPower",
    type: ["commandNotification"],
    convert: (model, msg, publish, options, meta) => {
      const commandID = msg.data.commandID;
      const lookup = {0x34: "cover_stop", 0x35: "cover_up", 0x36: "cover_down"};
      if(utils.hasAlreadyProcessedMessage(msg, model, msg.data.frameCounter, `${msg.device.ieeeAddr}_${commandID}`)) {
        meta.logger.debug(`GreenPower_3 command '${commandID}' already processed`);
        return;
      }
      if(commandID === 224) return;
      if(!lookup.hasOwnProperty(commandID)) {
        meta.logger.error(`GreenPower_3 error: missing command '${commandID}'`);
      }else{
        return {action: lookup[commandID]};
      }
    }
  }],
  toZigbee: [],
  exposes: [e.action(["cover_stop", "cover_up", "cover_down"])]
};

module.exports = definition;
```
And just in case, the file itself, because let's be extra cautious.
[Legrand_GreenPower_3.zip](https://github.com/Koenkk/zigbee2mqtt/files/11302166/Legrand_GreenPower_3.zip)

We did it !
Thank you for your patience and your time. I still don't know what was the error, but I have a working copy so that's a huge win. I just need to make a pull request on relevant repository. But that will be for another day. First, a full backup !

Koenkk, 2023-04-23:
Great, looking forward to the PR!

Pulsahr, 2023-04-24:
Hello,
Sorry to bother you with such trivial things, but it seems I don't have the possibility to create a branch on herdmans converter repo, see screnshot below.
[image]

I planned to create a branch named "Legrand-600087L" or whatever standards are expected, to add the converter mentioned in this issue.
I also plan to make another PR for 2 other devices (see [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/17381)).
I don't know if I missed a "how-to" somewhere, or if I made a mistake, or if you need to allow me something, but currently I can't help with what I coded. Can you help me?

Koenkk, 2023-04-24:
You have to create a [fork](https://docs.github.com/en/get-started/quickstart/fork-a-repo), then edit the files and make a [pull request](https://docs.github.com/en/desktop/contributing-and-collaborating-using-github-desktop/working-with-your-remote-repository-on-github-or-github-enterprise/creating-an-issue-or-pull-request)

Loic691, 2024-03-08:
Hi @Pulsahr 
Finaly did you push a PR ?

Loic691, 2024-03-08:
Thanks for your response. I don't have this device. I have to get one before !
Other things, do you know if green power legrand device are updatable with firmware ?
Regards