Koenkk/zigbee2mqtt #17371: Battery devices leaving network

adams-family, 2023-04-17:
### What happened?

With zigbee2mqtt:latest frunning from hub.docker.com and using a `SONOFF ZigBee 3.0 Dongle E USB stick EFR32MG21`, all my battery devices randomly leave the network for good, usually in 1-24 hours after pairing.

Only repairing solves the problem.

The fastest to leave the network are Ikea switches (such as: [IKEA E1524/E1810](https://www.zigbee2mqtt.io/devices/E1524_E1810.html#ikea-e1524%252Fe1810))

### What did you expect to happen?

Battery devices should stay on the network.

### How to reproduce it (minimal and precise)

- Obtain a battery device, ideally [IKEA E1524/E1810](https://www.zigbee2mqtt.io/devices/E1524_E1810.html#ikea-e1524%252Fe1810)
- Obtain a **SONOFF ZigBee 3.0 Dongle E USB coordinator**
- Run the latest **zigbee2mqtt** from **Docker Hub**
- Pair the device
- Wait between 1-24 hours until the device disconnects

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF ZigBee 3.0 Dongle E USB (coordinator type EZSP v8)

### Debug log

```
Zigbee2MQTT:warn 2023-04-12 20:52:10: Device 'IkeaSwitchKitchen' left the network
Zigbee2MQTT:info 2023-04-12 20:52:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xccccccfffe9e9ccc"},"type":"device_removed"}'
Zigbee2MQTT:warn 2023-04-17 10:04:22: Device 'TuyaThermometer5' left the network
Zigbee2MQTT:info 2023-04-17 10:04:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xa4c1385d6c29dccc"},"type":"device_removed"}'
```

kirovilya, 2023-04-17:
why don't you update z2m and try on it before writing an issue?

adams-family, 2023-04-17:
@kirovilya Because I'm running the latest from hub.docker.com

kirovilya, 2023-04-17:
@adams-family please update to 1.30.3 at least, or to edge/dev version

adams-family, 2023-04-17:
@kirovilya How can I do it in docker?

I'm running:
```
docker compose build         # doesn't do anything, since I'm up to date
docker compose up -d
```

My docker-compose.yml:
```
version: '2.1'

services:
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt
    restart: always
    volumes:
      - ./zigbee2mqtt_data:/app/data
      - /run/udev:/run/udev:ro
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    ports:
      - 8080:8080
```

kirovilya, 2023-04-17:
@adams-family https://www.zigbee2mqtt.io/guide/installation/02_docker.html#updating

kirovilya, 2023-04-18:
@adams-family "edge" = "dev" :)
what is bug? show logs

adams-family, 2023-04-19:
@kirovilya As per my first post:

## What happened? ##

With zigbee2mqtt:latest frunning from hub.docker.com and using a SONOFF ZigBee 3.0 Dongle E USB stick EFR32MG21, all my battery devices randomly leave the network for good, usually in 1-24 hours after pairing.

## Logs ##

```
Zigbee2MQTT:warn 2023-04-12 20:52:10: Device 'IkeaSwitchKitchen' left the network
Zigbee2MQTT:info 2023-04-12 20:52:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xccccccfffe9e9ccc"},"type":"device_removed"}'
Zigbee2MQTT:warn 2023-04-17 10:04:22: Device 'TuyaThermometer5' left the network
Zigbee2MQTT:info 2023-04-17 10:04:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xa4c1385d6c29dccc"},"type":"device_removed"}'
```

adams-family, 2023-05-20:
All right guys, I couldn't wait anymore and frankly got a bit frustrated about the answers in this topic.

Replaced Z2M with an Ikea Dirigera, working flawlessly. Leaving this topic as I don't have Z2M anymore, can't be of any help over here.

closed 2023-06-27 by github-actions[bot]