Koenkk/zigbee2mqtt #17373: Device option retain via configuration environment variable

OlafKocanda, 2023-04-17:
### What happened?

According to the [configuration documentation](https://www.zigbee2mqtt.io/guide/configuration/) it should be possible to set an environment variable. This worked for variables like "ZIGBEE2MQTT_CONFIG_MQTT_SERVER" or "ZIGBEE2MQTT_CONFIG_PERMIT_JOIN". However, it does not seem to work for the retain option:

```
            - name: ZIGBEE2MQTT_CONFIG_DEVICE_OPTIONS_RETAIN
              value: "true"
```

When adding this, it simply does not show in the configuration.yaml

As a sitenote: I tried editing / copying the configuration.yaml to the pod while running, but since it seems to read the file, I cannot edit / replace it during runtime: 
```tar: can't remove old file configuration.yaml: Resource busy```

### What did you expect to happen?

```
            - name: ZIGBEE2MQTT_CONFIG_DEVICE_OPTIONS_RETAIN
              value: "true"
```

should add the following to the configuration.yaml:

```
device_options:
  retain: true
```

### How to reproduce it (minimal and precise)

This is my zigbee2mqtt deployment.yaml on kubernetes:

```
---
apiVersion: apps/v1
kind: Deployment
metadata:
  creationTimestamp: null
  labels:
    app: zigbee2mqtt
  name: zigbee2mqtt
spec:
  replicas: 1
  selector:
    matchLabels:
      app: zigbee2mqtt
  strategy:
    type: Recreate
  template:
    metadata:
      creationTimestamp: null
      labels:
        app: zigbee2mqtt
    spec:
      containers:
        - name: zigbee2mqtt
          image: koenkk/zigbee2mqtt:1.30.3
          imagePullPolicy: IfNotPresent
          env:
            - name: TZ
              value: Europe/Berlin
            - name: "ZIGBEE2MQTT_DATA"
              value: "/data"
            - name: ZIGBEE2MQTT_CONFIG_MQTT_SERVER
              value: "mqtt://mosquitto"
            - name: ZIGBEE2MQTT_CONFIG_PERMIT_JOIN
              value: "false"
            - name: ZIGBEE2MQTT_CONFIG_DEVICE_OPTIONS_RETAIN
              value: "true"
          securityContext:
            privileged: true
          ports:
            - name: http
              containerPort: 8080
              protocol: TCP
          volumeMounts:
            - name: zigbee2mqtt-data
              mountPath: /data
            - name: zigbee2mqtt-config
              mountPath: /app/configuration.yaml
              subPath: configuration.yaml
            - name: zigbee2mqtt-udev
              mountPath: /run/udev
            - name: ttyacm
              mountPath: /dev/ttyACM0
          livenessProbe:
            tcpSocket:
              port: 8080
            initialDelaySeconds: 0
            failureThreshold: 3
            timeoutSeconds: 1
            periodSeconds: 10
          readinessProbe:
            tcpSocket:
              port: 8080
            initialDelaySeconds: 0
            failureThreshold: 3
            timeoutSeconds: 1
            periodSeconds: 10
          startupProbe:
            tcpSocket:
              port: 8080
            initialDelaySeconds: 0
            failureThreshold: 30
            timeoutSeconds: 1
            periodSeconds: 5
      nodeSelector:
        kubernetes.io/hostname: "hostname"
      volumes:
        - name: zigbee2mqtt-data
          persistentVolumeClaim:
            claimName: zigbee2mqtt-data
        - name: zigbee2mqtt-udev
          hostPath:
            path: /run/udev
        - name: ttyacm
          hostPath:
            path: /dev/ttyACM0
        - name: zigbee2mqtt-config
          configMap:
            name: zigbee2mqtt-config
```

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

OlafKocanda, 2023-05-23:
Still not solved

OlafKocanda, 2023-06-25:
still not solved

closed 2023-08-04 by github-actions[bot]

kirianguiller, 2024-05-27:
I have the same problem with same setup. 
 ZIGBEE2MQTT_CONFIG_DEVICE_OPTIONS_RETAIN=true is set in a docker compose file but not taken into account on config file creation.

niki-on-github, 2025-12-28:
I'm having the same problem. I came across this issue while searching. The problem seems to be that only settings specified in the default/existing config can be set. See https://github.com/Koenkk/zigbee2mqtt/blob/8916fb7b50836627030915962f07c181fc1d1e8d/lib/util/settings.ts#L380-L388