Koenkk/zigbee2mqtt #17374: INNR RB 251 C firmware update "does not stick"

DoerakkerHA, 2023-04-17:
### What happened?

see notification update is available
starting update
update completes
configuration occurs
configuration completes
somewhere a new notification appears: update available
rinse and repeat

### What did you expect to happen?

update completes, configuration completes end of story

Info 2023-04-17 08:51:34MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"0xc498860000077ed0"},"meta":{"device":"03 voorraam 1","from":{"dateCode":"20211013","softwareBuildID":"2.03.00"},"status":"update_succeeded","to":{"dateCode":"20211013","softwareBuildID":"2.03.00"}},"type":"ota_update"}'
Info 2023-04-17 08:51:34MQTT publish: topic 'zigbee2mqtt/03 voorraam 1', payload '{"brightness":48,"color":{"h":22,"hue":22,"s":97,"saturation":97,"x":0.5492,"y":0.4082},"color_mode":"color_temp","color_temp":555,"color_temp_startup":555,"last_seen":1681714294639,"level_config":{"on_level":"previous"},"linkquality":72,"power_on_behavior":null,"state":"OFF","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
Info 2023-04-17 08:51:34MQTT publish: topic 'zigbee2mqtt/03 voorraam 1', payload '{"brightness":48,"color":{"h":22,"hue":22,"s":97,"saturation":97,"x":0.5492,"y":0.4082},"color_mode":"color_temp","color_temp":555,"color_temp_startup":555,"last_seen":1681714294715,"level_config":{"on_level":"previous"},"linkquality":76,"power_on_behavior":null,"state":"OFF","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
Info 2023-04-17 08:51:34Successfully configured '03 voorraam 1'

### How to reproduce it (minimal and precise)

see notification update is available
starting update
update completes
configuration occurs
configuration completes
somewhere a new notification appears: update available
rinse and repeat

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

20211217

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

Info 2023-04-17 08:51:34MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"0xc498860000077ed0"},"meta":{"device":"03 voorraam 1","from":{"dateCode":"20211013","softwareBuildID":"2.03.00"},"status":"update_succeeded","to":{"dateCode":"20211013","softwareBuildID":"2.03.00"}},"type":"ota_update"}'
Info 2023-04-17 08:51:34MQTT publish: topic 'zigbee2mqtt/03 voorraam 1', payload '{"brightness":48,"color":{"h":22,"hue":22,"s":97,"saturation":97,"x":0.5492,"y":0.4082},"color_mode":"color_temp","color_temp":555,"color_temp_startup":555,"last_seen":1681714294639,"level_config":{"on_level":"previous"},"linkquality":72,"power_on_behavior":null,"state":"OFF","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
Info 2023-04-17 08:51:34MQTT publish: topic 'zigbee2mqtt/03 voorraam 1', payload '{"brightness":48,"color":{"h":22,"hue":22,"s":97,"saturation":97,"x":0.5492,"y":0.4082},"color_mode":"color_temp","color_temp":555,"color_temp_startup":555,"last_seen":1681714294715,"level_config":{"on_level":"previous"},"linkquality":76,"power_on_behavior":null,"state":"OFF","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
Info 2023-04-17 08:51:34Successfully configured '03 voorraam 1'

m0wlheld, 2023-04-18:
Same here. Same device, same Z2M version, same coordinator device.

m0wlheld, 2023-04-19:
Just did another OTA run for a Innr RB 251 C. The status after the update is as expected:

```json
{
    "brightness": 179,
    "color": {
        "h": 40,
        "hue": 40,
        "s": 51,
        "saturation": 51,
        "x": 0.3827,
        "y": 0.3781
    },
    "color_mode": "color_temp",
    "color_temp": 253,
    "level_config": {
        "on_level": "previous"
    },
    "linkquality": 36,
    "state": "OFF",
    "update": {
        "installed_version": 604050705,
        "latest_version": 604050705,
        "state": "idle"
    }
}
```

when checking for available OTA for that device, the status "resets" to the previous version:

```json
{
    "brightness": 179,
    "color": {
        "h": 40,
        "hue": 40,
        "s": 51,
        "saturation": 51,
        "x": 0.3827,
        "y": 0.3781
    },
    "color_mode": "color_temp",
    "color_temp": 253,
    "level_config": {
        "on_level": "previous"
    },
    "linkquality": 51,
    "state": "OFF",
    "update": {
        "installed_version": 587207953,
        "latest_version": 604050705,
        "state": "available"
    },
    "power_on_behavior": null,
    "update_available": null
}
```

wolterink, 2023-04-29:
Same here, but for a [RB 286 C]
installed_version: 587207953
latest_version: 604050705

d-berk, 2023-05-07:
same here.
RB 251 C
Coordinator: SONOFF Zigbee 3.0 USB Dongle Plus
z2m Version 1.30.4

DoerakkerHA, 2023-06-07:
The issue persists on install running all latest updates....

Please pick up

DoerakkerHA, 2023-06-21:
hello all, this issue persists ... just tried to update 2 of these lights, updates ok (appears) then when querying the ota it returns as update available.  I can live with it but would like to know what we can do about it to address this issue

DoerakkerHA, 2023-07-22:
The issue persists while all components/HaOs are up to date

m0wlheld, 2023-08-23:
Still present issue.

DoerakkerHA, 2023-08-23:
This issue has not been resolved; it still reports via update. When update executed and zigbee2mqtt restarted error reappears

m0wlheld, 2023-09-03:
Issue is still present with 1.33.0.

Affected devices (here) are Innr RB 251 C with firmware date 20211013 and firmware version 2.03.00 . OTA is reported available and gets applied without issues. But when checking for firmware updates afterwards, OTA availability is reported again.

jstaro, 2023-09-04:
I am also seeing this with:
- RB 266, firmware build date 20211022, version 2.03.01
- RB 245, firmware build date 20211022, version 2.03.01

Interestingly, it has once worked to upgrade a RB 245 to 2.04.01 (several months ago) but at some point, this stopped working.

anthraxn8b, 2023-09-13:
The interesting thing is, that this happens with INNR bulbs but not with PHILIPS HUE bulbs...

My log where the problem occurs that the new version does not seem be be installed:

`info 2023-09-13 08:47:30Update of 'diningroom/table_light_0' at 99.57%, ≈ 0 minutes remaining
info 2023-09-13 08:47:30MQTT publish: topic 'zigbee2mqtt/diningroom/table_light_0', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"color_temp_startup":370,"linkquality":188,"power_on_behavior":null,"state":"ON","update":{"installed_version":587207953,"latest_version":604050705,"progress":99.57,"remaining":3,"state":"updating"},"update_available":true}'
info 2023-09-13 08:47:38Update of 'diningroom/table_light_0' at 100.00%
info 2023-09-13 08:47:38MQTT publish: topic 'zigbee2mqtt/diningroom/table_light_0', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"color_temp_startup":370,"linkquality":188,"power_on_behavior":null,"state":"ON","update":{"installed_version":587207953,"latest_version":604050705,"progress":100,"remaining":3,"state":"updating"},"update_available":true}'
...
info 2023-09-13 08:49:38Finished update of 'diningroom/table_light_0'
info 2023-09-13 08:49:38Configuring 'diningroom/table_light_0'
info 2023-09-13 08:49:38MQTT publish: topic 'zigbee2mqtt/diningroom/table_light_0', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"color_temp_startup":370,"linkquality":188,"power_on_behavior":null,"state":"ON","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":true}'
info 2023-09-13 08:49:38MQTT publish: topic 'zigbee2mqtt/diningroom/table_light', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"state":"ON"}'
info 2023-09-13 08:49:38MQTT publish: topic 'zigbee2mqtt/diningroom/table_light_0', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"color_temp_startup":370,"linkquality":188,"power_on_behavior":null,"state":"ON","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":true}'
info 2023-09-13 08:49:38MQTT publish: topic 'zigbee2mqtt/fam_room/light', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"state":"ON"}'
info 2023-09-13 08:49:38Device 'diningroom/table_light_0' was updated from '{"dateCode":"20211013","softwareBuildID":"2.03.00"}' to '{"dateCode":"20211013","softwareBuildID":"2.03.00"}'
info 2023-09-13 08:49:38MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20211013","software_build_id":"2.03.00"},"id":"diningroom/table_light_0","to":{"date_code":"20211013","software_build_id":"2.03.00"}},"status":"ok","transaction":"ljsp7-3"}'
info 2023-09-13 08:49:39MQTT publish: topic 'zigbee2mqtt/diningroom/table_light_0', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"color_temp_startup":370,"linkquality":188,"power_on_behavior":null,"state":"ON","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":true}'
info 2023-09-13 08:49:39Successfully configured 'diningroom/table_light_0'
...
info 2023-09-13 09:01:26Checking if update available for 'diningroom/table_light_0'
info 2023-09-13 09:01:27Update available for 'diningroom/table_light_0'
info 2023-09-13 09:01:27MQTT publish: topic 'zigbee2mqtt/diningroom/table_light_0', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"color_temp_startup":370,"linkquality":188,"power_on_behavior":null,"state":"ON","update":{"installed_version":587207953,"latest_version":604050705,"state":"available"},"update_available":true}'
info 2023-09-13 09:01:27MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"diningroom/table_light_0","updateAvailable":true},"status":"ok","transaction":"ljsp7-4"}'
`

m0wlheld, 2023-10-14:
Still present.

DoerakkerHA, 2023-10-14:
Updated to version 11.0
Issue persists

Koenkk, 2023-10-14:
Does it still occur with the latest dev branch? It contains some OTA fixes.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

m0wlheld, 2023-10-14:
Checking with -dev now …

m0wlheld, 2023-10-14:
No change. Device reports firmware 2.03.00 with update available. Update initiated okay, takes about 25min, but device afterwards still reports 2.03.00

DoerakkerHA, 2023-10-14:
Same hereKind RegardsROnald GroeneveldSent from my iPhone +31 650 989 689

Koenkk, 2023-10-14:
Can you provide the herdsman debug log of the complete firmware upgrade?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

DoerakkerHA, 2023-10-16:
Hallo Koen, the log is overwritten during the update (takes about 30+ minutes) so a full log is barely possible to provide. I will run another update and try and capture relevant sections for the light in questionKind RegardsROnald GroeneveldSent from my iPhone +31 650 989 689

DoerakkerHA, 2023-10-16:
Zigbee2MQTT:info  2023-10-16 06:57:22: MQTT publish: topic 'zigbee2mqtt/03 zijraam', payload '{"brightness":1,"color":{"h":31,"hue":31,"s":85,"saturation":85,"x":0.4717,"y":0.4129},"color_mode":"color_temp","color_temp":390,"last_seen":1697432242610,"level_config":{"on_level":"previous"},"linkquality":25,"power_on_behavior":null,"state":"ON","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
Zigbee2MQTT:debug 2023-10-16 06:57:22: Received Zigbee message from '03 zijraam', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20211013","swBuildId":"2.03.00"}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-10-16 06:57:22: MQTT publish: topic 'zigbee2mqtt/03 zijraam', payload '{"brightness":1,"color":{"h":31,"hue":31,"s":85,"saturation":85,"x":0.4717,"y":0.4129},"color_mode":"color_temp","color_temp":390,"last_seen":1697432242729,"level_config":{"on_level":"previous"},"linkquality":25,"power_on_behavior":null,"state":"ON","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2023-10-16 06:57:22: Device '03 zijraam' was updated from '{"dateCode":"20211013","softwareBuildID":"2.03.00"}' to '{"dateCode":"20211013","softwareBuildID":"2.03.00"}'
Zigbee2MQTT:info  2023-10-16 06:57:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"0xc49886000007a473"},"meta":{"device":"03 zijraam","from":{"dateCode":"20211013","softwareBuildID":"2.03.00"},"status":"update_succeeded","to":{"dateCode":"20211013","softwareBuildID":"2.03.00"}},"type":"ota_update"}'
Zigbee2MQTT:info  2023-10-16 06:57:23: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20211013","software_build_id":"2.03.00"},"id":"0xc49886000007a473","to":{"date_code":"20211013","software_build_id":"2.03.00"}},"status":"ok"}'
Zigbee2MQTT:debug 2023-10-16 06:57:23: Received Zigbee message from '03 zijraam', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorTempPhysicalMax":555,"colorTempPhysicalMin":153}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-10-16 06:57:23: MQTT publish: topic 'zigbee2mqtt/03 zijraam', payload '{"brightness":1,"color":{"h":31,"hue":31,"s":85,"saturation":85,"x":0.4717,"y":0.4129},"color_mode":"color_temp","color_temp":390,"last_seen":1697432243032,"level_config":{"on_level":"previous"},"linkquality":25,"power_on_behavior":null,"state":"ON","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2023-10-16 06:57:23: Successfully configured '03 zijraam'
Zigbee2MQTT:debug 2023-10-16 06:57:23: Received Zigbee message from '10 logeer plafond', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":285605891,"imageType":256,"manufacturerCode":4454}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-10-16 06:57:23: MQTT publish: topic 'zigbee2mqtt/10 logeer plafond', payload '{"brightness":117,"last_seen":1697432243348,"linkquality":87,"power_on_behavior":null,"state":"OFF"}'
Zigbee2MQTT:debug 2023-10-16 06:57:23: Device '10 logeer plafond' requested OTA
Zigbee2MQTT:debug 2023-10-16 06:57:23: Responded to OTA request of '10 logeer plafond' with 'NO_IMAGE_AVAILABLE'
Zigbee2MQTT:debug 2023-10-16 06:57:23: Retrieving state of '03 zijraam' after reconnectKind RegardsROnald GroeneveldSent from my iPhone +31 650 989 689

Koenkk, 2023-10-21:
@m0wlheld since you know how to edit the zigbee-herdsman-converters source code, in `node_modules/zigbee-herdsman-converters/lib/ota/common.js` can you try changing `upgradeTime: 1,` to `upgradeTime: 0,` around line 342?

m0wlheld, 2023-10-21:
Did, but no (visible) effect. The OTA update is reported as available and before update, the status of the device is:

Status before update:

```json
{
    "brightness": 203,
    "color": {
        "h": 39,
        "hue": 39,
        "s": 53,
        "saturation": 53,
        "x": 0.3863,
        "y": 0.3803
    },
    "color_mode": "color_temp",
    "color_temp": 258,
    "level_config": {
        "on_level": "previous"
    },
    "linkquality": 61,
    "state": "OFF",
    "update": {
        "installed_version": 587207953,
        "latest_version": 604050705,
        "state": "available"
    }
}
```

Then update is initiated and completed, lines printed to logfile are:

```
Zigbee2MQTT:info  2023-10-21 12:59:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9c69370000cbc69d","ieee_address":"0x9c69370000cbc69d"},"type":"device_announce"}'
Zigbee2MQTT:info  2023-10-21 12:59:55: Finished update of '0x9c69370000cbc69d'
Zigbee2MQTT:info  2023-10-21 12:59:55: Configuring '0x9c69370000cbc69d'
Zigbee2MQTT:info  2023-10-21 12:59:55: MQTT publish: topic 'zigbee2mqtt/0x9c69370000cbc69d', payload '{"brightness":203,"color":{"h":39,"hue":39,"s":53,"saturation":53,"x":0.3863,"y":0.3803},"color_mode":"color_temp","color_temp":258,"level_config":{"on_level":"previous"},"linkquality":65,"power_on_behavior":"on","state":"OFF","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-21 12:59:55: MQTT publish: topic 'zigbee2mqtt/Lichter Felix', payload '{"brightness":203,"color":{"h":39,"hue":39,"s":53,"saturation":53,"x":0.3863,"y":0.3803},"color_mode":"color_temp","color_temp":258,"state":"OFF"}'
Zigbee2MQTT:info  2023-10-21 12:59:55: MQTT publish: topic 'zigbee2mqtt/0x9c69370000cbc69d', payload '{"brightness":203,"color":{"h":39,"hue":39,"s":53,"saturation":53,"x":0.3863,"y":0.3803},"color_mode":"color_temp","color_temp":258,"level_config":{"on_level":"previous"},"linkquality":65,"power_on_behavior":"on","state":"OFF","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-21 12:59:55: MQTT publish: topic 'zigbee2mqtt/Stehlampe Felix', payload '{"brightness":203,"color":{"h":39,"hue":39,"s":53,"saturation":53,"x":0.3863,"y":0.3803},"color_mode":"color_temp","color_temp":258,"state":"OFF"}'
Zigbee2MQTT:info  2023-10-21 12:59:55: Device '0x9c69370000cbc69d' was updated from '{"dateCode":"20211013","softwareBuildID":"2.03.00"}' to '{"dateCode":"20211013","softwareBuildID":"2.03.00"}'
Zigbee2MQTT:info  2023-10-21 12:59:56: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20211013","software_build_id":"2.03.00"},"id":"0x9c69370000cbc69d","to":{"date_code":"20211013","software_build_id":"2.03.00"}},"status":"ok","transaction":"itqnc-1"}'
Zigbee2MQTT:info  2023-10-21 12:59:56: MQTT publish: topic 'zigbee2mqtt/0x9c69370000cbc69d', payload '{"brightness":203,"color":{"h":39,"hue":39,"s":53,"saturation":53,"x":0.3863,"y":0.3803},"color_mode":"color_temp","color_temp":258,"level_config":{"on_level":"previous"},"linkquality":65,"power_on_behavior":"on","state":"OFF","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-21 12:59:56: Successfully configured '0x9c69370000cbc69d'
Zigbee2MQTT:info  2023-10-21 12:59:56: MQTT publish: topic 'zigbee2mqtt/0x60a423fffe6852a5', payload '{"current":10.15,"energy":1465.31,"indicator_mode":"off/on","linkquality":120,"power":2300,"power_outage_memory":"restore","state":"ON","voltage":225}'
Zigbee2MQTT:info  2023-10-21 12:59:57: MQTT publish: topic 'zigbee2mqtt/Lichter Felix', payload '{"brightness":203,"color":{"h":39,"hue":39,"s":53,"saturation":53,"x":0.3863,"y":0.3803},"color_mode":"color_temp","color_temp":258,"state":"OFF"}'
Zigbee2MQTT:info  2023-10-21 12:59:57: MQTT publish: topic 'zigbee2mqtt/0x9c69370000cbc69d', payload '{"brightness":203,"color":{"h":39,"hue":39,"s":53,"saturation":53,"x":0.3863,"y":0.3803},"color_mode":"color_temp","color_temp":258,"level_config":{"on_level":"previous"},"linkquality":61,"power_on_behavior":"on","state":"OFF","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-21 12:59:57: MQTT publish: topic 'zigbee2mqtt/Stehlampe Felix', payload '{"brightness":203,"color":{"h":39,"hue":39,"s":53,"saturation":53,"x":0.3863,"y":0.3803},"color_mode":"color_temp","color_temp":258,"state":"OFF"}'
```

The status after the update is 

```json
{
    "brightness": 203,
    "color": {
        "h": 39,
        "hue": 39,
        "s": 53,
        "saturation": 53,
        "x": 0.3863,
        "y": 0.3803
    },
    "color_mode": "color_temp",
    "color_temp": 258,
    "level_config": {
        "on_level": "previous"
    },
    "linkquality": 65,
    "state": "OFF",
    "update": {
        "installed_version": 604050705,
        "latest_version": 604050705,
        "state": "available"
    }
}
```

so the "installed_version" attribute has changed and reflects the "latest_version" attribute's value. But when returning to OTA page (event without checking for an OTA) the status is reverted to the previous version, as it was before the update.

m0wlheld, 2023-10-21:
Full debug log from start OTA to end OTA and returning to previous version.

[z2m-17374.log.gz](https://github.com/Koenkk/zigbee2mqtt/files/13060579/z2m-17374.log.gz)

Koenkk, 2023-10-22:
@m0wlheld could you send me the herdsman debug log?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

m0wlheld, 2023-10-22:
It‘s in the archive, I run Z2M in a container, so everything is logged to stdout.

Koenkk, 2023-10-22:
@m0wlheld as far as I can see, this file contains 1 log file which is the normal z2m debug logging (not the herdsman one)

m0wlheld, 2023-10-25:
Created a new archive using the instructions from https://www.zigbee2mqtt.io/guide/usage/debug.html . Result (gzip compressed) is attached.

[log.txt.gz](https://github.com/Koenkk/zigbee2mqtt/files/13162659/log.txt.gz)

Koenkk, 2023-10-25:
@m0wlheld In the log I see the following:

```
2023-10-25T06:58:35.372Z zigbee-herdsman:controller:endpoint CommandResponse 0x9c69370000cbc69d/1 genOta.upgradeEndResponse({"manufacturerCode":4454,"imageType":300,"fileVersion":604050705,"imageSize":291130,"currentTime":0,"upgradeTime":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
```

`"upgradeTime":1` -> it seems the change from https://github.com/Koenkk/zigbee2mqtt/issues/17374#issuecomment-1773708885 got lost? I would expect `"upgradeTime":0` here.

m0wlheld, 2023-10-30:
[log-20231030-z2m.txt.gz](https://github.com/Koenkk/zigbee2mqtt/files/13205656/log-20231030-z2m.txt.gz)

Next try logging with `upgradeTime: 1` enabled. No success with OTA update however.

Koenkk, 2023-10-30:
@m0wlheld this log still has `"upgradeTime":1`, I would expect `"upgradeTime":0`. Did you apply the change as I suggested in https://github.com/Koenkk/zigbee2mqtt/issues/17374#issuecomment-1773708885 ?

m0wlheld, 2023-10-30:
I did and restarted the container afterwards. This should not reset the image‘s content. Is it required to restart after changing the file?Regards Christoph Dahlen

Koenkk, 2023-10-30:
@m0wlheld the container needs to be restarted but with `docker restart CONTAINER_ID`, not via e.g. HA OS (if you are using the HA addon)

m0wlheld, 2023-10-30:
```json
                const payload = {
                    manufacturerCode: image.header.manufacturerCode, imageType: image.header.imageType,
                    fileVersion: image.header.fileVersion, imageSize: image.header.totalImageSize,
                    currentTime: 0, upgradeTime: 0,
                };
```

Repeating test with this setting now, gathering logs ...

m0wlheld, 2023-10-30:
[log.txt.gz](https://github.com/Koenkk/zigbee2mqtt/files/13209933/log.txt.gz)

m0wlheld, 2023-10-30:
No success however with "sticky" OTA update ...

t34wrj, 2023-11-02:
> I am also seeing this with:
> 
> * RB 266, firmware build date 20211022, version 2.03.01
> * RB 245, firmware build date 20211022, version 2.03.01
> 
> Interestingly, it has once worked to upgrade an RB 245 to 2.04.01 (several months ago) but at some point, this stopped working.

The same happened to me with the RB266 - one of three bulbs did upgrade to 2.04.01 with the remaining two stuck on 2.04.01 despite numerous attempts to update the firmware on these each time a new version of Zigbee2mqtt is released.

Koenkk, 2023-11-03:
I've asked Innr for more support, will update here.

peter1rhodes, 2023-11-20:
This issue is still present, and affects the Innr RS227T (white spectrum GU10 bulb) devices (even after the recent 1.33 release which includes some sort of fix for those devices in the notes). Same firmware versions as noted above

Koenkk, 2023-11-22:
Does anyone have sniffer hardware and could make a sniff of the update process? I've received a sniff from Innr of a successful update, then I can compare those: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

m0wlheld, 2023-12-04:
Both changes do no affect the outcome. With upgradeTime: 5 update completes but with the next OTA step it is reversed. With upgradeTime: 1 and the proposed changes update fails at 99% with timeout. Logs attached.

[upgrade-time-1-and-code_log.txt.gz](https://github.com/Koenkk/zigbee2mqtt/files/13548937/upgrade-time-1-and-code_log.txt.gz)
[upgrade-time-5_log.txt.gz](https://github.com/Koenkk/zigbee2mqtt/files/13548940/upgrade-time-5_log.txt.gz)

m0wlheld, 2023-12-05:
> @m0wlheld is there any chance you could sniff the OTA update process? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

From a hardware perspective: Yes. I do have the CC debugger and a spare CC2531. I've setup the devices and the software as explained from the link (for Windows). I see alot of data running through the window, but it's all mystery to me.

I can follow your instructions but also happy to donate the required hardware (incl. an Innr bulb) to you.

Koenkk, 2023-12-05:
>  I see alot of data running through the window, but it's all mystery to me.

Can you send the sniff + your network key to me?

You can use [this form](https://form.jotform.com/233385867866072) to privately share a file with me.

m0wlheld, 2023-12-06:
Uploaded sniff + key. An unmodified Z2M 1.34.0 was used. The protocol starts immediatly before OTA was requested and ends after an error message at the end of OTA was displayed (see below). The device in OTA affected has network address 0x6291.

```
Update of '0x9c69370000cbc69d' failed (Upgrade end response failed (CommandResponse 0x9c69370000cbc69d/1 genOta.upgradeEndResponse({"manufacturerCode":4454,"imageType":300,"fileVersion":604050705,"imageSize":291130,"currentTime":0,"upgradeTime":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 25233 - 1 - 228 - 25 - 11 after 10000ms)))
```

Koenkk, 2023-12-06:
I see that commands from the bulb are missing, could you sniff close to the bulb?

DoerakkerHA, 2023-12-28:
I just updated one of the 2 failing INNR bulbs and it appears to stick 👍 "update": {
        "installed_version": 604050705,
        "latest_version": 604050705,
... currently running the second one as we speak:
{
  "message": "Update of '03 zijraam' at 48.04%, ≈ 17 minutes remaining",
  "meta": {
    "device": "03 zijraam",
    "progress": 48.04,
    "status": "update_progress"
  },
  "type": "ota_update"
}

resulting in:
        "installed_version": 604050705,
        "latest_version": 604050705,

finally a reboot of HA: succes !

DoerakkerHA, 2023-12-28:
resulting in:
"installed_version": 604050705,
"latest_version": 604050705,

finally a reboot of HA: succes !

jstaro, 2024-01-01:
This was not resolved for RB 245 and RB 266.

DoerakkerHA, 2024-01-02:
Correction: it was indeed not resolved. After another reboot of HA the items returned to be updated, ran updates it appears successful but it wasn't 

Sorry for causing any confusion

jstaro, 2024-01-09:
@Koenkk Can we re-open this thread since the issue persists? I can look into sniffing an OTA update close to a bulb as mentioned to @m0wlheld .

Koenkk, 2024-01-11:
@GerryInnr I just checked, the `imageSize` is ignored (I've removed it now), in the sniff from @m0wlheld the `Upgrade End Response` is valid. One interesting thing is that it's sent after the device restarts:

[image]

I'm also missing the `Upgrade End Request`, @jstaro it would be great if you could make a new sniff.

jstaro, 2024-01-12:
I didn't manage to do it with a spare Sonoff Dongle Plus-E I had laying around (the stick wouldn't start in Windows with the SiLabs drivers). I will give it another try w/ Linux, and I've also ordered a CC2531 stick just in case.

wolterink, 2024-01-20:
version firmware 2.04.02 seems to have fixed the issue for me.

t34wrj, 2024-01-20:
Same for me.

christopher-david-smith, 2024-01-20:
Updating to 2.04.02 this morning appears to have resolved the issue for myself too

m0wlheld, 2024-01-20:
Confirmed. Still trying to understand, how a firmware update can fix a broken firmware update process, but maybe the previous firmware image was just broken …

Koenkk, 2024-01-21:
That's awesome! I assume we can close this now.

closed 2024-01-21 by Koenkk