Koenkk/zigbee2mqtt #17375: Tuya TS0026 Scene Switch

evangelo18, 2023-04-17:
### What happened?

Switch is not working. Database entry as follows

{"id":10,"type":"Router","ieeeAddr":"0x8cf681fffeaf27d2","nwkAddr":63913,"manufId":4098,"manufName":"_TZ3000_zua10nyz","powerSource":"Mains (single phase)","modelId":"TS0026","epList":[1,2,3,4,5,6],"endpoints":{"1":{"profId":260,"epId":1,"devId":4,"inClusterList":[0,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0026","manufacturerName":"_TZ3000_zua10nyz","stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"tuyaBacklightMode":1,"moesStartUpOnOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":4,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":4,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":4,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":4,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":260,"epId":6,"devId":4,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1674377020154,"defaultSendRequestWhen":"immediate"}

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

 20220219

### Adapter

conbee

closed 2023-05-25 by github-actions[bot]

carlva, 2023-09-30:
> ### What happened?
> Switch is not working. Database entry as follows [...]

did you ever get a resolution to this?

johncoles, 2023-12-22:
Hello, I too have the same issue with TS0026.