Koenkk/zigbee2mqtt #17379: Cannot enable permit_join via bridge/request

glynhudson, 2023-04-18:
### What happened?

I'm trying to enable pairing via MQTT, when I publish:

Topic:  `zigbee2mqtt/bridge/request/permit_join` 
Message: `"{"value": true}"`

zigbeetomqtt responds with

Topic: `zigbee2mqtt/bridge/response/permit_join`
Message: `{"data":{"value":false},"status":"ok"}`

and permit_join is not enabled 

I have `permit_join: false` in `configuration.yaml`

### What did you expect to happen?

I expected zigbee2mqtt to respond with `"value":true` and enable permit joining. 

I can enable permit join and pair devices no problem via the UI, but I need to be able to use command line MQTT to admin remove instance. 

### How to reproduce it (minimal and precise)

`mosquitto_pub -h localhost t zigbee2mqtt/bridge/request/permit_join -m "{"value": true}"`

### Zigbee2MQTT version

1.30.2 commit: cdf62ea

### Adapter firmware version

20221226

### Adapter

sonoff zigbee 3.0

### Debug log

```
Debug 2023-04-18 02:23:10Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{value: true}'
Info 2023-04-18 02:23:10Zigbee: disabling joining new devices.
Info 2023-04-18 02:23:11MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":false},"status":"ok"}'
```

sjorge, 2023-04-18:
Try properly escaping the json payload

```bash
mosquitto_pub -h localhost -t zigbee2mqtt/bridge/request/permit_join -m '{"value": true}'
```

glynhudson, 2023-04-18:
Thanks! That did the trick 👍

closed 2023-04-18 by glynhudson