Koenkk/zigbee2mqtt #17381: [Problem] Wrong recognition of device as a "ZLGP15"

Pulsahr, 2023-04-18:
### What happened?

I'm currently owning 3 new type of devices that I want to pair with zigbee2MQTT :
- Legrand green power switch blind control ref 600087L [manufacturer doc here (french)](https://www.legrand.fr/pro/catalogue/commande-sans-fils-sans-pile-self-e-dooxie-with-netatmo-pour-volets-roulants-connectes-blanc) (french)
- Legrand green power scenario switch departure/arrival ref 600085L [manufacturer doc here (french)](https://www.legrand.fr/pro/catalogue/commande-generale-depart-arrivee-sans-fils-sans-pile-self-e-dooxie-with-netatmo-blanc)
- Legrand green power scenario switch day/night ref 600084L [manufacturer doc here (french)](https://www.legrand.fr/pro/catalogue/commande-levercoucher-sans-fils-sans-pile-self-e-dooxie-with-netatmo-blanc)

As they are not supported, I'm following the process for "support new devices".
I am currently working on the blind switch 600087L, with various problems I'm trying to solve (I made an issue [here](https://github.com/Koenkk/zigbee2mqtt/issues/17369)), but I wanted to try other switches.

### Problem with 600084L
I started today to work on the day/night switch 600084L: when attempting to pair it's immediatly recognized, but as the ZLGP15 ([manufacturer doc here](https://www.legrand.fr/pro/catalogue/commande-sans-fils-sans-pile-self-e-dooxie-with-netatmo-4-scenarios-blanc), real ref is 600091L), which is a 4 scenario green power switch. This switch has 4 commands with ID= 20;21;22;23, but the 600084L has command IDs= 18;19
Database value for this item:
```json
{
    "id": 8,
    "type": "GreenPower",
    "ieeeAddr": "0x000000000055b282",
    "nwkAddr": 45698,
    "manufId": null,
    "modelId": "GreenPower_254",
    "epList": [
        242
    ],
    "endpoints": {
        "242": {
            "epId": 242,
            "inClusterList": [],
            "outClusterList": [],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "interviewCompleted": true,
    "meta": {},
    "lastSeen": 1681811653956,
    "defaultSendRequestWhen": "immediate"
}
```

### Problem with 600085L
Identified as ZLGP15 too, emitted command IDS= 16;17
Database value for this item:
```json
{
    "id": 10,
    "type": "GreenPower",
    "ieeeAddr": "0x0000000000573940",
    "nwkAddr": 14656,
    "manufId": null,
    "modelId": "GreenPower_254",
    "epList": [
        242
    ],
    "endpoints": {
        "242": {
            "epId": 242,
            "inClusterList": [],
            "outClusterList": [],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "interviewCompleted": true,
    "meta": {},
    "lastSeen": 1681813233989,
[... 28 lines cut ...]

SONOFF ZigBee 3.0 USB Dongle Plus

### Debug log

```
Zigbee2MQTT:info  2023-04-18 11:53:35: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"f2t6k-1"}'
Zigbee2MQTT:info  2023-04-18 11:53:41: Device '0x000000000055b282' joined
Zigbee2MQTT:info  2023-04-18 11:53:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000000000055b282","ieee_address":"0x000000000055b282"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-04-18 11:53:41: MQTT publish: topic 'homeassistant/sensor/0x000000000055b282/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x000000000055b282"],"manufacturer":"Legrand","model":"Wireless and batteryless 4 scenes control (ZLGP15)","name":"0x000000000055b282"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"0x000000000055b282 action","state_topic":"zigbee2mqtt/0x000000000055b282","unique_id":"0x000000000055b282_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Zigbee2MQTT:info  2023-04-18 11:53:41: MQTT publish: topic 'homeassistant/sensor/0x000000000055b282/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x000000000055b282"],"manufacturer":"Legrand","model":"Wireless and batteryless 4 scenes control (ZLGP15)","name":"0x000000000055b282"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x000000000055b282 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x000000000055b282","unique_id":"0x000000000055b282_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2023-04-18 11:53:41: Successfully interviewed '0x000000000055b282', device has successfully been paired
Zigbee2MQTT:info  2023-04-18 11:53:41: Device '0x000000000055b282' is supported, identified as: Legrand Wireless and batteryless 4 scenes control (ZLGP15)
Zigbee2MQTT:info  2023-04-18 11:53:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Wireless and batteryless 4 scenes control","exposes":[{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["press_1","press_2","press_3","press_4"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZLGP15","options":[],"supports_ota":false,"vendor":"Legrand"},"friendly_name":"0x000000000055b282","ieee_address":"0x000000000055b282","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:debug 2023-04-18 11:53:41: Received Zigbee message from '0x000000000055b282', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"applicationInfo":4,"deviceID":254,"extendedOptions":242,"gpdClientClusters":{"data":[],"type":"Buffer"},"gpdCommandIdList":{"data":[18,19,26,27],"type":"Buffer"},"gpdServerClusters":{"data":[],"type":"Buffer"},"keyMic":4028049532,"manufacturerID":0,"modelID":0,"numClientClusters":0,"numGdpCommands":4,"numServerClusters":0,"options":197,"outgoingCounter":23,"securityKey":{"data":[254,34,90,17,112,248,165,132,242,25,254,192,70,168,168,3],"type":"Buffer"}},"commandID":224,"frameCounter":23,"gppGddLink":142,"gppNwkAddr":24389,"options":2048,"payloadSize":33,"srcID":5616258}' from endpoint 242 with groupID 0
Zigbee2MQTT:debug 2023-04-18 11:53:41: Received MQTT message on 'homeassistant/sensor/0x000000000055b282/action/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x000000000055b282"],"manufacturer":"Legrand","model":"Wireless and batteryless 4 scenes control (ZLGP15)","name":"0x000000000055b282"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"0x000000000055b282 action","state_topic":"zigbee2mqtt/0x000000000055b282","unique_id":"0x000000000055b282_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Zigbee2MQTT:debug 2023-04-18 11:53:41: Received MQTT message on 'homeassistant/sensor/0x000000000055b282/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x000000000055b282"],"manufacturer":"Legrand","model":"Wireless and batteryless 4 scenes control (ZLGP15)","name":"0x000000000055b282"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x000000000055b282 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x000000000055b282","unique_id":"0x000000000055b282_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:debug 2023-04-18 11:53:50: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-04-18 11:54:06: Received Zigbee message from '0x000000000055b282', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[69,95,142],"type":"Buffer"}},"commandID":18,"frameCounter":25,"gppGddLink":142,"gppNwkAddr":24389,"options":21664,"payloadSize":0,"srcID":5616258}' from endpoint 242 with groupID 2948
Zigbee2MQTT:error 2023-04-18 11:54:06: ZLGP15: missing command '18'
[... cut ...] pe":"Buffer"}},"commandID":19,"frameCounter":26,"gppGddLink":76,"gppNwkAddr":24389,"options":21664,"payloadSize":0,"srcID":5616258}' from endpoint 242 with groupID 2948
Zigbee2MQTT:error 2023-04-18 11:54:13: ZLGP15: missing command '19'
Zigbee2MQTT:debug 2023-04-18 11:54:44: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"f2t6k-2","value":false}'
Zigbee2MQTT:info  2023-04-18 11:54:44: Zigbee: disabling joining new devices.
```

luchendo, 2023-04-28:
Hi. I am owning also the "Legrand green power scenario switch departure/arrival ref 600085L" and share the same observation that there is either an incorrect identification to the 4 scene buttons, or a need to integrate all commands in the current integration. 

I agree that command ID missing are 16 and 17. 
If you have any advice for a manual fix until an official comes, I would be happy to learn :-). 
Thanks for your work.

Pulsahr, 2023-04-30:
I plan to submit a PR when I have time to do it. Basically it's just a few things to do:
- replace the current device image by an image with 3 switches (4 button, departure/arrival, day/nigh)
- - edit [zigbee-herdsman-converters/converters/fromZigbee.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js) to rename function "legrand_zlgp15" (what is the choice of naming ? why not "legrand_gp254" ?) and add commands 16 to 19 into const lookup
  - 16=home_arrival, 17=home_departure, 18=daytime_day, 19=daytime_night (I just checked) (do not alter the existing ones)
- edit [zigbee-herdsman-converters/devices/legrand.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/legrand.js) "GreenPower_254" (currently line 387) to udpate description (something like "Wireless and batteryless scenario switches"), call the new name of the function, and add the new commands in `exposes` property (do not alter existing ones).

I just don't have time currently, and I don't know when I will.

Pulsahr, 2023-05-13:
Made a Pull Request: https://github.com/Koenkk/zigbee-herdsman-converters/pull/5745