Koenkk/zigbee2mqtt #17383: [Feature request]: Better handling of ephemeral states for eg "action"

nvx, 2023-04-18:
### Is your feature request related to a problem? Please describe

Some ZigBee devices such as https://www.zigbee2mqtt.io/devices/66492-001.html include information as to what caused the state to change (eg, key unlock vs manual thumbturn vs zigbee control).

Unfortunately, the way this information comes in appears to be ephemeral, and not at the same time as the lock state changing, so this makes it difficult to use this information in a Home Assistant automation.

For example, when manually unlocking the thumb turn, these are the MQTT messages produced:

```
zigbee2mqtt/deadbolt: {"action":"manual_unlock","action_source":2,"action_source_name":"manual","action_user":65535,"battery":10,"last_seen":"2023-04-18T12:42:46.692Z","linkquality":153,"lock_state":"locked","state":"LOCK"}
zigbee2mqtt/deadbolt: {"action":"","action_source_name":null,"action_user":null,"battery":10,"last_seen":"2023-04-18T12:42:46.692Z","linkquality":153,"lock_state":"locked","state":"LOCK"}
zigbee2mqtt/deadbolt/action: manual_unlock
zigbee2mqtt/deadbolt: {"action":null,"action_source_name":null,"action_user":null,"battery":10,"last_seen":"2023-04-18T12:42:46.699Z","linkquality":142,"lock_state":"unlocked","state":"UNLOCK"}
```

Only the very last message has the state change from LOCK to UNLOCK, and by that stage the action and action source information has been lost and isn't visible to an automation reacting on the state change. I've confirmed this is the case by looking at the trace view of an automation that fires on the lock state change.

### Describe the solution you'd like

I can see two ways of solving this, one would be to cache the last seem values and always expose them, eg a "last_action" field that would contain the last non-null/empty value seen for the action field, ditto for the rest.

The other option I thought of (which may or may not make sense, as I'm unsure how these devices work at the zigbee layer or internally in zigbee2mqtt that may dictate certain behaviours) would be to send the event data all at once, instead of having a mqtt message that has the updated action but not the updated state rather hold back the message and send just one that has all of the information together.

### Describe alternatives you've considered

It may be possible to react to the changing attribute value instead, but then you'd need to figure out which action is going to result in which state which may not always be possible, and would certainly make the HA automation rather complex.

### Additional context

zigbee2mqtt 1.30.3

nvx, 2023-05-23:
Still relevant...

nvx, 2023-06-25:
ping

sjorge, 2023-06-25:
Setting debounce to 1 sec would probably group all of them together. You can even set it per-device

nvx, 2023-07-03:
> Setting debounce to 1 sec would probably group all of them together. You can even set it per-device

Nice idea, this seems to work well. Is it worth adding this to the docs for this device (https://www.zigbee2mqtt.io/devices/66492-001.html)?

nvx, 2023-08-03:
Still relevant, but perhaps updating docs is enough to consider this closed?

closed 2023-09-10 by github-actions[bot]