Koenkk/zigbee2mqtt #17384: [New device support]: Namron Stove Guard

GD-Dal, 2023-04-18:
### Link

https://www.namron.com/products/namron-zigbee-komfyrvakt-med-veggsensor-hvit/

### Database entry

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024c94537","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"}

### Comments

The zigbee functionality in this the device is described at page 9 in this [link](https://www.elektroimportoren.no/docs/lib/1402793-Brukerveiledning-5.pdf)

I have started to create a converter, but hit the wall

In Z2M, it looks like this:
[image]

The temperature varies. Often down to -300 C.
The link quality does seem way to low.
And the logs say the battery reports status every 5 seconds, which I would think is totally unnecessary.

Help is appreciated 

### External converter

```shell
const definition = {
    zigbeeModel: ['1402790', '1402791', '1402792', '1402793', '1402794', '1402795', '1402796', '1402797', '1402798'],
    model: '140279X',
    vendor: 'Namron',
    description: 'Namron Stove Guard',
    fromZigbee: [fz.on_off, fz.battery, fz.temperature, fz.metering, fz.electrical_measurement],
    toZigbee: [tz.on_off],
    exposes: [e.temperature(), e.battery(), e.switch(), e.power(), e.energy()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1) || device.getEndpoint(10);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'genOnOff', 'haElectricalMeasurement', 'genPowerCfg', 'msTemperatureMeasurement']);
            await endpoint.read('haElectricalMeasurement', ['acVoltageMultiplier', 'acVoltageDivisor']);
            await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
            await endpoint.read('haElectricalMeasurement', ['acCurrentMultiplier', 'acCurrentDivisor']);
            await reporting.onOff(endpoint);
            await reporting.temperature(endpoint);
            await reporting.activePower(endpoint);
	    await reporting.batteryPercentageRemaining(endpoint);
        },
};

module.exports = definition;
```

closed 2023-05-26 by github-actions[bot]

GD-Dal, 2023-05-27:
Not possible to repoen?

damtjern, 2023-06-07:
Any updates on support?

GD-Dal, 2023-07-04:
2 releases has come and gone, but still no support for this /sadface
How to reopen a device support request like this?

Thanks

PetterVikenes, 2023-08-07:
Have been in contact with Brunel Holding (the company that have the design/certificates) they distribute under the brand Friefence.org. But they also started license them to Namron, who I believe sells the most of them ([elektroinportøren](https://www.elektroimportoren.no/))

I was told there is two versions of firmware, one where Brun Holding is Manufacturer and one where Namron is.
My understanding was that Manufacturer name was the only difference (Namron only receives hex).
But the code base and practises of Namron would be different.

BTW due to "stove safety regulations" in Norway: "You need to visually inspect the stove before resetting,"
Makes kinda sens... 
To apply to regulations, you can only trip the device (relay off) But seems like relay ON cmd  won't respond at all, and throw an error. ON/OFF state works though
Not sure how to go about "ONLY Trip" in Zigbee

The Wattage part works in the converter above (my main gold), so knowing about the ON/OFF kind of works.
i'l see if i dig any deeper, but...

here's the Brun Holding Zigbee Doc
[FireFence_Zigbee_description_rev1.2.pdf](https://github.com/Koenkk/zigbee2mqtt/files/12283044/FireFence_Zigbee_description_rev1.2.pdf)

GD-Dal, 2023-09-11:
Another update came and went, but still no support.
Maybe because the case is closed and therefore not noticed?

Is it possible to repoen cases, or do I have to create a new one?

dreamcatcher2075, 2023-10-09:
I am also watching this with interest as I own two of these devices.

CyberSecDog, 2024-01-08:
I wanted to get this to work this weekend.

This is what i have manage to create.
This works ok for me. But Its to problems: 

- [ ] When the Namron Device do not know the temperature it sends a value that show up at -327.68° (If i hold the button on the sensor for 5 sec and put it into test mode and increases the temp on the stove above 150 so the stove turns off the temp is ok)
- [ ] The power stat can only be used to turn off and not on. You need to manualy tun it on at the device (by design) How can i show this in the config?

This is my Namron.js file:
```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['1402790', '1402791', '1402792', '1402793', '1402794', '1402795', '1402796', '1402797', '1402798'],
    model: '1402790',
    vendor: 'Namron',
    description: 'Namron Stove Guard',
    fromZigbee: [fz.on_off, fz.battery, fz.temperature, fz.metering, fz.electrical_measurement],
    toZigbee: [tz.on_off],
    exposes: [e.temperature(), e.battery(), e.switch(), e.power(), e.energy()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1) || device.getEndpoint(10);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'genOnOff', 'haElectricalMeasurement', 'seMetering', 'genPowerCfg', 'msTemperatureMeasurement']);
        await reporting.onOff(endpoint);
        await reporting.temperature(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.activePower(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.currentSummDelivered(endpoint);
    },
};

module.exports = definition;
```
[image]

damtjern, 2024-01-30:
@CyberSecDog Dig you have manually set reporting interval?
I have to press read data to make it update, but the data seems to work mostly, exept power and energy.
However i used the Generate external config from the latest zigbee2mqtt and not yours

CyberSecDog, 2024-01-31:
@damtjern  the config from  @GD-Dal try to get the power and energy from a old method, haElectricalMeasurement. This might be sorting that worked on old firmware. But my and new stove guards use seMetering and is the newer standard. 

If you use my config you get the power and Energy if you use a new firmware. I have a updated config on my system theta translate the temp to unknow if its not set that I'm trying out. But i whit the config in my post you can use it to monitor energy use in the energy dashboard.

CyberSecDog, 2024-01-31:
The temp only work if you do the step to synk sensor and base whit a "test mode" to sync.

CyberSecDog, 2024-01-31:
@CyberSecDog when you use my config whit seMetering  you auto get newer updates intervals automaticly  do the default of the  atthrebutes whit the firs post i only goth 2 .. whit my setup auto ghet this :
[image]

ChristianPB, 2024-06-24:
I have used it with zigbee2mqtt for a month. I worked perfekt from 25.may to 22.june. Then suddenly the stovegard was lost... This has happened before. Also when the stovegard was only 3m from the antenna. (Sonoff)

GD-Dal, 2024-11-14:
Yet another release passed, but no support for this.
Asking like someone that know how to add devices, how hard is it to add a device like this if you know the zigbee profile like provide in this [document](https://www.elektroimportoren.no/docs/lib/1402793-Brukerveiledning-5.pdf)?

Can someone please have mercy and add support for this?