Koenkk/zigbee2mqtt #17385: IKEA E1746 repeater won't pair with

befi-ibm, 2023-04-18:
### What happened?

My newly bought IKEA repeater will not pair with my SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N) Zigbee USB-Stick

### What did you expect to happen?

I expected the repeater to be paired

### How to reproduce it (minimal and precise)

press reset button on the ikea repaeter for 5+ seconds and wait
IKEA device starts blinking
the messages in the log appear (see below)
the device appears in the zigbee2MQTT list as "not supported"

### Zigbee2MQTT version

1.28.2-dev commit: a2f85348

### Adapter firmware version

 20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N) Zigbee USB-Stick

### Debug log

Debug:
Apr 18 12:22:21 server npm[20017]: Zigbee2MQTT:info  2023-04-18 12:22:21: Starting interview of 'repeater-01'
Apr 18 12:22:21 server npm[20017]: Zigbee2MQTT:info  2023-04-18 12:22:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"repeater-01","ieee_address":"0xb4e3f9fffefd9bbf","status":"started"},"type":"device_interview"}'
Apr 18 12:22:21 server npm[20017]: Zigbee2MQTT:info  2023-04-18 12:22:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"repeater-01"},"type":"pairing"}'

Wikibear, 2023-04-19:
Please update to the latest version: https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.30.3 and try again.

TheLordVader, 2023-04-29:
I have the same issue, running 1.30.3

closed 2023-07-07 by github-actions[bot]

Wurli23, 2024-06-08:
I am using 1.38.0 and I have the same problem. New IKEA E 1746 won't pair