Koenkk/zigbee2mqtt #17387: _TZ3000_dlug3kbc Should be ZIGBEE-B09-UK

joesoon, 2023-04-18:
### Link

https://shopee.com.my/Tuya-Smart-Wall-Socket-Combo-QC3.0-USB-Charging-Port-WiFi-ZigBee-works-with-Smart-Life-Google-Home-Alexa-Home-Assistant2-i.272772351.12002744288

### Database entry

0xa4c1387cf8a088c2

### Comments

After pairing, it shows up as TS011F_plug_1 rather than Zemismart ZIGBEE-B09-UK (The model I purchased looked exactly like it)

Zigbee Model
TS011F
Zigbee Manufacturer
_TZ3000_dlug3kbc

Koenkk, 2023-04-19:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-04-19 by Koenkk

joesoon, 2023-04-20:
Thanks for your quick reponse! I have just updated the hassio version of edgr branch and the problem persists, do you have any tips to rectify that?

Koenkk, 2023-04-20:
What zigbee2mqtt version is listed under z2m frontend -> settings -> about?

joesoon, 2023-04-21:
Hey, thanks again for your response.
Version :
1.30.3-dev commit: 17947aa

Koenkk, 2023-04-21:
I forgot to update it, when you update to the latest edge it should work (uninstall -> install the addon)

joesoon, 2023-04-21:
Just did, it now recognizes correctly, thanks! however there are two states exposed, state_l1 and state_l2, with each referring to the 2 USB ports on the device, there's no way to control the AC outlet, am I missing some configurations?

Koenkk, 2023-04-22:
Can you provide the `data/database.db` entry of this device?

joesoon, 2023-04-22:
Sure thing. Here it is

```
{"id":24,"type":"Router","ieeeAddr":"0xa4c1387cf8a088c2","nwkAddr":30332,"manufId":4417,"manufName":"_TZ3000_dlug3kbc","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"6��+\u00136��+\u0012","65506":54,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_dlug3kbc","zclVersion":3,"appVersion":192,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":0,"activePower":0,"rmsCurrent":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0257","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0257","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0257","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0257","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"3":{"profId":260,"epId":3,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1948765142},"lastSeen":1682145087320,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2023-04-22:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition  = {
    fingerprint: tuya.fingerprint('TS011F', ['_TZ3000_zigisuyh', '_TZ3000_v4mevirn', '_TZ3000_mlswgkc3', '_TZ3000_dlug3kbc']),
    model: 'ZIGBEE-B09-UK',
    vendor: 'Zemismart',
    description: 'Zigbee smart outlet universal socket with USB port',
    extend: tuya.extend.switch({powerOutageMemory: true, endpoints: ['l1', 'l2', 'l3']}),
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2, 'l3': 3};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(device.getEndpoint(1));
        await reporting.onOff(device.getEndpoint(2));
        await reporting.onOff(device.getEndpoint(3));
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

joesoon, 2023-04-22:
Hey, that did it, you're fantastic. L1 and L2 are the USB and L3 is the AC outlet. Cheers.

Koenkk, 2023-04-22:
Great, added, you can keep using the external converter until updating to the next release.