Koenkk/zigbee2mqtt #17388: [New device as external]: Schneider Electric PowerLogic EM4302

jonathandreyer, 2023-04-18:
### What happened?

Hello everyone,

I have a few Schneider Electric EM4302 and I would like to use it without the Schneider Gateway. Before starting with Zigbee2MQTT, I have already sniffer a complete pairing-reporting-leave process.

In the documentation (available [here](https://download.schneider-electric.com/files?p_Doc_Ref=7EN02-0356&p_enDocType=User+guide&p_File_Name=7EN02-0356-02.pdf)), it is indicated that the device joined only coordinators with a specific EPID. So I have sniffed the MAC address of Schneider dongle, and get the PanID & EPID from the Gateway frontend (to clone the dongle).

Actually, my setup is composed of a RaspberryPI with a Sonoff USB Dongle-P (flashed with the firmware indicated in the documentation). I also have a VM for running Wireshark with another Sonoff dongle (also flashed with the correct firmware).

I have started with a minimal external converter ([here](https://gist.github.com/jonathandreyer/19a8f3e8f8990973f22984c0ec5c907e/f73767b8346200d68517f862dbedff99cdd13010)), but that not work properly. In Zigbee2MQTT, the device is "successfully been paired", but the LED to the device indicated that the device wait to bind with the gateway. So I have taken a look of others implemented devices and the initial capture to try to understand the communication. I have tested a new version of the external converter ([here](https://gist.github.com/jonathandreyer/19a8f3e8f8990973f22984c0ec5c907e/51bcd230669f46ab2e5df753b488ae56dd496055)), to add multiple bind requests as that is in the record of the initial capture. But now, there is an error in the log:
```
zigbee2mqtt         | Zigbee2MQTT:error 2023-04-18 23:33:30: Failed to configure '0x00124b00xxxxxxxx', attempt 1 (TypeError: Cannot read properties of undefined (reading 'bind')
zigbee2mqtt         |     at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24)
zigbee2mqtt         |     at Object.configure (/app/data/extension/externally-loaded.js:19:25)
zigbee2mqtt         |     at Configure.configure (/app/lib/extension/configure.ts:117:37)
zigbee2mqtt         |     at EventEmitter.<anonymous> (/app/lib/extension/configure.ts:81:62)
zigbee2mqtt         |     at EventEmitter.emit (node:events:525:35)
zigbee2mqtt         |     at EventBus.emitDeviceInterview (/app/lib/eventBus.ts:74:22)
zigbee2mqtt         |     at Controller.<anonymous> (/app/lib/zigbee.ts:86:27)
zigbee2mqtt         |     at Controller.emit (node:events:513:28)
zigbee2mqtt         |     at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
zigbee2mqtt         |     at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:570:22
```

Do you have any idea what the problem is?

In the log, I have few information regarding definitions that are not defined, but that seems to me OK, because that it is not yet implemented. 

Regarding the fields "fromZigbee, toZigbee & exposes", I prefer to leave them empty and go back to these once the communication is OK.

### Zigbee2MQTT version

1.30.2 (#24c6b2e)

### Adapter firmware version

20221226

### Adapter

Sonoff ZBDongle-P

### Debug log

zigbee2mqtt         | Zigbee2MQTT:info  2023-04-18 23:32:02: Zigbee2MQTT started!
zigbee2mqtt         | Zigbee2MQTT:info  2023-04-18 23:32:57: Device '0x00124b00xxxxxxxx' joined
zigbee2mqtt         | Zigbee2MQTT:info  2023-04-18 23:32:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00xxxxxxxx","ieee_address":"0x00124b00xxxxxxxx"},"type":"device_joined"}'
zigbee2mqtt         | Zigbee2MQTT:info  2023-04-18 23:32:57: Starting interview of '0x00124b00xxxxxxxx'
zigbee2mqtt         | Zigbee2MQTT:info  2023-04-18 23:32:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00xxxxxxxx","ieee_address":"0x00124b00xxxxxxxx","status":"started"},"type":"device_interview"}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:32:57: Device '0x00124b00xxxxxxxx' announced itself
zigbee2mqtt         | Zigbee2MQTT:info  2023-04-18 23:32:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00xxxxxxxx","ieee_address":"0x00124b00xxxxxxxx"},"type":"device_announce"}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:28: Received Zigbee message from '0x00124b00xxxxxxxx', type 'readResponse', cluster 'genBasic', data '{"modelId":"EM4300"}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:28: Skipping message, definition is undefined and still interviewing
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Schneider Electric"}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Skipping message, definition is undefined and still interviewing
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Skipping message, definition is undefined and still interviewing
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Skipping message, definition is undefined and still interviewing
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Skipping message, definition is undefined and still interviewing
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Skipping message, definition is undefined and still interviewing
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:29: Skipping message, definition is undefined and still interviewing
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:30: Received Zigbee message from '0x00124b00xxxxxxxx', type 'readResponse', cluster 'genBasic', data '{"dateCode":"\u0000"}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:30: Skipping message, definition is undefined and still interviewing
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:30: Received Zigbee message from '0x00124b00xxxxxxxx', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-18 23:33:30: Skipping message, definition is undefined and still interviewing
zigbee2mqtt         | Zigbee2MQTT:info  2023-04-18 23:33:30: Successfully interviewed '0x00124b00xxxxxxxx', device has successfully been paired
zigbee2mqtt         | Zigbee2MQTT:info  2023-04-18 23:33:30: Device '0x00124b00xxxxxxxx' is supported, identified as: Schneider Electric PowerLogic Wireless Energy Meter (EM4300)
zigbee2mqtt         | Zigbee2MQTT:info  2023-04-18 23:33:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"PowerLogic Wireless Energy Meter","exposes":[{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"EM4300","options":[],"supports_ota":false,"vendor":"Schneider Electric"},"friendly_name":"0x00124b00xxxxxxxx","ieee_address":"0x00124b00xxxxxxxx","status":"successful","supported":true},"type":"device_interview"}'
zigbee2mqtt         | Zigbee2MQTT:info  2023-04-18 23:33:30: Configuring '0x00124b00xxxxxxxx'
[... cut ...]
zigbee2mqtt         |     at EventEmitter.emit (node:events:525:35)
zigbee2mqtt         |     at EventBus.emitDeviceInterview (/app/lib/eventBus.ts:74:22)
zigbee2mqtt         |     at Controller.<anonymous> (/app/lib/zigbee.ts:86:27)
zigbee2mqtt         |     at Controller.emit (node:events:513:28)
zigbee2mqtt         |     at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
zigbee2mqtt         |     at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:570:22))

Koenkk, 2023-04-19:
It seems your device has no endpoint 1, in the data/database.db entry of your device you can find which endpoints are available.

jonathandreyer, 2023-04-19:
@Koenkk Thanks you for your feedback. I will study more in detail the initial capture and find the correct endpoint value for each cluster.

jonathandreyer, 2023-04-20:
@Koenkk Thank you for your support. I have updated the endpoint to 10 and now it is OK. I have updated the [file](https://gist.github.com/jonathandreyer/19a8f3e8f8990973f22984c0ec5c907e/0f73cd94b4a58d9c77cf4e73dcf4adfdff722161) with the correct value and profiles.

I have two questions regarding the decoding of the received data:
1. I have already studied multiples existing converters in the *zigbee-herdsman-converters*, but what is the difference between *exposes* (like [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/2861a72762fab96952a9077bb1551770a53b3200/devices/schneider_electric.js#L463)) and *reporting* (in *configure*, like [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/2861a72762fab96952a9077bb1551770a53b3200/devices/schneider_electric.js#L466))?
2. I have also seen that the backend received data in buffer type. Are there any examples with this data type?

Best regards,

Jonathan Dreyer

Koenkk, 2023-04-21:
1. Exposes defines what is discoverd to e.g. Home Assistant and what is shown under the z2m frontend -> device -> exposes tab, configuring reporting makes the device report a attribute (e.g. on/off state)
2. Can you give an example of this?

jonathandreyer, 2023-04-26:
@Koenkk, Thanks you for your reply and sorry for my delay. For my side, I need to get the information received by the concentrator to the MQTT, but I do not need it to be integrated in Home Automation. So it is not necessary to focus on this point at this time.
In a second time when the decoding of all received data is OK, we can see how to integrate with the front end.

Thank you for the clarification regarding the exposes field. But another question comes to mind, what is the difference between *fromZigbee* & e.g. *reporting.activePower(…)* (like [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/2861a72762fab96952a9077bb1551770a53b3200/devices/schneider_electric.js#L472))? I understood that fromZigbee was dedicated to "registered" converters, but is it correct?
To be back to the decoding received data, should I focus on *fromZigbee* or *reporting*?

For the second point, I need a bit more time to test again the device and sniffing the received data, probably this weekend. I would like to provide a valid transmission received and the corresponding capture.

Best regards,

Jonathan Dreyer

Koenkk, 2023-04-27:
- reporting makes the device send Zigbee messages to Zigbee2MQTT
- even that the device now sends a certain Zigbee messages to Zigbee2MQTT, as long as `fromZigbee` is empty, nothing will be published to MQTT. What a fromZigbee converter does is parse the Zigbee messages to a more human friendly messages which will be published to MQTT.

jonathandreyer, 2023-04-28:
@Koenkk Thanks you for your reply. Now I'm getting a better understanding. As there is no register modification in the capture (with the original concentrator), I suppose there is no need to configure anything.

Regarding the buffer type, here is an example of communication:
```
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:28: Received Zigbee message from '0x00124b00xxxxxxxx', type 'attributeReport', cluster 'genBasic', data '{"57856":10}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:28: Received Zigbee message from '0x00124b00xxxxxxxx', type 'attributeReport', cluster 'seMetering', data '{"readingSnapshotTime":3674}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:28: No converter available for 'EM4300' with cluster 'seMetering' and type 'attributeReport' and data '{"readingSnapshotTime":3674}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'raw', cluster 'seMetering', data '{"data":[28,94,16,3,10,16,64,45,51,166,238,0,0,0,17,64,45,26,219,191,255,255,255,18,64,45,195,110,69,1,0,0,20,64,45,51,166,238,0,0,0,21,64,45,26,219,191,255,255,255,22,64,45,195,110,69,1,0,0,0,68,34,1,0,0],"type":"Buffer"}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:29: No converter available for 'EM4300' with cluster 'seMetering' and type 'raw' and data '{"data":[28,94,16,3,10,16,64,45,51,166,238,0,0,0,17,64,45,26,219,191,255,255,255,18,64,45,195,110,69,1,0,0,20,64,45,51,166,238,0,0,0,21,64,45,26,219,191,255,255,255,22,64,45,195,110,69,1,0,0,0,68,34,1,0,0],"type":"Buffer"}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'raw', cluster 'seMetering', data '{"data":[28,94,16,4,10,0,65,45,242,185,33,0,0,0,1,65,45,238,241,252,255,255,255,2,65,45,173,93,49,0,0,0,3,65,45,242,185,33,0,0,0,4,65,45,238,241,252,255,255,255,5,65,45,173,93,49,0,0,0,1,68,34,232,3,0],"type":"Buffer"}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:29: No converter available for 'EM4300' with cluster 'seMetering' and type 'raw' and data '{"data":[28,94,16,4,10,0,65,45,242,185,33,0,0,0,1,65,45,238,241,252,255,255,255,2,65,45,173,93,49,0,0,0,3,65,45,242,185,33,0,0,0,4,65,45,238,241,252,255,255,255,5,65,45,173,93,49,0,0,0,1,68,34,232,3,0],"type":"Buffer"}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'raw', cluster 'seMetering', data '{"data":[28,94,16,5,10,0,66,45,155,121,199,0,0,0,1,66,45,244,87,190,255,255,255,2,66,45,183,90,237,0,0,0,3,66,45,155,121,199,0,0,0,4,66,45,244,87,190,255,255,255,5,66,45,183,90,237,0,0,0,2,68,34,1,0,0],"type":"Buffer"}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:29: No converter available for 'EM4300' with cluster 'seMetering' and type 'raw' and data '{"data":[28,94,16,5,10,0,66,45,155,121,199,0,0,0,1,66,45,244,87,190,255,255,255,2,66,45,183,90,237,0,0,0,3,66,45,155,121,199,0,0,0,4,66,45,244,87,190,255,255,255,5,66,45,183,90,237,0,0,0,2,68,34,1,0,0],"type":"Buffer"}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'raw', cluster 'seMetering', data '{"data":[28,94,16,6,10,0,67,45,166,114,5,0,0,0,1,67,45,56,145,4,0,0,0,2,67,45,95,182,38,0,0,0,3,67,45,166,114,5,0,0,0,4,67,45,56,145,4,0,0,0,5,67,45,95,182,38,0,0,0,3,68,34,232,3,0],"type":"Buffer"}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:29: No converter available for 'EM4300' with cluster 'seMetering' and type 'raw' and data '{"data":[28,94,16,6,10,0,67,45,166,114,5,0,0,0,1,67,45,56,145,4,0,0,0,2,67,45,95,182,38,0,0,0,3,67,45,166,114,5,0,0,0,4,67,45,56,145,4,0,0,0,5,67,45,95,182,38,0,0,0,3,68,34,232,3,0],"type":"Buffer"}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:29: Received Zigbee message from '0x00124b00xxxxxxxx', type 'attributeReport', cluster 'seMetering', data '{"17412":1,"17413":1000,"17665":0,"17920":0}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:29: No converter available for 'EM4300' with cluster 'seMetering' and type 'attributeReport' and data '{"17412":1,"17413":1000,"17665":0,"17920":0}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:30: Received Zigbee message from '0x00124b00xxxxxxxx', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"acFrequency":500,"acFrequencyDivisor":10,"acFrequencyMultiplier":1,"activePower":0,"powerDivisor":1000,"powerMultiplier":21,"rmsVoltage":237,"totalActivePower":2,"totalApparentPower":218,"totalReactivePower":-211}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:30: No converter available for 'EM4300' with cluster 'haElectricalMeasurement' and type 'attributeReport' and data '{"acFrequency":500,"acFrequencyDivisor":10,"acFrequencyMultiplier":1,"activePower":0,"powerDivisor":1000,"powerMultiplier":21,"rmsVoltage":237,"totalActivePower":2,"totalApparentPower":218,"totalReactivePower":-211}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 01:34:30: Received Zigbee message from '0x00124b00xxxxxxxx', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"acCurrentDivisor":1,"acCurrentMultiplier":1,"acPowerDivisor":1,"acPowerMultiplier":21,"acVoltageD [... cut ...]
The full [log](https://github.com/Koenkk/zigbee2mqtt/files/11354864/Logs.txt) can be seen in attachment

It is interesting that there is no registry value/information like for the cluster "haElectricalMeasurement". The [list of reported attributes](https://github.com/Koenkk/zigbee2mqtt/files/11354869/EM4300_attributes-reported-typeBuffer.pdf) can also be seen in attachment. I have got it from a capture with the original concentrator, and validated with the datasheet registers information. I have indicated which registers are in buffer type.

Best regards,

Jonathan Dreyer

Koenkk, 2023-04-28:
`raw` means the message cannot be parsed, probably the message does not comply with the Zigbee spec.

Can you sniff the traffic of this device when it reports such messages to z2m?

jonathandreyer, 2023-04-28:
@Koenkk Of course, yes. I have captured a new one transmission. I have selected the first problematic frame. The log in the Z2M is:
```
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 21:17:02: Received Zigbee message from '0x00124b0005a60f8c', type 'raw', cluster 'seMetering', data '{"data":[28,94,16,3,10,16,64,45,212,165,238,0,0,0,17,64,45,243,217,191,255,255,255,18,64,45,78,112,69,1,0,0,20,64,45,212,165,238,0,0,0,21,64,45,243,217,191,255,255,255,22,64,45,78,112,69,1,0,0,0,68,34,1,0,0],"type":"Buffer"}' from endpoint 10 with groupID 0
zigbee2mqtt         | Zigbee2MQTT:debug 2023-04-28 21:17:02: No converter available for 'EM4300' with cluster 'seMetering' and type 'raw' and data '{"data":[28,94,16,3,10,16,64,45,212,165,238,0,0,0,17,64,45,243,217,191,255,255,255,18,64,45,78,112,69,1,0,0,20,64,45,212,165,238,0,0,0,21,64,45,243,217,191,255,255,255,22,64,45,78,112,69,1,0,0,0,68,34,1,0,0],"type":"Buffer"}'
```

The sniffed frame in Wireshark gives:
[image]

The raw data seems to be received correctly, but it is correct that Wireshark does not insert attribute information as it does for the OK data.

PS: I also captured all the transmissions, but I'm not sure if this is necessary at this point. Don't hesitate to ask me if you need more frames.

Best regards,

Jonathan Dreyer

Koenkk, 2023-04-29:
> PS2: I have seen [here](https://github.com/Koenkk/zigbee-herdsman/blob/23597b8e18575aa1ea12566c6ee41ba2413f2ec8/src/zcl/definition/cluster.ts#L3448) that all registers that are causing problems are not listed. If necessary, I can create the necessary lines to add.

Please do so, I believe after that the message is not `raw` anymore.

jonathandreyer, 2023-04-29:
@Koenkk Thanks you for your fast support. I have added the registers that seem to cause problems in a new [branch](https://github.com/jonathandreyer/zigbee-herdsman/tree/dev-schneider-powerlogic). I also saw in the log that other frames have numbers and not register names. In a second step I might include them, in order to match the received frames properly.

While searching available types to fit with the datasheet, I found [this message](https://github.com/Koenkk/zigbee-herdsman/blob/23597b8e18575aa1ea12566c6ee41ba2413f2ec8/src/zcl/definition/dataType.ts#L55) that indicated the ```int48``` type are `not in BuffaloZcl and thus cannot be parsed/written`. Maybe it is the reason why Z2M aren't able to parse the frame, because the ```raw``` message is only there when the ```int48``` type is in the message. What do you think?

Last question, there is an option to use this modification in a docker container (without installing all sources & dependency)? If necessary, I can build docker images with the appropriate dependencies, but I do not have good Node.js skills.

Best regards,

Jonathan Dreyer

Koenkk, 2023-04-30:
I've added support for `int48`.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

jonathandreyer, 2023-04-30:
@Koenkk Thanks to you for the fast modification. Now it is OK, I have no more raw data into received frames. The modified cluster `seMetering` works well, but I have a few attributes that are not correctly interpreted for clusters `haElectricalMeasurement ` and `haDiagnostic `. I will try to fix that and will be back.

Best regards,

Jonathan Dreyer

jonathandreyer, 2023-04-30:
@Koenkk I have added the missing attributes and checked that they are all properly processed. The changes are on the [dev-schneider-powerlogic](https://github.com/jonathandreyer/zigbee-herdsman/compare/dev-schneider-powerlogic) branch. Now I can start working on the `fromZigbee` converter.

I have another general question, is it possible to write to an attribute (e.g. to change the reporting frequency) without implementing that in the  toZigbee converter? I would like the reduce the time between each reporting, to control what is the maximal reporting speed. The default frequency is 1min, but I would to the reduce to 5-10sec.

Best regards,

Jonathan Dreyer

Koenkk, 2023-04-30:
> I have added the missing attributes and checked that they are all properly processed. The changes are on the [dev-schneider-powerlogic](https://github.com/jonathandreyer/zigbee-herdsman/compare/dev-schneider-powerlogic) branch. Now I can start working on the fromZigbee converter.

Great, please make a PR

> I have another general question, is it possible to write to an attribute (e.g. to change the reporting frequency) without implementing that in the toZigbee converter (e.g. by the frontend UI)? I would like the reduce the time between each reporting, to control what is the maximal reporting speed. The default frequency is 1min, but I would to the reduce to 5-10sec.

Reporting frequency has nothing to do with the toZigbee converter. You can change the reporting frequency in the `configure`, e.g. https://github.com/Koenkk/zigbee-herdsman-converters/blob/c0b0bcc59e0066aaee7b807b651f30d5d1fea681/devices/shinasystem.js#L271

jonathandreyer, 2023-04-30:
@Koenkk Ok, I have created the [PR#703](https://github.com/Koenkk/zigbee-herdsman/pull/703).

Regarding the reporting frequency, my explanation was a little oversimplified. In the whole device attributes, there are two attributes that can change the behavioral of the device. The `Energy Counters Reporting Period` (attribute 0x4600 - cluster `seMetering`) changes the frequency at which the values are reported by the device. The `Demande Interval Duration` (attribute 0x4319 - cluster `haElectricalMeasurement`) changes the interval at which some values are calculated (min. voltages & max. currents) by the device.
These attributes are available in the document attached above in the conversation. Here is a capture with both values:
[image]

Best regards,

Jonathan Dreyer

jonathandreyer, 2023-05-02:
Hello @Koenkk, I have made some progress to convert all received values. The external converter has been [updated](https://gist.github.com/jonathandreyer/19a8f3e8f8990973f22984c0ec5c907e/7b0ab73c4ece11a0147af621f605027179d8bb0f). At this point, all converted values seem correct.

As there are 15 received frames (in 4 different clusters), that generate multiples MQTT message.
To reduce this amount to only one messages with all values, I have tried to configure the *endpoint.configureReporting()* function, but that seems not working as expected. With the new external converter, I received 6 MQTT messages (each minute).
I have probably not understand how it is working. Does it possible to reduce into one message? I would like to avoid having multiple time the same value in multiple successive message. In fact, I would like to push the received data into a database and, if I have the same value several times, that will generate weird marks on the graphs.

At this point, I have not looked into this interval modification (by writing into one attribute in the device), but it is the next point that I will look into.

Best regards,

Jonathan Dreyer

jonathandreyer, 2023-05-03:
@Koenkk Thanks to you for your reply. I have tested this option, but that generate some delay, because the conversion of received frame is started as soon as the device is linked (as soon as the _bind_ is done). Is it possible to disable the converters (and ignore _attributeReport_), until the configure steps are not done?

I tried to configure the _endpoint.configureReporting()_ (with this [converter version](https://gist.github.com/jonathandreyer/19a8f3e8f8990973f22984c0ec5c907e/53e3ebf40ed90ecc65240b27718c631cef478a7a)), but it does not seem to work. With this configuration, I receive 14 MQTT messages. The quantity does not change if I remove the four _endpoint.configureReporting()_ configurations.

Is it possible to bypass this problematic, by sending each values in different MQTT topic, like _zigbee2mqtt/0x00124b00xxxxxxxx/value1_, _zigbee2mqtt/0x00124b00xxxxxxxx/value2_, ... (and not one message in _zigbee2mqtt/0x00124b00xxxxxxxx with several values in one payload)?

Best regards,

Jonathan Dreyer

Koenkk, 2023-05-04:
> Is it possible to disable the converters (and ignore attributeReport), until the configure steps are not done?

No this is not possible. 

> The quantity does not change if I remove the four endpoint.configureReporting() configurations.

The configure has already been setup, removing it from the configure won't remove it from the device, you have to call it with different parameters to change it.

jonathandreyer, 2023-05-05:
@Koenkk Thanks to you for your reply. Ok, I understand that it is not possible to enable/disable « on the fly » the configure.

Regarding your second reply, I compare two different situations. To testing each situation, I erase the database, reset the device (unbind the device) and start with a default configuration (same _configuration.yml_ template). With this cleaned setup, I bind the device, wait until one cycle of reporting attributes has been arrived and count how many messages I have received in an MQTT client (MQTT Explorer). In one situation (with _endpoint.configureReporting()_), I use the external converter available in the link. The other one, I remove all codes related to the _endpoint.configureReporting()_ (not published in the gist). If necessary, I can publish it.

For the last question, do you think that it is possible to publish each values in a separated topic? I ask that because, I have the feeling that will solve my issue.

Best regards,

Jonathan Dreyer

Koenkk, 2023-05-05:
Could you provide some debug logging to help me understand the problem better?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

jonathandreyer, 2023-05-06:
@Koenkk Thanks to you for your support. Sure, I will do it with both converters ([with](https://gist.github.com/jonathandreyer/19a8f3e8f8990973f22984c0ec5c907e/53e3ebf40ed90ecc65240b27718c631cef478a7a) _endpoint.configureReporting()_ and [without](https://gist.github.com/jonathandreyer/19a8f3e8f8990973f22984c0ec5c907e)).
I will clean the setup and, install zigbee2mqtt from the dev branch (to be sure). Currently, I use the master branch of zigbee2mqtt with custom zigbee-herdsman (on branch [dev-schneider-powerlogic](https://github.com/jonathandreyer/zigbee-herdsman/tree/dev-schneider-powerlogic)).

I will post both logs ASAP.

Best regards,

Jonathan Dreyer

jonathandreyer, 2023-05-06:
@Koenkk Here are the two log files ([Log_with.txt](https://github.com/Koenkk/zigbee2mqtt/files/11412915/Log_with.txt) & [Log_without.txt](https://github.com/Koenkk/zigbee2mqtt/files/11412918/Log_without.txt)).

I find it odd that when an frame is received (such as the log below), this values converted are append to previous published values. For me, only received values should be in the message.
```
debug 2023-05-06 16:51:25: Received Zigbee message from '0x00124b00xxxxxxxx', type 'attributeReport', cluster 'haDiagnostic', data '{"lastMessageLqi":255,"lastMessageRssi":-48}' from endpoint 10 with groupID 0
info  2023-05-06 16:51:25: MQTT publish: topic 'zigbee2mqtt/0x00124b00xxxxxxxx', payload '{"active_energy_phase1":1215.565,"active_energy_phase2":1727.065,"active_energy_phase3":1340.005,"active_energy_total":4282635,"active_power_demand_phase1":0,"active_power_demand_phase2":0,"active_power_demand_phase3":0,"active_power_demand_total":0,"active_power_phase1":-21,"active_power_phase2":-21,"active_power_phase3":0,"active_power_total":-0.042,"apparent_energy_phase1":1946800,"apparent_energy_phase2":1826260,"apparent_energy_phase3":1494675,"apparent_energy_total":5267735,"apparent_power_demand_phase1":0,"apparent_power_demand_phase2":0,"apparent_power_demand_phase3":0,"apparent_power_demand_total":0,"apparent_power_phase1":42,"apparent_power_phase2":0,"apparent_power_phase3":0,"apparent_power_total":0.042,"communication_quality":100,"current_sensor_multiplier":1,"demand_date_time":8100,"demand_interval_duration":900,"demand_interval_maximal_current_phase1":45,"demand_interval_maximal_current_phase2":0,"demand_interval_maximal_current_phase3":0,"demand_interval_minimal_voltage_phase1":235,"demand_interval_minimal_voltage_phase2":2,"demand_interval_minimal_voltage_phase3":1,"diagnostic_register_1":226,"energy_counters_reporting_period":0,"energy_reset_date_time":0,"frequency":50,"linkquality":123,"lqi_meter":255,"meter_radio_power":10,"meter_status":0,"partial_active_energy_phase1":1215.565,"partial_active_energy_phase2":1727.065,"partial_active_energy_phase3":1340.005,"partial_active_energy_total":4282635,"partial_apparent_energy_phase1":1946800,"partial_apparent_energy_phase2":1826260,"partial_apparent_energy_phase3":1494675,"partial_apparent_energy_total":5267735,"partial_reactive_energy_phase1":419450,"partial_reactive_energy_phase2":310005,"partial_reactive_energy_phase3":182.965,"partial_reactive_energy_total":912420,"power_factor_phase1":-50,"power_factor_phase2":100,"power_factor_phase3":100,"reactive_energy_phase1":419450,"reactive_energy_phase2":310005,"reactive_energy_phase3":182.965,"reactive_energy_total":912420,"reactive_power_demand_phase1":0,"reactive_power_demand_phase2":0,"reactive_power_demand_phase3":0,"reactive_power_demand_total":0,"reactive_power_phase1":0,"reactive_power_phase2":-21,"reactive_power_phase3":0,"reactive_power_total":-0.021,"rssi_meter":-48,"voltage_phase1":237}'
```

Best regards,

Jonathan Dreyer

Koenkk, 2023-05-07:
> I find it odd that when an frame is received (such as the log below), this values converted are append to previous published values. For me, only received values should be in the message.

You can disable the state caching with the [`cache_state` option](https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#server-connection). Are there any other issues I can help with?

jonathandreyer, 2023-05-14:
Hello @Koenkk, sorry for my delay. Many hanks you for your reply and your feedback. Maybe a note about *cache_state* in the documentation could be useful, as I had read the documentation more than once without finding it. With your message I was able to stumble upon it, but maybe it would be useful to have an introductory message with some default behavior setting. I understand the need for the option (especially for home automation), but I am not sure it is specially intuitive for people who are looking to use it in another area.

Now, that it works pretty well as expected with the new [version](https://gist.github.com/jonathandreyer/19a8f3e8f8990973f22984c0ec5c907e) of the converter. I will launch a round of measuring with multiple devices to verify that all messages are properly collected. I will be back as soon as this checking phase is done.

Best regards,

Jonathan Dreyer

closed 2023-06-21 by github-actions[bot]