Koenkk/zigbee2mqtt #17389: devices offline after few minutes

MrMontesa, 2023-04-19:
### What happened?

Hey there,
I have a problem that half of my light go offline after 10mins. While troubleshooting it does not look like a range problem, but something else. First a bit of background, then current error description.
Coming from Openhab with deconz docker container with a conbee2 stick. All smooth, no issues, but all lights were on outdated firmware.

Then migrated to Home-assistant (2023.4.4) with M2Z add-on. Conbe2 stick did not work well, so I bought the Sonoff ZBDonge-E, flashed it with 6.10.3.0 build 297 and setup everything from scratch. Its just 14 devices. Mostly Hue lights and Ikea Tradfri remotes. The coordinator is in basement mounted to the ceiling, most Hue bulbs on mains, are on ground floor. Remotes are on 1st floor.

Including the devices was already a pain, since devices were not found. I had to bring some lights into basement next to the coordinator. Easy for a lightbulb, but hard for some mounted LED strips. Also wasnt able to include one of the strips at all. So I optimized the coordinator position. Long USB cable, played with antenna position, move a Hue light to basement to better routing, etc.

But I still have devices going offline really soon. When I restart Z2M I can control everything nicely for 10mins, but then they drop off and become unresponsive.
Interestingly I think it cant be a range issue, because the remotes on 1st floor, still work nicely after hours, whereas the Hue bulbs on ground floor are offline.
Here are some screenshots of the current setup:
[image]

[image]

Also saw in the map that everything connects back to the USB dongle and no routing via bulbs, what I would have expected. 

Any idea on how to keep the ground floor lights online would be appreciated.
Thanks much
Sebastian

### What did you expect to happen?

Lights should not drop off and become offline

### How to reproduce it (minimal and precise)

Not sure. In my environment, I restart Z2M, all devices are online and I can control them. After 10mins or so they drop off and become offline, while the far away battery driven remotes still operate.

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDonge-E

### Debug log

```
Warning 2023-04-19 08:04:50Failed to ping 'Esszimmer_Iris' (attempt 1/1, Read 0x0017880100c05c61/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41 - 11 - 69 - 0 - 1 after 10000ms))
Error 2023-04-19 08:04:51Publish 'set' 'state' to 'TV_Bank_LED_Leiste' failed: 'Error: Command 0x0017880100cbb565/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 18304 - 11 - 68 - 6 - 11 after 10000ms)'
```

Wikibear, 2023-04-19:
I'm not really sure if this is really an error. But you didn't connect well to a device if you set permit all. If you want to have more control, i suggest you to delete all devices and start again with drop down and choose the coordinator or a repeater device like a lamp. You can say a lamp in a room connect to the coordinator directly and a battery device can connect to the lamp. 
If you say all the time permit all, then the devices route by themself to the coordinator or a repeater device and you don't have any control about the routing.

MrMontesa, 2023-04-19:
Thanks for your feedback. I was indeed adding the devices via permit all. I was assuming the devices will connect to the "shortest path" by themselves automatically.
Will reset the coordinator and try once more.
Thanks

MrMontesa, 2023-04-19:
Seems like it didnt work. I removed all devices and started to add a Hue lightbulb (Kugellampe) from groundfloor as first routing device by "permit coordinator". The distance to coordinator is quite less and it worked like charm.
Then I added a Hue strip with "permit Kugellampe" which is 1 meter away from the first added bulb. Worked like charm as well. 

Then tried adding a 3rd bulb (Hue Iris) via "permit Kugellampe" as well and it failed.
Error:
```
Info 2023-04-19 22:07:52Zigbee: allowing new devices to join via Kugellampe.
Error 2023-04-19 22:08:02Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: '{"address":33526,"clusterId":32822,"sequence":126} after 10000ms'
Info 2023-04-19 22:08:02MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{},"error":"{\"address\":33526,\"clusterId\":32822,\"sequence\":126} after 10000ms","status":"error","transaction":"rpaym-2"}'
```

On the next try, the "Kugellampe" Hue bulb was already offline again.
[image]

While trying to add devices I also had error message, where device initially joined, but then bailed out with "interview failed" error.
Any idea what might be wrong?
Thanks

closed 2023-05-27 by github-actions[bot]

evd1ser, 2023-09-07:
Same problem, how you are resolve this?