Koenkk/zigbee2mqtt #17396: [New device support]: TS0601 - _TZE200_hi5qczzt

majo053, 2023-04-20:
### Link

https://www.aliexpress.com/item/1005004696749635.html

### Database entry

{"id":30,"type":"EndDevice","ieeeAddr":"0xa4c1382f7e65639f","nwkAddr":34917,"manufId":4417,"manufName":"_TZE200_hi5qczzt","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1681975126703,"defaultSendRequestWhen":"immediate"} {"id":31,"type":"EndDevice","ieeeAddr":"0xa4c1380a3ac7ba34","nwkAddr":61530,"manufId":4417,"manufName":"_TZE200_hi5qczzt","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_hi5qczzt","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1681980555332,"defaultSendRequestWhen":"immediate"}

### Comments

Hello, so I just adjusted one of the other examples, and now it works fine with Smoke detection and Battery Low.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    tuya_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.state:
                    result.smoke = value === 0;
                    break;
                case 14:
                    result.battery_low = value === 0;
                    break;
                case 16:
                    result.silence_siren = value;
                    break;
                case 20: {
                    const alarm = {0: true, 1: false};
                    result.alarm = alarm[value];
                    break;
                    }
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    }
}

const definition = {
    zigbeeModel: ['TS0601'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_hi5qczzt'}],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [fzLocal.tuya_smoke, fz.battery],
    toZigbee: [],
    exposes: [e.smoke(), e.battery(), e.battery_low()],
    meta: {
        tuyaDatapoints: [
        [1, 'smoke', {from: (v) => v === 0}],
        [6, 'silence_siren', {from: (v) => v === 0}],
        [14, 'battery_low', {from: (v) => v === 0}],
        ],
    },
};

module.exports = definition;
```

majo053, 2023-04-20:
`Zigbee2MQTT:debug 2023-04-20 08:37:21: Received Zigbee message from 'Smoke sensor 154', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":6144}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:21: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Zigbee2MQTT:debug 2023-04-20 08:37:22: Received Zigbee message from 'Smoke sensor 154', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":6144}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:22: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Zigbee2MQTT:debug 2023-04-20 08:37:23: Received Zigbee message from 'Server socket', type 'attributeReport', cluster 'genBasic', data '{"65506":48,"65508":0,"appVersion":68}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-04-20 08:37:25: Received Zigbee message from 'Smoke sensor 154', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":6400}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:25: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[1]}}
Zigbee2MQTT:debug 2023-04-20 08:37:28: Received Zigbee message from 'Smoke sensor 153', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":1280}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:28: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Zigbee2MQTT:debug 2023-04-20 08:37:29: Received Zigbee message from 'Smoke sensor 153', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":1280}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:29: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Zigbee2MQTT:debug 2023-04-20 08:37:31: Received Zigbee message from 'Smoke sensor 153', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":1536}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:31: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[1]}}
Zigbee2MQTT:debug 2023-04-20 08:37:32: Received Zigbee message from 'Smoke sensor 153', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":1536}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:32: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[1]}}
Zigbee2MQTT:debug 2023-04-20 08:37:42: Received Zigbee message from 'Smoke sensor 153', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":1792}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:42: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Zigbee2MQTT:debug 2023-04-20 08:37:44: Received Zigbee message from 'Smoke sensor 153', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":1792}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:44: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Zigbee2MQTT:debug 2023-04-20 08:37:45: Received Zigbee message from 'Smoke sensor 153', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":2048}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:45: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[1]}}
Zigbee2MQTT:debug 2023-04-20 08:37:47: Received Zigbee message from 'Smoke sensor 153', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":2048}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2023-04-20 08:37:47: zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[1]}}`

Koenkk, 2023-04-20:
- What does `silence_siren` do? Can you add it to the `exposes`?
- Probably `[4, 'tamper', tuya.valueConverter.raw],` with `e.tamper()` as exposes also works.

majo053, 2023-04-24:
> * What does `silence_siren` do? Can you add it to the `exposes`?
> * Probably `[4, 'tamper', tuya.valueConverter.raw],` with `e.tamper()` as exposes also works.

Sorry, silence_siren doesnt work, works only smoke and battery.

Koenkk, 2023-04-24:
It does not work at all? (so should be removed?)

majo053, 2023-04-25:
> It does not work at all? (so should be removed?)

Works only smoke detection, battery is not working. Can you please help mi with this?

majo053, 2023-04-25:
> Can you try with:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ytibqbra', '_TZE200_hi5qczzt']),
>     model: 'TS0601_smoke_5',
>     vendor: 'TuYa',
>     description: 'Smoke sensor',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     configure: tuya.configureMagicPacket,
>     exposes: [e.smoke(), e.tamper(), e.battery(), tuya.exposes.faultAlarm(),
>         tuya.exposes.silence(), exposes.binary('alarm', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable the alarm')],
>     meta: {
>         tuyaDatapoints: [
>             [1, 'smoke', tuya.valueConverter.trueFalse0],
>             [4, 'tamper', tuya.valueConverter.raw],
>             [11, 'fault_alarm', tuya.valueConverter.trueFalse1],
>             [15, 'battery', tuya.valueConverter.raw],
>             [16, 'silence', tuya.valueConverter.raw],
>             [17, 'alarm', tuya.valueConverter.onOff],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```

Thank you for this converter, smoke was detected, but tamper, battery, fault alarm, silence and alarm doest work :(

`Debug 2023-04-25 20:17:16Received MQTT message on 'zigbee2mqtt/Smoke sensor 154/set/alarm' with data 'ON'
Debug 2023-04-25 20:19:01Received MQTT message on 'zigbee2mqtt/Smoke sensor 154/set/silence' with data 'true'
Debug 2023-04-25 20:19:01Publishing 'set' 'silence' to 'Smoke sensor 154'
Info 2023-04-25 20:19:01MQTT publish: topic 'zigbee2mqtt/Smoke sensor 154', payload '{"alarm":null,"battery":null,"fault_alarm":null,"linkquality":111,"silence":true,"smoke":true,"tamper":null}'
Debug 2023-04-25 20:19:02Received MQTT message on 'zigbee2mqtt/Smoke sensor 154/set/silence' with data 'false'
Debug 2023-04-25 20:19:02Publishing 'set' 'silence' to 'Smoke sensor 154'
Info 2023-04-25 20:19:02MQTT publish: topic 'zigbee2mqtt/Smoke sensor 154', payload '{"alarm":null,"battery":null,"fault_alarm":null,"linkquality":111,"silence":false,"smoke":true,"tamper":null}'
Debug 2023-04-25 20:19:03Received MQTT message on 'zigbee2mqtt/Smoke sensor 154/set/silence' with data 'true'
Debug 2023-04-25 20:19:03Publishing 'set' 'silence' to 'Smoke sensor 154'
Info 2023-04-25 20:19:03MQTT publish: topic 'zigbee2mqtt/Smoke sensor 154', payload '{"alarm":null,"battery":null,"fault_alarm":null,"linkquality":111,"silence":true,"smoke":true,"tamper":null}'
Debug 2023-04-25 20:19:30Received MQTT message on 'zigbee2mqtt/Smoke sensor 154/set/silence' with data 'false'
Debug 2023-04-25 20:19:30Publishing 'set' 'silence' to 'Smoke sensor 154'
Info 2023-04-25 20:19:30MQTT publish: topic 'zigbee2mqtt/Smoke sensor 154', payload '{"alarm":null,"battery":null,"fault_alarm":null,"linkquality":111,"silence":false,"smoke":true,"tamper":null}'
Debug 2023-04-25 20:19:30Received MQTT message on 'zigbee2mqtt/Smoke sensor 154/set/silence' with data 'true'
Debug 2023-04-25 20:19:30Publishing 'set' 'silence' to 'Smoke sensor 154'
Info 2023-04-25 20:19:30MQTT publish: topic 'zigbee2mqtt/Smoke sensor 154', payload '{"alarm":null,"battery":null,"fault_alarm":null,"linkquality":111,"silence":true,"smoke":true,"tamper":null}'`

[image]

[image]

[image]

Koenkk, 2023-04-27:
Probably the other features are not supported by the device, can you check if dp 14 is battery_state? Try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_hi5qczzt'}, // [KnockautX / Brelag AG, Switzerland](https://www.brelag.com)
    ],
    vendor: 'TuYa',
    model: 'TS0601_smoke_3',
    description: 'Photoelectric smoke detector',
    whiteLabel: [
        {vendor: 'KnockautX', model: 'SMOAL024'},
    ],
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [e.smoke(), tuya.exposes.batteryState()],
    meta: {
        tuyaDatapoints: [
            /**
             * According to the Vendor "KnockautX / Brelag AG" DP 16 "muffling"
             * is supported as well. But it was not possible to verify this using
             * SMOLA024 devices - therefore it is not included in the device definition.
             *
             * Data Transfer Type: Send and Report
             * Data Type: Bool
             * muffling: 16,
             */
            [1, 'smoke', tuya.valueConverter.trueFalse0],
            [14, 'battery_state', tuya.valueConverter.batteryState],
        ],
    },
};

module.exports = definition;
```

majo053, 2023-04-28:
> Probably the other features are not supported by the device, can you check if dp 14 is battery_state? Try with:
>
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const [...]

Hi, dp 14 doesnt work. Its strange. Works only smoke alarm. Do you have any other idea?

If I press button, this show in logs:

```
2023-04-28 21:13:22Datapoint 9 not defined for '_TZE200_hi5qczzt' with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[0]}}
2023-04-28 21:13:23Datapoint 9 not defined for '_TZE200_hi5qczzt' with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[1]}}
2023-04-28 21:13:25Datapoint 9 not defined for '_TZE200_hi5qczzt' with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[1]}}
```

Maybe test alarm?

Koenkk, 2023-04-29:
- 9 is probably self test result, added it in https://github.com/Koenkk/zigbee2mqtt/issues/17396#issuecomment-1524894506
- when leaving the device running for 24 hours, do you see any other `Datapoint X not defined for '_TZE200_hi5qczzt' with data` in the log?

majo053, 2023-05-09:
> * 9 is probably self test result, added it in [[New device support]: TS0601 - _TZE200_hi5qczzt #17396 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/17396#issuecomment-1524894506)
> * when leaving the device running for 24 hours, do you see any other `Datapoint X not defined for '_TZE200_hi5qczzt' with data` in the log?

Hello, during 3 days a cant see in syslog messages any "Datapoint X not defined for...". Can you please add support for smoke and self test?

Koenkk, 2023-05-09:
@majo053 I need to know the datapoints (https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)

majo053, 2023-05-19:
> @majo053 I need to know the datapoints (https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)

Hello, sorry I dont have tuya zigbee router. I try your tutorial, but i stuck at "6. Find your devices". What I can do?

Thank you for your support!

Koenkk, 2023-05-20:
Without TuYa gateway it is not possible to do this.

majo053, 2023-06-02:
> Without TuYa gateway it is not possible to do this.

Hello, thakn you anyway, i will wait for "others" and support for this smoke detector.

closed 2023-07-11 by github-actions[bot]