Koenkk/zigbee2mqtt #17397: [New device support]: Fan controller _TZE200_fvldku9h

Big-Szu, 2023-04-20:
### Link

https://pl.aliexpress.com/item/4001237395363.html

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x60a423fffe7ec839","nwkAddr":27932,"manufId":4098,"manufName":"_TZE200_fvldku9h","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���+\u0012���+\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_fvldku9h","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xdc8e95fffe257ec8","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0xdc8e95fffe257ec8","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1681982517620,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to create custom converter and was able to make it turn on and off but got stuck with changing fan speed.

Log entries after changing speed on the device:
2023-04-19 19:22:29Datapoint 4 not defined for '_TZE200_fvldku9h' with data {"dp":4,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,23]}}
2023-04-19 19:22:29Received Zigbee message from '0x60a423fffe7ec839', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,23],"type":"Buffer"},"datatype":2,"dp":4}],"seq":20992}' from endpoint 1 with groupID 0,

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_fvldku9h',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Fan switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
exposes: [
            tuya.exposes.switch(), e.power_on_behavior(['off', 'on']).withAccess(ea.STATE_SET),
          //  tuya.exposes.countdown().withValueMin(0).withValueMax(43200).withUnit('s').withDescription('Max ON time in seconds'),
            exposes.numeric('fan_speed', ea.STATE_SET).withValueMin(1).withValueMax(5).withValueStep(1)
                .withDescription('Speed off the fan'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff],
            //    [2, 'countdown', tuya.valueConverter.countdown],
                [4, 'fan_speed', tuya.valueConverterBasic
                    .lookup({'1': tuya.enum(0), '2': tuya.enum(1), '3': tuya.enum(2), '4': tuya.enum(3), '5': tuya.enum(4)})],
              //  [11, 'power_on_behavior', tuya.valueConverterBasic.lookup({'off': tuya.enum(0), 'on': tuya.enum(1)})],
            ],
        },
};

module.exports = definition;
```

Big-Szu, 2023-04-20:
Extract from states
`{
    "fan_speed": 3,
    "linkquality": 100,
    "state": "ON"
}`

Big-Szu, 2023-04-21:
Today I connected the switch to Tuya gateway and got some details from the Tuya API

[image]

And here is a list of supported DP IDs
`    "category": "fskg",
    "functions": [
      {
        "code": "switch_fan",
        "dp_id": 1,
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "countdown_fan",
        "dp_id": 2,
        "type": "Integer",
        "values": "{\"unit\":\"s\",\"min\":0,\"max\":43200,\"scale\":0,\"step\":1}"
      },
      {
        "code": "fan_speed",
        "dp_id": 4,
        "type": "Integer",
        "values": "{\"unit\":\"%\",\"min\":0,\"max\":100,\"scale\":0,\"step\":1}"
      },
      {
        "code": "relay_status",
        "dp_id": 11,
        "type": "Enum",
        "values": "{\"range\":[\"off\",\"on\",\"memory\"]}"
      },
      {
        "code": "backlight_switch",
        "dp_id": 13,
        "type": "Boolean",
        "values": "{}"
      }
    ],
    "status": [
      {
        "code": "switch_fan",
        "dp_id": 1,
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "countdown_fan",
        "dp_id": 2,
        "type": "Integer",
        "values": "{\"unit\":\"s\",\"min\":0,\"max\":43200,\"scale\":0,\"step\":1}"
      },
      {
        "code": "fan_speed",
        "dp_id": 4,
        "type": "Integer",
        "values": "{\"unit\":\"%\",\"min\":0,\"max\":100,\"scale\":0,\"step\":1}"
      },
      {
        "code": "relay_status",
        "dp_id": 11,
        "type": "Enum",
        "values": "{\"range\":[\"off\",\"on\",\"memory\"]}"
      },
      {
        "code": "backlight_switch",
        "dp_id": 13,
        "type": "Boolean",
        "values": "{}"
      }
    ]`

Big-Szu, 2023-04-21:
So, eventually got it working! 🥇 Fan speed and count down are being exposed to HA as numeric values and the device works as a switch, but it functions properly :)

Updated External converter in first post.

Big-Szu, 2023-04-21:
Closing as completed

closed 2023-04-21 by Big-Szu

Big-Szu, 2023-04-21:
Made correction to Converter as the ON/OFF switch was not working and Power On Behaviour was missing 'memory' value.

Issues is stil with the backlight

Lotambena, 2023-08-30:
Thanks for the good work. it applied your code and worked. the device that is formed is not recognized as a fan . after changing device type on mqtt and the fan speed is not available on the lovelace. only the on/off functions are available. did you experience the same challenges and do you have some pointers.

CIPH3RD, 2024-11-16:
bit late here, but I have the same switch, this works for all features of the fan switch:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_fvldku9h']),
    model: 'ZB-1SRW-US',
    vendor: 'Somgoms',
    description: 'Fan switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    whiteLabel: [{vendor: 'Somgoms', model: 'ZB-1SRW-US'},
        tuya.whitelabel('Somgoms', 'ZB-1SRW-US', 'Fan switch', ['_TZE200_fvldku9h']),
    ],
    exposes: [
        e.binary('backlight_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Backlight switch'),
        e.binary('switch_fan', ea.STATE_SET, 'ON', 'OFF').withDescription('Fan On Off'),
        e.numeric('fan_speed', ea.STATE_SET).withValueMin(0).withValueMax(100).withValueStep(1).withUnit('%').withDescription('Fan Speed'),
        e.enum('relay_status', ea.STATE_SET, ['off', 'on', 'memory']).withDescription('Recover state after power outage'),
        e.numeric('countdown_fan', ea.STATE_SET).withValueMin(0).withValueMax(43200).withValueStep(1).withUnit('s').withDescription('Max ON time in seconds'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'switch_fan', tuya.valueConverter.onOff],
            [2, 'countdown_fan', tuya.valueConverter.countdown],
            [4, 'fan_speed', tuya.valueConverter.raw],
            [11, 'relay_status', tuya.valueConverter.raw],
            [13, 'backlight_switch', tuya.valueConverter.onOff],
        ],
    },
};
module.exports = definition;`

Lotambena, 2025-01-13:
this converter is no longer, loading in the new update, device are unsupported. anyone experiencing this problem