Koenkk/zigbee2mqtt #17400: external converter breaks down z2m

maxmib, 2023-04-20:
### What happened?

external converter breaks down z2m ,z2m works well after removing it ,but I need it to use my device 
this is the external file 
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const bind = async (endpoint, target, clusters) => {
    for (const cluster of clusters) {
        await endpoint.bind(cluster, target);
    }
};

const withEpPreffix = (converter) => ({
    ...converter,
    convert: (model, msg, publish, options, meta) => {
        const epID = msg.endpoint.ID;
        const converterResults = converter.convert(model, msg, publish, options, meta);
        return Object.keys(converterResults)
            .reduce((result, key) => {
                result[`${key}_${epID}`] = converterResults[key];
                return result;
            }, {});
    },
});	
const device = {
    zigbeeModel: ['Sensor_Hub_01'],
    model: 'Sensor_Hub_01',
    vendor: 'Sensor_Hub',
    description: '[Sensor hub]',
    supports: 'temperature,humidity,illuminance,pm2.5',
    fromZigbee: [
        fz.temperature,        
		fz.humidity,
		fz.illuminance,
		fz.heiman_pm25		
    ],
    toZigbee: [], 
    exposes: [
        e.humidity(),
        e.temperature(),
		e.illuminance(),
		e.pm25(),        
        e.linkquality()		
    ]
};

module.exports = device;
```

### What did you expect to happen?

help with modifying the external converter file or add device support in z2m directly

### How to reproduce it (minimal and precise)

add my external converter file and restart z2m

### Zigbee2MQTT version

1.30.3 commit: 24c6b2e

### Adapter firmware version

20221226

### Adapter

sonoff zigbee dongle plus (cc2652p)

### Debug log

```
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Timeout.restart [as _onTimeout] (/app/index.js:17:5)

```

Koenkk, 2023-04-20:
`fz.heiman_pm25	` should be `fz.pm25`

maxmib, 2023-04-21:
> `fz.heiman_pm25	` should be `fz.pm25`

Thanks,it works .
And about writing external converter file ,is there any guide documents I can follow as a newbie?Thanks again.

Koenkk, 2023-04-21:
Most is available under https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html, do you have any specific questions? (what features of the device do not work yet?)

maxmib, 2023-04-22:
> `fz.heiman_pm25 ` should be `fz.pm25`

this device works well,but its converter file is writted by someone else ,I just want to learn how converter file works and try to write some converters  to test other unsupported devices by myself .I will start from the link you mentioned above .thanks.

closed 2023-04-22 by maxmib