Koenkk/zigbee2mqtt #17403: [New device support]: S3 003 Smart in wall plug

maxmib, 2023-04-20:
### Link

no product page

### Database entry

{"id":25,"type":"Router","ieeeAddr":"0xedd9a514006f0d00","nwkAddr":43324,"manufId":4638,"manufName":"umeinfo","powerSource":"Unknown","modelId":"zigbee_transfer","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":43690,"inClusterList":[0,3,4,5,6,8,768,3073],"outClusterList":[25,3073],"clusters":{"genBasic":{"attributes":{"modelId":"zigbee_transfer","manufacturerName":"umeinfo","powerSource":0,"zclVersion":2,"appVersion":1,"stackVersion":98,"hwVersion":133}},"haElectricalMeasurement":{"attributes":{}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b000be958f7","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b000be958f7","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":1,"stackVersion":98,"hwVersion":133,"zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1682005880346,"defaultSendRequestWhen":"immediate"}

### Comments

z2m pairs this plug and receives data ,but I don't know how to transfer the data 

data received 
```
info  2023-04-20 23:44:29: Device 'S3' joined
info  2023-04-20 23:44:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"S3","ieee_address":"0xedd9a514006f0d00"},"type":"device_joined"}'
info  2023-04-20 23:44:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"S3"},"type":"device_connected"}'
info  2023-04-20 23:44:30: Starting interview of 'S3'
info  2023-04-20 23:44:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"S3","ieee_address":"0xedd9a514006f0d00","status":"started"},"type":"device_interview"}'
info  2023-04-20 23:44:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"S3"},"type":"pairing"}'
debug 2023-04-20 23:44:30: Device 'S3' announced itself
info  2023-04-20 23:44:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"S3","ieee_address":"0xedd9a514006f0d00"},"type":"device_announce"}'
info  2023-04-20 23:44:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"S3"},"type":"device_announced"}'
debug 2023-04-20 23:44:30: Received Zigbee message from 'S3', type 'raw', cluster '3073', data '{"data":[28,30,18,10,128,47,83,56,53,47,49,47,50,44,56],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:30: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:31: Received Zigbee message from 'S3', type 'raw', cluster '3073', data '{"data":[28,30,18,11,128,47,83,56,52,47,49,47,48],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:31: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:31: Received Zigbee message from 'S3', type 'readResponse', cluster 'genBasic', data '{"modelId":"zigbee_transfer"}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:31: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:31: Received Zigbee message from 'S3', type 'raw', cluster '3073', data '{"data":[28,30,18,12,128,47,83,56,53,47,49,47,50,44,56],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:31: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:31: Received Zigbee message from 'S3', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"umeinfo"}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:31: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:31: Received Zigbee message from 'S3', type 'readResponse', cluster 'genBasic', data '{"powerSource":0}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:31: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:31: Received Zigbee message from 'S3', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:31: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:31: Received Zigbee message from 'S3', type 'readResponse', cluster 'genBasic', data '{"appVersion":1}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:31: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:31: Received Zigbee message from 'S3', type 'readResponse', cluster 'genBasic', data '{"stackVersion":98}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:31: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:31: Received Zigbee message from 'S3', type 'readResponse', cluster 'genBasic', data '{"hwVersion":133}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:31: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:32: Received Zigbee message from 'S3', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:32: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:32: Received Zigbee message from 'S3', type 'raw', cluster '3073', data '{"data":[28,30,18,13,128,47,83,56,53,47,49,47,50,44,56],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:32: Skipping message, definition is undefined and still interviewing
debug 2023-04-20 23:44:32: Received Zigbee message from 'S3', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2023-04-20 23:44:32: Skipping message, definition is undefined and still interviewing
info  2023-04-20 23:44:32: Successfully interviewed 'S3', device has successfully been paired
info  2023-04-20 23:44:32: Device 'S3' is supported, identified as: umeinfo S3 Wall Plug (S3 003)
info  2023-04-20 23:44:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"S3 Wall Plug","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"S3 003","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device.","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for current, takes into effect on next report of device.","name":"current_precision","property":"current_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for voltage, takes into effect on next report of device.","name":"voltage_precision","property":"voltage_precision","type":"numeric","value_max":3,"value_min":0}],"supports_ota":false,"vendor":"umeinfo"},"friendly_name":"S3","ieee_address":"0xedd9a514006f0d00","status":"successful","suppor [... cut ...]
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint);
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.rmsCurrent(endpoint);
        await reporting.activePower(endpoint);
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

closed 2023-05-29 by github-actions[bot]

t9080, 2023-06-05:
我也有这种插座，因为服务器无法访问，请问有方法重新刷固件吗

Nova2003-emoji, 2024-07-22:
解决了吗