Koenkk/zigbee2mqtt #17408: Can' connect to Sonoff Zigbee 3.0 plus USB Dongle

julred, 2023-04-20:
### What happened?

I already had my Sonoff Stick working on an old Proxmox installation of zigbee2mqtt.
But I switched to a new server and re-installed zigbee2mqtt once more.
Afterwards it was not possible to start zigbee2mqtt because of this log:
```

alpine-zigbee2mqtt:/var/lib/zigbee2mqtt/log/2023-04-20.19-28-00# cat log.txt 
info  2023-04-20 19:28:00: Logging to console and directory: '/var/lib/zigbee2mqtt/log/2023-04-20.19-28-00' filename: log.txt
info  2023-04-20 19:28:00: Starting Zigbee2MQTT version 1.28.2-dev (commit #null)
info  2023-04-20 19:28:00: Starting zigbee-herdsman (0.14.68)
error 2023-04-20 19:28:00: Error while starting zigbee-herdsman
error 2023-04-20 19:28:00: Failed to start zigbee
error 2023-04-20 19:28:00: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-04-20 19:28:00: Exiting...
error 2023-04-20 19:28:00: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port001'
    at SerialPort.<anonymous> (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:128:28)
    at SerialPort._error (/usr/lib/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:198:14)
    at /usr/lib/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:242:12
alpine-zigbee2mqtt:/var/lib/zigbee2mqtt/log/2023-04-20.19-28-00# 
```

### Zigbee2MQTT version

not able to verify

### Adapter firmware version

not able to verify

### Adapter

Sonoff Zigbee 3.0 plus

juanchula, 2023-04-20:
Did you change the port in configuration.yaml? It's possible that the new server detects the coordinator with a different name. To see what the new name of the port is, you can execute `ls /dev/serial/by-id/*`

julred, 2023-04-21:
> Did you change the port in configuration.yaml? It's possible that the new server detects the coordinator with a different name. To see what the new name of the port is, you can execute `ls /dev/serial/by-id/*`

Thanks for your reply.

Of course I changed the configuration.yaml in /etc/zigbee2mqtt folder:

```
alpine-zigbee2mqtt:/etc/zigbee2mqtt# cat configuration.yaml 
frontend:
  port: 9442
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.178.150:1883
  user: usr
  password: passwd****
  keepalive: 60
  reject_unauthorized: true
  version: 4
serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port001
advanced:
  pan_id: GENERATE
  network_key: GENERATE
  channel: 20
```

closed 2023-06-20 by github-actions[bot]