Koenkk/zigbee2mqtt #17412: SONOFF Zigbee dongle E not working with Z2MQTT since HA OS 10.0 update

hardwareadictos, 2023-04-21:
### What happened?

That's it. Worked just fine til the update, this is what i am getting on logs:

Rolled back two times to discardf any update issues, with no luck.

### Zigbee2MQTT version

1.30.3-1 

### Adapter firmware version

6.10.3_V1.0.0

### Adapter

SONOFF ZBDongle-E

### Debug log

[11:34:09] INFO: Preparing to start...
[11:34:09] INFO: Socat not enabled
[11:34:10] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-04-21 11:34:12: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-04-21 11:34:12: Logging to console and directory: '/config/zigbee2mqtt/log/2023-04-21.11-34-12' filename: log.txt
Zigbee2MQTT:debug 2023-04-21 11:34:12: Removing old log directory '/config/zigbee2mqtt/log/2023-04-21.11-31-09'
Zigbee2MQTT:info  2023-04-21 11:34:12: Starting Zigbee2MQTT version 1.30.3 (commit #unknown)
Zigbee2MQTT:info  2023-04-21 11:34:12: Starting zigbee-herdsman (0.14.103)
Zigbee2MQTT:debug 2023-04-21 11:34:12: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812172611-if00"}}'
Error: Reset error: Error: {"sequence":-1} after 10000ms
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:298:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

hardwareadictos, 2023-04-21:
Enabled debug mode:

```
[15:14:23] INFO: Preparing to start...
[15:14:23] INFO: Socat not enabled
[15:14:23] INFO: Zigbee Herdsman debug logging enabled
[15:14:24] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-04-21 15:14:27: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-04-21 15:14:27: Logging to console and directory: '/config/zigbee2mqtt/log/2023-04-21.15-14-27' filename: log.txt
Zigbee2MQTT:debug 2023-04-21 15:14:27: Removing old log directory '/config/zigbee2mqtt/log/2023-04-21.14-57-55'
Zigbee2MQTT:info  2023-04-21 15:14:27: Starting Zigbee2MQTT version 1.30.3 (commit #unknown)
Zigbee2MQTT:info  2023-04-21 15:14:27: Starting zigbee-herdsman (0.14.103)
Zigbee2MQTT:debug 2023-04-21 15:14:27: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812172611-if00"}}'
2023-04-21T13:14:27.210Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-04-21T13:14:27.212Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812172611-if00","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-04-21T13:14:27.213Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812172611-if00","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-04-21T13:14:27.241Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2023-04-21T13:14:27.242Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2023-04-21T13:14:27.242Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2023-04-21T13:14:27.243Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2023-04-21T13:14:27.244Z zigbee-herdsman:adapter:ezsp:uart -?- waiting reset
2023-04-21T13:14:37.247Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":-1} after 10000ms
2023-04-21T13:14:37.248Z zigbee-herdsman:adapter:ezsp:driv Reset connection. Try 0
2023-04-21T13:14:37.248Z zigbee-herdsman:adapter:ezsp:driv Stop driver
2023-04-21T13:14:37.249Z zigbee-herdsman:adapter:ezsp:ezsp Stop ezsp
2023-04-21T13:14:37.249Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-04-21T13:14:37.250Z zigbee-herdsman:adapter:ezsp:driv Close driver
Error: Reset error: Error: {"sequence":-1} after 10000ms
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:298:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

hardwareadictos, 2023-04-28:
Updated to HAOS 10.1, same issue. Can anyone help me please?

hardwareadictos, 2023-05-31:
More than a month and te issue still persists. Fixed it temporary by commenting out #adapter: ezsp on configuration.yml.

Today i updated to 10.2 OS and it doesnt't even care of this commit, my HA installation is USELESS:

```
[18:12:51] INFO: Preparing to start...
[18:12:52] INFO: Socat not enabled
[18:12:52] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-05-31 18:12:55: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-05-31 18:12:55: Logging to console and directory: '/config/zigbee2mqtt/log/2023-05-31.18-12-55' filename: log.txt
Zigbee2MQTT:debug 2023-05-31 18:12:55: Removing old log directory '/config/zigbee2mqtt/log/2023-05-31.18-02-18'
Zigbee2MQTT:info  2023-05-31 18:12:55: Starting Zigbee2MQTT version 1.30.4 (commit #unknown)
Zigbee2MQTT:info  2023-05-31 18:12:55: Starting zigbee-herdsman (0.14.111)
Zigbee2MQTT:debug 2023-05-31 18:12:55: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220812172611-if00"}}'
Error: Reset error: Error: {"sequence":-1} after 10000ms
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:299:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

lx2905, 2023-06-20:
I got the same issue. Only thing that fixes this for me for a few hours is to unplug and plug the Dongle again, restart zigbee2mqtt on HA.

hardwareadictos, 2023-06-20:
what pisses me off is that i didn't even get an answer or any supoort. E-Dongle is on beta stage and will be always on beta if no one takes care of those issues. This issue was even stale (why?)!!!

Updated to HAOS 10.3 today, same situation...

lx2905, 2023-06-21:
I tried uncommenting the ezsp, but didnt change anything thought. Would be nice if somebody is going to fix this. Otherwise I'll try to get a Dongle-P and change the Hardware.

kirovilya, 2023-06-21:
@hardwareadictos Greetings.
1. I hope you do not have ZHA integration running in parallel, it must be disabled so that it does not try to connect to the USB port.
2. I hope you use a USB extension cable so that the stick is away from the USB3 ports.
3. I hope you don't have nearby devices (disks, flash drives) plugged into USB3.

kirovilya, 2023-06-21:
@lx2905 What hardware is z2m running on? maybe not enough power?
what version of z2m?

hardwareadictos, 2023-06-21:
> I have not installed ZHA. The Error persists with and without extension cable. I even unplugged all other usb-devices nearby.
> 
> If I plug and unplug the stick and restart z2m, it works for a few hours, then crashes again.

Same situation here

hardwareadictos, 2023-06-21:
> @hardwareadictos Greetings.
>
> 1. I hope you do not have ZHA integration running in parallel, it must be disabled so that it does not try to connect to the USB port.
> 2. I hope you use a USB extension [...]

1 - ZHA is disabled on Home Assistant
2 - Using extension cable since day 1 as i dont want the stick to be inside my rack.
3 - Don't. That stick is the only USB on the machine and its fully passthrough to HAOS VM on Proxmox, issue started to appear on HAOS 10.0, that's a good point to start with. On HAOS 9.XX never had issues, maybe it's a compatibility problem with HAOS 10.X or it's kernel.

Thanks a lot for the answer :+1:

lx2905, 2023-06-21:
> @lx2905 What hardware is z2m running on? maybe not enough power? what version of z2m?

It's running perfectly on that machine with proxmox. Home Assistant got 8gb RAM and 4 Kernels, never reaching the limits. 

Newest Update of that z2m stable version.

kirovilya, 2023-06-21:
@hardwareadictos @lx2905 I don't know what else it could be... I don't know what problem could be with HAOS 10 :((

lx2905, 2023-06-21:
Are there any debugs or logs that we could provide?

kirovilya, 2023-06-21:
@lx2905 https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging
you can record herdsman logs ... so that you can see what is happening before the error

lx2905, 2023-06-21:
Ok. I activated that rn. Posting the log as soon as it crashes the next time.

hardwareadictos, 2023-06-21:
> @lx2905 https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging you can record herdsman logs ... so that you can see what is happening before the error

Done also

lx2905, 2023-06-23:
Is it just for me that activating the logs fixed the issue? I mean it didnt crashed since then and thats 2 Days ago.

Hotelk52339, 2023-06-28:
same issue. But i can't even start z2m

Nilleftw, 2023-06-30:
> @hardwareadictos Greetings.
> 
> 1. I hope you do not have ZHA integration running in parallel, it must be disabled so that it does not try to connect to the USB port.

This fixed it for me. I don't know if some recent update had installed or re-enabled ZHA, but disabling it fixed the problem immediately. Thank you so much for pointing this out.

lx2905, 2023-06-30:
The Error did not show up for me again, after enabling logs. #close

hardwareadictos, 2023-08-03:
Updated to HAOS 10.4 and issue reappeared

closed 2023-09-10 by github-actions[bot]

xelemorf, 2023-10-28:
I have read that 7.1.1.0 FW might help, hopefully I'll have some time to test in the upcoming days.

- ncp-uart-sw_7.1.1.0_115200.gbl (259 KB) @ https://cloud.time-vortex.net/s/Jy2x6tj2GNx2QZn
- Web based flasher (use Chrome or Edge): https://darkxst.github.io/silabs-firmware-builder/
- If you are flashing using any other method you'll need the initfile aswell: https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/nvm3_initfile.gbl
- 7.1.1.0 FW mentioned @ https://github.com/darkxst/silabs-firmware-builder/issues/14#issuecomment-1676338682
- Additional workaround (janky but helps): https://community.home-assistant.io/t/tenacious-watchdog-for-add-ons-example-for-zigbee2mqtt/521888

Script based on above "workaround" article:
```
alias: Restart Zigbee2Mqtt
sequence:
  - if:
      - condition: state
        entity_id: binary_sensor.zigbee2mqtt_edge_running
        state: "on"
    then:
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/bridge/request/restart
          qos: "2"
    alias: If running then restart
  - if:
      - condition: or
        conditions:
          - condition: state
            entity_id: binary_sensor.zigbee2mqtt_edge_running
            state: "off"
          - condition: state
            entity_id: binary_sensor.zigbee2mqtt_edge_running
            state: unavailable
          - condition: state
            entity_id: binary_sensor.zigbee2mqtt_edge_running
            state: unknown
    then:
      - service: hassio.addon_start
        data:
          addon: 45df7312_zigbee2mqtt_edge
    alias: If not running then start
  - delay:
      hours: 0
      minutes: 30
      seconds: 0
      milliseconds: 0
  - service: homeassistant.update_entity
    data: {}
    target:
      entity_id: binary_sensor.zigbee2mqtt_edge_running
mode: single

```

jomaskm, 2023-11-09:
@xelemorf 
Did you get around to testing the firmware you mentioned in your last post?

The adapter/dongle is simply not behaving well on my installation. 
I'm thinking of going back to Conbee II. It was working a lot better.

xelemorf, 2023-11-09:
@jomaskm 
I have tested FW ncp-uart-sw_7.1.1.0_115200.gbl for a few days and although It was noticeably better I still had issues with my Zigbee mesh network. 
- devices straight up dissappearing from the network and inventory all together, 
- devices disconnecting from the nearest router then connecting to coordinator but with 0 LQI and not responding for more than a day becoming zombies
- manually repaired devices just straight up having issues every morning

I still believe HAOS 10.x is the issue, as previously I did not have any major issues with Dongle-E coordinators (I have 2 revisions of them and a total 3 piece).

I had enough of the multiple constant issues every day for 4 weeks so I ended up buying a Dongle-P directly from Sonoff on Aliexpress. Chose Ali and Sonoff because local sellers sent me Dongle-E instead of the ordered Dongle-P without them knowing the difference.

Dongle-P is working outstandingly fine, even the direct coordinator connections (with end devices)%are working as they should which was never the case with Dongle-E.

jomaskm, 2023-11-09:
Thank you for the detailed reply.

I went ahead and ordered the P-model.
I'm running HASS OS on top of Proxmox, so I got the backups covered.

Did you manage to migrate all your devices to the new dongle without setting up everything from scratch btw.?

xelemorf, 2023-11-09:
@jomaskm 

I knew ahead of time that Coordinator change will require repairing all devices, but I wanted a clean start so I made an excel speadsheet for a list of all devices and their location/room, stopped and removed Zigbee2Mqtt and Zigbee2Mqtt Edge completely, deleted all the files from "/config/zigbee2mqtt/", and started clean. New Z2M versions can now generate the security keys so I have put "GENERATE" for those, ref below.

- network_key: GENERATE
- ext_pan_id: GENERATE
- pan_id: _still needs to be manually defined_

I needed to also change the below values, make sure **rtscts** is set to false and you can toggle on the physical dip switch on the board.
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus__yourdeviceid_
  disable_led: true
  baudrate: 115200
  rtscts: false
  
For the firmware upgrade I have used the python method, the Texas Instrument provided flashing software cannot handle the latest firmwares. I'll provide a guideline for you on that process soon.

stire81, 2023-12-30:
Hi!
Had the same issue...
used Sonoff zigbee plus dongle before with ZHA for months (with only 1 zigbee device)
Than i bought 4 more zigbee devices (tuya), but it doesnt displayed the temperature & humidity data.
I deleted ZHA and set up the zigbee2mqtt.
After 1 day its running! :)
My settings:
[image]

changyeamoon, 2024-01-13:
I ended up trying all the adapter options until it worked! I used adapter: zstack to get it working!

All options in the documentation - https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter