Koenkk/zigbee2mqtt #17414: Switches fail after updating efr32 to 7.2.2.0 and Z2M to last-dev

juanchula, 2023-04-21:
### What happened?

My switches stop responding after a few hours. They do send state changes when triggered physically, but do not respond to any state changes made through the Z2M HA interface. This is happening with both Sonoff ZBMINI and Moes MS-104BZ. In addition to this, I'm not able to generate the network map. Everything works again when I restart zigbee2mqtt but it fails again after hours

This issue started occurring after I updated the coordinator firmware to https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_6.10.3_115200.gbl and then to https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.2.2.0_115200.gbl, and switched to the last-dev version of Z2Q from Dockerhub.

### What did you expect to happen?

That the switches update change their value correctly

### How to reproduce it (minimal and precise)

1) Installed Z2M 1.30.3
2) Flashed efr32mg21 with ncp-uart-sw_6.10.3_115200.gbl
3) Paired a couple of switches to the network
4) Stopped Z2M
5) Flashed efr32mg21 with ncp-uart-sw_7.2.2.0_115200.gbl and updated Z2M to 1.30.3-dev commit: 17947aa
6) Started Z2M and waited for a few hours
7) Tried changing the states of the switches.

### Zigbee2MQTT version

1.30.3-dev commit: 17947aa

### Adapter firmware version

7.2.2.0 build 190

### Adapter

EFR32MG21 SM-011 V1.0 EZSP v11

### Debug log

Debug 2023-04-21 12:33:16Received MQTT message on 'zigbee2mqtt/Living Exterior/set' with data '{"state":"ON"}'
Debug 2023-04-21 12:33:16Publishing 'set' 'state' to 'Living Exterior'
Debug 2023-04-21 12:33:34Received MQTT message on 'zigbee2mqtt/Living Interior Puerta Lateral/set' with data '{"state_l1":"ON"}'
Debug 2023-04-21 12:33:34Publishing 'set' 'state' to 'Living Interior Puerta Lateral'
Debug 2023-04-21 12:33:34Received MQTT message on 'zigbee2mqtt/Living Interior Puerta Lateral/set' with data '{"state_l2":"ON"}'
Debug 2023-04-21 12:33:34Publishing 'set' 'state' to 'Living Interior Puerta Lateral'
Debug 2023-04-21 12:33:48Received MQTT message on 'zigbee2mqtt/Living Exterior/set' with data '{"state":"ON"}'
Debug 2023-04-21 12:33:48Publishing 'set' 'state' to 'Living Exterior'

mamrai1, 2023-04-21:
I had similar problems but after repairing the devices without deleting them everything seems ok now.

juanchula, 2023-04-21:
@mamrai1 With repair, do you mean to pair them again by pressing the pair button of the switches?

mamrai1, 2023-04-22:
Yes and put the coordinator in pairing mode.. Best to select to pair to the nearest router

juanchula, 2023-04-23:
I solved it by going back to the stable version of Z2M

wastez, 2023-05-07:
Updated to 7.2.2 right now. 
Everything seems to be good until now.

wastez, 2023-05-07:
Everything is working as it should but after a while one device gets unavailable and i'm not able to scan the network....
Does anybody has also problems with 7.2.2?

wastez, 2023-05-07:
Switched back to 7.1.1 and the problem is gone....

juanchula, 2023-05-21:
I update the title of the issue because I confirmed that the problem is due to Z2M version 1.30.4. When I revert back to version 1.30.3, some of the devices that were experiencing the issue give me this error: 

> Publish 'set' 'state' to 'Living Interior Puerta Lateral' failed: 'Error: Command 0x385b44fffe332150/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
> Zigbee2MQTT:debug 2023-05-21 20:14:42: Error: Command 0x385b44fffe332150/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)
>     at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:464:19)
>     at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

 However, it corrects itself within a few seconds and does not fail again.

closed 2023-06-28 by github-actions[bot]