Koenkk/zigbee2mqtt #17418: Only one device appearing on mobile

Daniel15, 2023-04-22:
### What happened?

When I load Zigbee2MQTT's web UI on my phone (Samsung Galaxy S22), I only see one device:
[image]

This happens in both Firefox and Chrome, and still happens when "desktop mode" is enabled.

### What did you expect to happen?

I expect all devices to appear, like what I see on desktop:
[image]

### How to reproduce it (minimal and precise)

All I'm doing is loading the UI

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

20211217

### Adapter

Sonoff USB stick

### Debug log

N/A

Koenkk, 2023-04-22:
You've filtered the list with a search (`Exter`), removing it should show all devices again.

[image]

closed 2023-04-22 by Koenkk

Daniel15, 2023-04-22:
Hahaha wow, I can't believe I didn't notice that. I didn't realise it saved searches, and the saved searches could be different per device (I guess it's in localstorage?)

Sorry for wasting your time 😅

Koenkk, 2023-04-22:
No problem, enjoy! 😄