Koenkk/zigbee2mqtt #17420: IKEA LED1949C5 OTA Update fail

chris777c, 2023-04-22:
### What happened?

Hello,

When I would like update firmware of my IKEA LED1949C5, web interface show me this message : 
`Failed to check if update available for 'Ikea-tradfriBulb' (No image available for imageType '8708')`

Can you help me please ?

### What did you expect to happen?

Firmware version of my IKEA LED1949C5 is 1.0.032.

I can see on : http://fw.ota.homesmart.ikea.net/feed/version_info.json
`"fw_binary_url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.19.052/bin/10080506-zingo_kt_bulb_hwpwmcs_ws-1.1.003.ota.ota.signed",
	"fw_file_version_LSB": 4099,
	"fw_file_version_MSB": 1,
	"fw_filesize": 259294,
	"fw_image_type": 8708,
	"fw_manufacturer_id": 4476,
	"fw_type": 2`

so I think, I can update 1.0.032 to 1.1.003 but don't work.

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Debug log

`debug 2023-04-22 06:49:15: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Ikea-tradfriBulb2","transaction":"xcosh-6"}'
info  2023-04-22 06:49:15: Checking if update available for 'Ikea-tradfriBulb2'
debug 2023-04-22 06:49:15: Check if update available for '0x84ba20fffe38513b' (TRADFRIbulbE14WScandleopal470lm)
debug 2023-04-22 06:49:15: Using endpoint '1'
debug 2023-04-22 06:49:15: Received Zigbee message from 'Ikea-tradfriBulb2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65586,"imageType":8708,"manufact$
debug 2023-04-22 06:49:15: Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":8708,"fileVersion":65586}'
info  2023-04-22 06:49:15: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Ikea-tradfriBulb2"},"error":"Failed to check if update available for 'Ikea-t$
error 2023-04-22 06:49:15: Failed to check if update available for 'Ikea-tradfriBulb2' (No image available for imageType '8708')
debug 2023-04-22 06:49:15: Error: No image available for imageType '8708'
    at getImageMeta (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/tradfri.js:18:15)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at isNewImageAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:260:18)
    at Object.isUpdateAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:29)
    at OTAUpdate.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/otaUpdate.ts:203:45)
debug 2023-04-22 06:49:18: Received Zigbee message from 'Ikea-tradfriBulb2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65586,"imageType":8708,"manufact$
debug 2023-04-22 06:49:18: Device 'Ikea-tradfriBulb2' requested OTA
`

Koenkk, 2023-04-22:
Did you by any chance set `ikea_ota_use_test_url` to `true` in the z2m `configuration.yaml`?

chris777c, 2023-04-22:
Yes :
`ota:
  ikea_ota_use_test_url: true
  disable_automatic_update_check: true
`

MattWestb, 2023-04-22:
The test feed is dead for over 2 years and your light is also newer so its not in the test feed. Disabling it so the system is using the normal product feed and it shall being updated OK.

Koenkk, 2023-04-22:
Indeed, remove `ikea_ota_use_test_url` to use the feed you linked in OP

chris777c, 2023-04-23:
Thank you for reply but it still not work ( I tried test url because is write [here](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#using-the-ikea-tradfri-test-server) )

Koenkk, 2023-04-23:
Can you provide the debug log when doing the check?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

chris777c, 2023-04-23:
yes , here 
`debug 2023-04-23 09:57:35: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Ikea-tradfriBulb2","transaction":"331tf-8"}'
info  2023-04-23 09:57:35: Checking if update available for 'Ikea-tradfriBulb2'
debug 2023-04-23 09:57:35: Check if update available for '0x84ba20fffe38513b' (TRADFRIbulbE14WScandleopal470lm)
debug 2023-04-23 09:57:35: Using endpoint '1'
debug 2023-04-23 09:57:35: Received Zigbee message from 'Ikea-tradfriBulb2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65586,"imageType":8708,"manufact$
debug 2023-04-23 09:57:35: Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":8708,"fileVersion":65586}'
info  2023-04-23 09:57:36: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Ikea-tradfriBulb2"},"error":"Failed to check if update available for 'Ikea-t$
error 2023-04-23 09:57:36: Failed to check if update available for 'Ikea-tradfriBulb2' (No image available for imageType '8708')
debug 2023-04-23 09:57:36: Error: No image available for imageType '8708'
    at getImageMeta (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/tradfri.js:18:15)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at isNewImageAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:260:18)
    at Object.isUpdateAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:29)
    at OTAUpdate.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/otaUpdate.ts:203:45)
debug 2023-04-23 09:57:38: Received Zigbee message from 'Ikea-tradfriBulb2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65586,"imageType":8708,"manufact$
debug 2023-04-23 09:57:38: Device 'Ikea-tradfriBulb2' requested OTA
debug 2023-04-23 09:57:38: Responded to OTA request of 'Ikea-tradfriBulb2' with 'NO_IMAGE_AVAILABLE'`

Koenkk, 2023-04-23:
How are you running z2m? We need to add some extra logging statements to debug this

chris777c, 2023-04-23:
This method: https://www.zigbee2mqtt.io/guide/installation/01_linux.html

Koenkk, 2023-04-24:
Great, if you go to the z2m directory, open the file: `node_modules/zigbee-herdsman-converters/lib/ota/tradfri.js` now change `getImageMeta` as follows:

```js
async function getImageMeta(current, logger, device) {
    const url = useTestURL ? testURL : productionURL;
    const imageType = current.imageType;
    const images = (await axios.get(url)).data;
    logger.warn('URL ' + url); // <- add this
    logger.warn('ALL IMAGES ' + JSON.stringify(images)); // <- add this
    const image = images.find((i) => i.fw_image_type === imageType);

    if (!image) {
        throw new Error(`No image available for imageType '${imageType}'`);
    }

    return {
        fileVersion: (image.fw_file_version_MSB << 16) | image.fw_file_version_LSB,
        url: image.fw_binary_url,
    };
}
```

Then restart z2m and provide the z2m log file when checking for an update

chris777c, 2023-04-24:
here 

`Apr 24 19:03:37 [17343]: Zigbee2MQTT:info  2023-04-24 19:03:37: Checking if update available for 'Ikea-tradfriBulb2'
Apr 24 19:03:37 [17343]: Zigbee2MQTT:debug 2023-04-24 19:03:37: Check if update available for '0x84ba20fffe38513b' (TRADFRIbulbE14WScandleopal470lm)
Apr 24 19:03:37 [17343]: Zigbee2MQTT:debug 2023-04-24 19:03:37: Using endpoint '1'
Apr 24 19:03:37 [17343]: Zigbee2MQTT:debug 2023-04-24 19:03:37: Received Zigbee message from 'Ikea-tradfriBulb2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65586,"imageType":8708,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Apr 24 19:03:37 [17343]: Zigbee2MQTT:debug 2023-04-24 19:03:37: Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":8708,"fileVersion":65586}'
Apr 24 19:03:37 [17343]: Zigbee2MQTT:info  2023-04-24 19:03:37: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Ikea-tradfriBulb2"},"error":"Failed to check if update available for 'Ikea-tradfriBulb2' (No image available for imageType '8708')","status":"error","transaction":"331tf-18"}'
Apr 24 19:03:37 [17343]: Zigbee2MQTT:error 2023-04-24 19:03:37: Failed to check if update available for 'Ikea-tradfriBulb2' (No image available for imageType '8708')
Apr 24 19:03:37 [17343]: Zigbee2MQTT:debug 2023-04-24 19:03:37: Error: No image available for imageType '8708'
Apr 24 19:03:37 [17343]:     at getImageMeta (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/tradfri.js:18:15)
Apr 24 19:03:37 [17343]:     at runMicrotasks (<anonymous>)
Apr 24 19:03:37 [17343]:     at processTicksAndRejections (node:internal/process/task_queues:96:5)
Apr 24 19:03:37 [17343]:     at isNewImageAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:260:18)
Apr 24 19:03:37 [17343]:     at Object.isUpdateAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:29)
Apr 24 19:03:37 [17343]:     at OTAUpdate.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/otaUpdate.ts:203:45)
Apr 24 19:03:40 [17343]: Zigbee2MQTT:debug 2023-04-24 19:03:40: Received Zigbee message from 'Ikea-tradfriBulb2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65586,"imageType":8708,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Apr 24 19:03:40 [17343]: Zigbee2MQTT:debug 2023-04-24 19:03:40: Device 'Ikea-tradfriBulb2' requested OTA
Apr 24 19:03:41 [17343]: Zigbee2MQTT:debug 2023-04-24 19:03:41: Responded to OTA request of 'Ikea-tradfriBulb2' with 'NO_IMAGE_AVAILABLE'
Apr 24 19:03:48 [17343]: Zigbee2MQTT:debug 2023-04-24 19:03:48: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"error"}},"transaction":"331tf-19"}'
`

Koenkk, 2023-04-24:
- I don't see the added log messages, did you edit the correct file?
- Are you sure you are not using an external converter? (check for `external_converters` in your z2m `configuration.yaml`)

chris777c, 2023-04-25:
hello,

Yes i know , I had surprised to no see log me too but i have only on file tradfri.js on my distribution.

and my configuration.yaml , i have only this line for OTA :

`ota:
  disable_automatic_update_check: true
  ikea_ota_use_test_url: false
`

chris777c, 2023-04-25:
Finally it work, thank you for reply and sorry  to have disturbed you

closed 2023-04-25 by chris777c