Koenkk/zigbee2mqtt #17421: TS0121_TZ3000_rdtixbnu Power (Instantaneous measured power) stops reporting.

brunotfr, 2023-04-22:
### What happened?

As soon as I upgrade from version 1.28.4-1 to a newer one, the Power (Instantaneous measured power) of all plugs stops reporting.
Strangely the Energy (Sum of consumed energy) still keeps reporting.
The Addon perfectly runs with Out Of the Box settings with version 1.28.4-1
After a stop/start running version 1.30.3-1, Power does report for a short period of time an then stops.
Tried running it in debug, but there's no error in the log regarding this issue

### What did you expect to happen?

I expected the Power to keep reporting every minute.

### How to reproduce it (minimal and precise)

After upgrading to the laters version (1.30.3-1), stop/start the Addon.
Open the Zigbee2MQTT dashboard, open the Powerplug and switch on the Powerplug with (in my case) the washing machine plugged in.
In the dashboard you will see the current Power used by the washing machine.
As it runs through it's program you'll see the current Power change almost every minute.
After a period of time it stops changing, meaning the reporting stopped.
At this point the washing machine has not finished yet.
Only after a stop/start of the Addon, it wil start reporting again for a short period of time.

Running version 1.28.4-1 there is no issue.
The plug keeps reporting Power over and over, like expected.

### Zigbee2MQTT version

 1.30.3-1

### Adapter firmware version

20230411.1 - latest

### Adapter

Deconz Conbee II

PastCoder, 2023-04-22:
I have the same issue with Blitzwolf BW-SHP13 (TZ3000_g5xawfcq). It worked well in the past. But some time (I don't know excatly since when) it does noz update anymore (or only after a very long time - sometime there is an update after some hours).
I did not find out if it is always.

PastCoder, 2023-04-22:
Maybe it is related to this merge? https://github.com/Koenkk/zigbee-herdsman-converters/pull/5461

brunotfr, 2023-04-22:
My settings are exactly like in the screenshot of stokket.
Looks like it has become an out of the box setting.

With version 1.28.4-1 the reports promptly come every minute and continue to come.
Every version after 1.28, with the same settings, will report for a short period of time and then stop.
Weird thing is, it's only the power that stops reporting

Koenkk, 2023-04-22:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/fb4539bb570720f915774e28ef78a8bb

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

brunotfr, 2023-04-22:
Thanks for the swift respons.
Downloaded the file, renamed it to ext_converter_ts0121.js and added it to the configuration.yaml 

 external_converters:
  - ext_converter_tuya_battery.js
  - ext_converter_ts0601.js
  - ext_converter_ts0121.js

Next upgraded to version 1.30.3-1 and restarted z2m.
The Addon doesn't start with this error in the log:

[21:06:01] INFO: Preparing to start...
[21:06:01] INFO: Socat not enabled
[21:06:02] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:15
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, tuya.fz.power_outage_memory,
                 ^
ReferenceError: fz is not defined
    at /app/data/extension/externally-loaded.js:15:18
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:150:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:157:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:167:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

Tried to figure out where to set the reference, but did find it yet :-)

Koenkk, 2023-04-23:
Updated https://gist.github.com/Koenkk/fb4539bb570720f915774e28ef78a8bb, power outage memory and indicator mode support will be added back later.

brunotfr, 2023-04-23:
Did the same routine with the new file.
This time I get this error in the log during startup (version 1.30.3-1):

[image]

Koenkk, 2023-04-23:
Whooops, fixed with https://gist.github.com/Koenkk/fb4539bb570720f915774e28ef78a8bb

brunotfr, 2023-04-23:
Haha, no problem.
Same routine with the latest file and this time it did startup :-)
Will run some test today and get back to you asap.

brunotfr, 2023-04-23:
Looks like the problem is still there.
The washing machine started around 10:02 and reported every minute until around 10:15

[image]

The history showed an update/report every minute, now the latest update is an hour ago:

[image]

Koenkk, 2023-04-23:
Can you provide the debug log on 1.28.4 and 1.30.3 when letting z2m run for a few hours (and receive power measurements from the device)? (without using any external converter)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

brunotfr, 2023-04-23:
Sure, will try to run some tests on both versions tommorrow with debug enabled

brunotfr, 2023-04-24:
Attached the debug log files of both versions.
As the log file is rotating approximately every 10 minutes, I’ve downloaded several in a sequence and put them together in one file for each version.

Maybe I should rephrase the topic of the case to: 'It keeps reporting, but no longer the actual power consumption'.

For testing purposes I’ve connected an infrared heating panel to a plug called 'Wasmachine’.
So this is the object to look for in the logs.

In the ‘Version 1.30.3.log’ file the last actual power reading for 'Wasmachine' was 409 and then it froze:

[image]

After 30 minutes in this state I switched over to version 1.28.4
As you can see in the ‘Version 1.28.4.log’ file, even after a period of time the actual Power consumption of 'Wasmachine’ is still reported correctly.

[Version 1.28.4.log](https://github.com/Koenkk/zigbee2mqtt/files/11310676/Version.1.28.4.log)
[Version 1.30.3.log](https://github.com/Koenkk/zigbee2mqtt/files/11310677/Version.1.30.3.log)

Koenkk, 2023-04-24:
Could you check with the following external converter (on 1.30.3): https://gist.github.com/Koenkk/b6e8a5ead526d5c1bc737013c1676e96

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m
- let it run for some time until power measurements freeze
- provide me the STDOUT logging (not the log file), you should see messages like `SET TIMER`

brunotfr, 2023-04-24:
Sure, downloaded the file and will run a test tomorrow.

Koenkk, 2023-04-27:
@brunotfr I see, that's strange indeed. Please provide the log once it stops working again.

brunotfr, 2023-04-30:
Issue is still there.
Today when using it for the first time again it stopped working.
At 10:52 the power was 67 and it stil is at this moment.

This is version 1.30.3 with the external_converter.

[image]

[image]

Attached the STDOUT
The IEEE address you are looking for is 0x60a423fffe621677

[zigbee-herdsman.log](https://github.com/Koenkk/zigbee2mqtt/files/11360977/zigbee-herdsman.log)

martinusgithubius, 2023-05-05:
I have similar problems, with TS011F  (_TZ3000_gvn91tmx), z2m version 1.30.4-1. The values of 'power' and 'current' remain unchanged after a certain period, especially when the device has been turned off. The recorded values during the 'on' period are still being displayed.

Koenkk, 2023-05-06:
Thanks, I think I found & fixed the issue.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

brunotfr, 2023-05-07:
Jus stopped z2m and installed z2m Edge next to it.
Edited the configuration yaml and it started.
However after a couple of minutes the add-on stopped with this error:

[08:28:07] INFO: [32mPreparing to start...[0m
[08:28:08] INFO: [32mSocat not enabled[0m
[08:28:09] INFO: [32mStarting Zigbee2MQTT...[0m
Zigbee2MQTT:warn  2023-05-07 08:28:14: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-05-07 08:28:14: Allowing new devices to join.
Zigbee2MQTT:warn  2023-05-07 08:28:14: Set `permit_join` to `false` once you joined all devices.
Error: APS TIMEOUT
    at Driver.processApsBusyQueue (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:721:28)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:161:56)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)

Koenkk, 2023-05-07:
@brunotfr could you give me the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

brunotfr, 2023-05-07:
Followed the same steps and added herdsman debug.
After 7-10 minutes the devices became unavailable again (the add-on showed that is was still running).
Grabbing the log appeared to be not possible as the container was no longer available (forgot to copy the ID upfront).

For the next attempt I've rebooted everything and at the moment it's still running in Edge and herdsman debug after 35 minutes.
Also the power is still reporting every minute.
No clue why it crashed the first two times, but rebooting may have taken care of it.
Will keep an eye on it the next couple of hours.

brunotfr, 2023-05-07:
Until now z2m Edge is still working as expected (with herdsman debug on).
Three rounds of the washing machine and one of the tumble dryer where reported neatly.
Looks like you found and fixed the issue :-)

brunotfr, 2023-06-02:
@Koenkk The dev branch is still running fine.
Did not see this fix in the current releases.
Is it possible to add it to the next release?

Koenkk, 2023-06-02:
@brunotfr I expect it is, can you check with z2m 1.31.0?

brunotfr, 2023-06-02:
Sure, will give it a try and let you know.

brunotfr, 2023-06-04:
Checked and it's fixed in z2m version 1.31.0-1

closed 2023-06-04 by brunotfr