Koenkk/zigbee2mqtt #17423: [New device support]:

mdevaal, 2023-04-22:
### Link

https://www.philips-hue.com/nl-nl/p/hue-white-and-color-ambiance-resonate-down-muurlamp/8719514466777

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0x001788010cbb0a95","nwkAddr":44196,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"929003553001","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210305","swBuildId":"1.82.10","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1682172950030,"defaultSendRequestWhen":"immediate"}

### Comments

Just purchased an Philips hue resonate downward wall light but for this model there seems no support yet. That why I am reaching out en hope you can help me.

Not sure which attribute reflects Supported color modes, happy to check that if someone could point the right attribute.

*) could not find English product page for this specific Philips hue light bulb.

### Color temperature range

'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

Koenkk, 2023-04-22:
Added! You can use the following external converter until the next release:

```js
const philips = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['929003553001'],
    model: '929003553001',
    vendor: 'Philips',
    description: 'Hue outdoor Resonate downward wall lamp',
    extend: philips.extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

closed 2023-04-22 by Koenkk

mdevaal, 2023-04-22:
Confirmed working!
Really happy with your support @koenk, you just made my day (or maybe even week).

Thank you so much,
Martin

Koenkk, 2023-04-23:
Graag gedaan! 😄