Koenkk/zigbee2mqtt #17424: [New device support]: Zigbee Model TS0202 Zigbee Manufacturer _TZ3040_bb6xaihh

fredqm, 2023-04-22:
### Link

https://www.aliexpress.com/item/1005005099780623.html?spm=a2g0o.order_list.order_list_main.78.71f7caa4ODwHtj&gatewayAdapt=glo2bra

### Database entry

{"id":22,"type":"EndDevice","ieeeAddr":"0xa4c13892b7f9a3ff","nwkAddr":52943,"manufId":4417,"manufName":"_TZ3040_bb6xaihh","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,4,1280,0],"outClusterList":[4,6,4096,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024c23a5d","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1682168862967,"defaultSendRequestWhen":"immediate"}

### Comments

I'm a noob.

Koenkk, 2023-04-22:
`_TZ3040_bb6xaihh` will be supported in the 1 May Zigbee2MQTT release.

closed 2023-04-22 by Koenkk

posreg, 2023-05-10:
mind me asking, this device has been paired successsfully, however TS0202_1 points to different picture / model than the linked product

[https://www.zigbee2mqtt.io/devices/TS0202_1.html](url)
versus
[https://www.aliexpress.com/item/1005005099780623.html?spm=a2g0o.order_list.order_list_main.78.71f7caa4ODwHtj&gatewayAdapt=glo2bra](url)

is this expected behaviour ?

thanks!

Koenkk, 2023-05-10:
@posreg please fill in https://github.com/Koenkk/zigbee2mqtt/issues/new?assignees=&labels=wrong+device&projects=&template=wrong_device.yaml&title=%5BWrong+device%5D%3A+