Koenkk/zigbee2mqtt #17432: Hue Dimmer Switch 324131092621 not reporting action

bendamat, 2023-04-23:
### What happened?

I sucessfully included a Hue Dimmer Switch (324131092621) into my network. The device is correctly recognised. 
I have a sensor device topic with the following sub-topics:
* `zigbee2mqtt-ad/sensor/0x0017880104f544b0/battery`
* `zigbee2mqtt-ad/sensor/0x0017880104f544b0/action_duration`
* `zigbee2mqtt-ad/sensor/0x0017880104f544b0/linkquality`
* `zigbee2mqtt-ad/sensor/0x0017880104f544b0/update_state`
But I'm missing an `action` topic.
Instead, I see the different actions as topics under `device_automation`: 
* `zigbee2mqtt-ad/device_automation/0x0017880104f544b0/action_on_press`
* `zigbee2mqtt-ad/device_automation/0x0017880104f544b0/action_on_press_release`
* `zigbee2mqtt-ad/device_automation/0x0017880104f544b0/action_off_press`
and so on...

### What did you expect to happen?

I expected to see an action topic under the sensor device topic. 

### How to reproduce it (minimal and precise)

Include a Hue Dimmer Switch.
Press any button.  

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZB DONGLE PLUS E

### Debug log

Zigbee2MQTT:debug 2023-04-23 14:03:38: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":"Coordinator","time":254,"transaction":"bcd70-2","value":true}'
Zigbee2MQTT:info  2023-04-23 14:03:38: Zigbee: allowing new devices to join via Coordinator.
Zigbee2MQTT:info  2023-04-23 14:03:38: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"device":"Coordinator","time":254,"value":true},"status":"ok","transaction":"bcd70-2"}'
Zigbee2MQTT:info  2023-04-23 14:03:45: Device '0x0017880104f544b0' joined
Zigbee2MQTT:info  2023-04-23 14:03:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880104f544b0","ieee_address":"0x0017880104f544b0"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-04-23 14:03:45: MQTT publish: topic 'zigbee2mqtt-ad/sensor/0x0017880104f544b0/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880104f544b0"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880104f544b0","sw_version":"6.1.1.28573"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x0017880104f544b0 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x0017880104f544b0","unique_id":"0x0017880104f544b0_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2023-04-23 14:03:45: MQTT publish: topic 'zigbee2mqtt-ad/sensor/0x0017880104f544b0/action_duration/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880104f544b0"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880104f544b0","sw_version":"6.1.1.28573"},"enabled_by_default":true,"name":"0x0017880104f544b0 action duration","state_topic":"zigbee2mqtt/0x0017880104f544b0","unique_id":"0x0017880104f544b0_action_duration_zigbee2mqtt","unit_of_measurement":"second","value_template":"{{ value_json.action_duration }}"}'
Zigbee2MQTT:info  2023-04-23 14:03:45: MQTT publish: topic 'zigbee2mqtt-ad/sensor/0x0017880104f544b0/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880104f544b0"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880104f544b0","sw_version":"6.1.1.28573"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x0017880104f544b0 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x0017880104f544b0","unique_id":"0x0017880104f544b0_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2023-04-23 14:03:45: MQTT publish: topic 'zigbee2mqtt-ad/sensor/0x0017880104f544b0/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880104f544b0"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880104f544b0","sw_version":"6.1.1.28573"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"0x0017880104f544b0 update state","state_topic":"zigbee2mqtt/0x0017880104f544b0","unique_id":"0x0017880104f544b0_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Zigbee2MQTT:info  2023-04-23 14:03:45: MQTT publish: topic 'zigbee2mqtt-ad/binary_sensor/0x0017880104f544b0/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880104f544b0"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880104f544b0","sw_version":"6.1.1.28573"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","name":"0x0017880104f544b0 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x0017880104f544b0","unique_id":"0x0017880104f544b0_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
Zigbee2MQTT:info  2023-04-23 14:03:45: MQTT publish: topic 'zigbee2mqtt-ad/update/0x0017880104f544b0/update/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/bridge/request/device/ota_update/update","device":{"identifiers":["zigbee2mqtt_0x0017880104f544b0"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880104f544b0","sw_version":"6.1.1.28573"},"device_class":"firmware","entity_picture":"https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png","latest_version_template":"{{ value_json['update']['latest_version'] }}","latest_version_topic":"zigbee2mqtt/0x0017880104f544b0","name":"0x0017880104f544b0","payload_install":"{\"id\": \"0x0017880104f544b0\"}","state_topic":"zigbee2mqtt/0x0017880104f544b0","unique_id":"0x0017880104f544b0_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'
Zigbee2MQTT:info  2023-04-23 14:03:45: Starting interview of '0x0017880104f544b0'
Zigbee2MQTT:info  2023-04-23 14:03:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880104f544b0","ieee_address":"0x0017880104f544b0","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:debug 2023-04-23 14:03:45: Received MQTT message on 'zigbee2mqtt-ad/sensor/0x0017880104f544b0/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880104f544b0"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880104f544b0","sw_version":"6.1.1.28573"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x0017880104f544b0 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x0017880104f544b0","unique_id":"0x0017880104f544b0_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:debug 2023-04-23 14:03:45: Received MQTT message on 'zigbee2mqtt-ad/sensor/0x0017880104f544b0/action_duration/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880104f544b0"],"manufacturer":"Philips","model":"Hue dimmer switch (324131092621)","name":"0x0017880104f544b0","sw_version":"6.1.1.28573"},"enabled_by_default":true,"name":"0x0017880104f544b0 action duration","state_topic":"zigbee2mqtt/0x0017880104f544b0","unique_id":"0x0017880104f544b0_action_duration_zigbee2mqtt","unit_of_measurement":"second","value_template":"{{ value_json.action_duration }}"}'
Zigbee2MQTT:debug 2023-04-23 14:03:45: Received MQTT message on 'zigbee2mqtt-ad/sensor/0x0017880104f544b0/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0017880104f544b0"],"manufacturer":"Philips","model":"Hue d [... cut ...]
Zigbee2MQTT:info  2023-04-23 14:03:57: MQTT publish: topic 'zigbee2mqtt/0x0017880104f544b0/action', payload 'on_press'
Zigbee2MQTT:debug 2023-04-23 14:03:57: Received Zigbee message from '0x0017880104f544b0', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-04-23 14:03:57: Received Zigbee message from '0x0017880104f544b0', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":8,"type":1,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID 0

bendamat, 2023-05-12:
Update: I just updated to v1.30.4 but the issue persists..

scyllaL, 2023-05-16:
Same for me as well, I just needed to migrate zigbee2mqtt from one machine to another via docker. However I just recognized now, that the automation in Homeassistant which is (or was) triggered by on_press no longer works. In Homeassistant I even do no langer see on_press as an action, the only actions I see are on_hold, on_hold_released and the same for off. However in the web frontend of zigbee2mqtt i clearly see the action on_press.

bendamat, 2023-05-19:
Thanks! I tried that but didn't have any luck. But I've managed to make it work by temporarily turn on the legacy options for the Home Assistant integration. Since then, I have the correct AD topics.

RezzZ, 2023-05-24:
same here, just started with a blueprint for hue dimmer switches in HA but it requires an action sensor.. Hope this can be fixed. Ticking legacy to true didnt help (yet)

RezzZ, 2023-06-24:
i use a different ha integration called switch manager (HACS) which works fine with the dimmer switches now. so no more issue regarding the absence of the action sensor.

closed 2023-08-02 by github-actions[bot]