Koenkk/zigbee2mqtt #17435: New Moes Thermostat

Lazze85, 2023-04-23:
### Link

https://it.aliexpress.com/item/1005005202931454.html?spm=a2g0o.productlist.main.17.44885c19cVX2AD&algo_pvid=af4a488f-d680-4148-b99b-957bbd750b87&algo_exp_id=af4a488f-d680-4148-b99b-957bbd750b87-8&pdp_npi=3%40dis%21EUR%2167.26%2127.74%21%21%21%21%21%402100bfe316822530504695447d0703%2112000032224629524%21sea%21IT%210&curPageLogUid=iekBh2YjfToF7

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0xa4c1380481607da2","nwkAddr":23047,"manufId":4417,"manufName":"_TZE204_aoclfnxz","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"� �+i� �+i� �+i� �+i� �+i� �+i� �+i� �+i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1682262343356,"defaultSendRequestWhen":"immediate"}

### Comments

L'unica cosa che non sono riuscito a fare è l'impostazione automatica della data e l'ora al momento della connessione.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const exposesLocal = {
    hour: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('h').withValueMin(0).withValueMax(23),
    minute: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('m').withValueMin(0).withValueMax(59),
    program_temperature: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('°C')
        .withValueMin(5).withValueMax(35).withValueStep(0.5),
}

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'WHT-006-GC',
    vendor: 'Moes',
    description: 'Moes Thermostat',
    fromZigbee: [fz.moes_thermostat],
    toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
        tz.moes_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
        tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit, tz.moes_thermostat_min_temperature_limit,
        tz.moes_thermostat_program_schedule],
    exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(), e.min_temperature_limit(),
        exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
            .withLocalTemperature(ea.STATE).withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
            .withSystemMode(['off', 'heat'], ea.STATE_SET).withRunningState(['idle', 'heat', 'cool'], ea.STATE)
            .withPreset(['hold', 'program']),
        e.temperature_sensor_select(['IN', 'AL', 'OU']),
        exposes.composite('program', 'program', ea.STATE_SET).withDescription('Time of day and setpoint to use when in program mode')
            .withFeature(exposesLocal.hour('weekdays_p1_hour'))
            .withFeature(exposesLocal.minute('weekdays_p1_minute'))
            .withFeature(exposesLocal.program_temperature('weekdays_p1_temperature'))
            .withFeature(exposesLocal.hour('weekdays_p2_hour'))
            .withFeature(exposesLocal.minute('weekdays_p2_minute'))
            .withFeature(exposesLocal.program_temperature('weekdays_p2_temperature'))
            .withFeature(exposesLocal.hour('weekdays_p3_hour'))
            .withFeature(exposesLocal.minute('weekdays_p3_minute'))
            .withFeature(exposesLocal.program_temperature('weekdays_p3_temperature'))
            .withFeature(exposesLocal.hour('weekdays_p4_hour'))
            .withFeature(exposesLocal.minute('weekdays_p4_minute'))
            .withFeature(exposesLocal.program_temperature('weekdays_p4_temperature'))
            .withFeature(exposesLocal.hour('saturday_p1_hour'))
            .withFeature(exposesLocal.minute('saturday_p1_minute'))
            .withFeature(exposesLocal.program_temperature('saturday_p1_temperature'))
            .withFeature(exposesLocal.hour('saturday_p2_hour'))
            .withFeature(exposesLocal.minute('saturday_p2_minute'))
            .withFeature(exposesLocal.program_temperature('saturday_p2_temperature'))
            .withFeature(exposesLocal.hour('saturday_p3_hour'))
            .withFeature(exposesLocal.minute('saturday_p3_minute'))
            .withFeature(exposesLocal.program_temperature('saturday_p3_temperature'))
            .withFeature(exposesLocal.hour('saturday_p4_hour'))
            .withFeature(exposesLocal.minute('saturday_p4_minute'))
            .withFeature(exposesLocal.program_temperature('saturday_p4_temperature'))
            .withFeature(exposesLocal.hour('sunday_p1_hour'))
            .withFeature(exposesLocal.minute('sunday_p1_minute'))
            .withFeature(exposesLocal.program_temperature('sunday_p1_temperature'))
            .withFeature(exposesLocal.hour('sunday_p2_hour'))
            .withFeature(exposesLocal.minute('sunday_p2_minute'))
            .withFeature(exposesLocal.program_temperature('sunday_p2_temperature'))
            .withFeature(exposesLocal.hour('sunday_p3_hour'))
            .withFeature(exposesLocal.minute('sunday_p3_minute'))
            .withFeature(exposesLocal.program_temperature('sunday_p3_temperature'))
            .withFeature(exposesLocal.hour('sunday_p4_hour'))
            .withFeature(exposesLocal.minute('sunday_p4_minute'))
            .withFeature(exposesLocal.program_temperature('sunday_p4_temperature')),
    ]
};

module.exports = definition;
```

closed 2023-06-02 by github-actions[bot]

EstebanSteffovski, 2023-06-14:
same thing for me. _TZE204_aoclfnxz

Nikolay-Ch, 2023-06-22:
Same problem...

alfieso, 2023-11-03:
Hi @Lazze85, I am using your extension with thermostat model BHT-006GAZB. I realised that when setting the target temperature in the thermostat the pace is 0.5º but I think this extension just take this value as integer. Could be possible to support 1 decimal for current_heating_setpoint?
Thanks!

PulsarFX, 2023-12-06:
have a look at #19991 if this of any use for your model