Koenkk/zigbee2mqtt #17436: [New device support]: AliExpress Roller Blind Cover (Tuya, Zigbee) C-Type (3 Button)

acbee, 2023-04-23:
### Link

https://www.aliexpress.com/item/1005002954586484.html?spm=a2g0o.order_list.order_list_main.34.25771802eEeu0j

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x187a3efffe01db9d","nwkAddr":27572,"manufId":4098,"manufName":"_TZE200_cpbo62rn","powerSource":"Battery","modelId":"TS0601","epList":[1],"endp

### Comments

I've managed to get the STATE (open/close/stop) working, and am reading a number of other DPs but can't determine what they are. 

My biggest issue is that I can't figure out a way to calibrate / set the lower and upper limits / set a run time so open/close only moves the motor for 2 seconds.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_cpbo62rn' },],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Roller Blinds Motor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
    ],
    meta: {
        tuyaDatapoints: [
            [1,  'state',       tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
            [3,  'position',    tuya.valueConverter.raw],
            [7,  'dp007',       tuya.valueConverter.raw],
            [12, 'motor_fault', tuya.valueConverter.trueFalse1],
            [13, 'battery',     tuya.valueConverter.raw],
            [101, 'dp101',      tuya.valueConverter.raw],
            [102, 'dp102',      tuya.valueConverter.raw],
            [103, 'dp103',      tuya.valueConverter.raw],
            [104, 'dp104',      tuya.valueConverter.raw],
            [105, 'dp105',      tuya.valueConverter.raw],
            [106, 'dp106',      tuya.valueConverter.raw],
            [108, 'dp108',      tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

wraeuk, 2023-04-24:
I just recieved these and I'm just starting to play around with getting them working.

With any luck, the dps 103, 104, 105 are the limit settings
https://github.com/Koenkk/zigbee2mqtt/issues/11164#issuecomment-1086580037

acbee, 2023-04-24:
I added tz.tuya_data_point_test to the toZigbee section of my custom file, reloaded, and was able to successfully issue commands such as "bool,103,0" and "bool,105,0" which I would have expected to clear the limits as suggested by @nilvanis. But pressing the up or down buttons on the physical device only spins the motor for two seconds, suggesting they were not cleared?

piwko28, 2023-05-05:
Any luck with the resets? I fight with the same boss.

wraeuk, 2023-05-05:
I went and grabbed a gateway from AliExpress and it finally turned up. The DPs according to Tuya are (I've added data where I can see it).

1 = Control (open,close,stop)
2 = Curtain position setting (0-100)
3 = Current curtain position (0-100)
4 = Mode (?)
101 = Curtain closed initial position (left,right)
102 = Start learning ("start" - bool?)
7 = Work State (standby, success)
103 = Current Temperature (2.40摄氏度)
104 = Current Luminosity (0-100)
105 = Set Temperature (50摄氏度)
106 = Set Luminosity (?)
107 = Restore iniital state (?)
12 = Fault (0)
108 = to bind (AAAAAAAAAA==)
13 = battery (0-100)
109 =  update timing information (?)

Will need to spend some time working this out, but in the meantime attacking 102 and maybe 107 should yield some results

acbee, 2023-05-07:
I also received a Tuya gateway/hub this week and picked up a CC2351 USB device which I flashed to sniff the Zigbee traffic via Wireshark. After setting the device up with the Tuya gateway/hub and Smart Life app. and capturing the packets, I re-paired it with w/Z2M and using the dev. console was able to replicate the app. setup procedure. 

NOTE: Because my motor is mounted on the left side of my blind, everything is reverse what the app. states, so when the app. says to fully close the blind before starting, I fully open. 

Position the blind in the starting position manually, with the motor disconnected, then connect the motor and issue in this sequence: 

RESET LIMITS, START STUDY, wait until the blind is as open/closed as you like, STOP STUDY

Dev. Console Values:

Cluster: 0xef00
Command: 0

RESET LIMITS
{
    "dpValues": [{
            "dp": 107,
            "datatype": 2,
            "data": {
                "data":[0],
                "type":"Buffer"
            }
    }],
    "seq":1
}

START STUDY (Limit Setting)
{
    "dpValues": [{
            "dp": 102,
            "datatype": 2,
            "data": {
                "data":[0],
                "type":"Buffer"
            }
    }],
    "seq":1
}

STOP STUDY (Limit Setting)
{
    "dpValues": [{
            "dp": 102,
            "datatype": 2,
            "data": {
                "data":[1],
                "type":"Buffer"
            }
    }],
    "seq":1
}

OPEN
{
    "dpValues": [{
            "dp": 1,
            "datatype": 2,
            "data": {
                "data":[0],
                "type":"Buffer"
            }
    }],
    "seq":1
}

STOP
{
    "dpValues": [{
            "dp": 1,
            "datatype": 2,
            "data": {
                "data":[1],
                "type":"Buffer"
            }
    }],
    "seq":1
}

CLOSE
{
    "dpValues": [{
            "dp": 1,
            "datatype": 2,
            "data": {
                "data":[2],
                "type":"Buffer"
            }
    }],
    "seq":1
}

piwko28, 2023-05-11:
@wraeuk thanks man! It really works.

I attach custom converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_cpbo62rn' },],
    model: 'LY-108',
    vendor: 'TuYa',
    description: 'Cover',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        exposes.text('work_state', ea.STATE),
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.battery(),
        exposes.enum('opening_mode', ea.STATE_SET, ['tilt', 'lift']).withDescription('Opening mode'),
        exposes.enum('motor_direction', ea.STATE_SET, ['left', 'right']).withDescription('Motor side'),
        exposes.enum('set_upper_limit', ea.STATE_SET, ['start', 'stop']).withDescription('Learning'),
        exposes.enum('factory_reset', ea.STATE_SET, ['SET']).withDescription('Remove limits'),
    ],
    whiteLabel: [
        tuya.whitelabel('Tuya', 'LY-108', 'Cover', ['_TZE200_cpbo62rn']),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
            [2, 'position', tuya.valueConverter.coverPosition],
            [3, 'position', tuya.valueConverter.raw],
            [4, 'opening_mode', tuya.valueConverterBasic.lookup({'tilt': tuya.enum(0), 'lift': tuya.enum(1)})],
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'learning': tuya.enum(2)})],
            [13, 'battery', tuya.valueConverter.raw],
            [101, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],
            [107, 'factory_reset', tuya.valueConverter.setLimit],
        ],
    },
};

module.exports = definition;
```

piwko28, 2023-05-11:
I've created a pull request to add support for this device globally:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5738

Andrew-Joakimsen, 2023-05-23:
It kind of works but one open causes it to close and close causes it to open, and it reports the wrong state. How do I reverse it? Changing motor direction doesn't appear to do anything.

Andrew-Joakimsen, 2023-05-31:
@piwko28 What is the proper procedure to use this to set limit? The option in Z2M set_upper_limit does not do anything.

I have to send the manual commands, but then the open/close state is not accurate.

Andrew-Joakimsen, 2023-05-31:
Actually in my case set_upper_limit  = STOP starts the learn process where it CLOSES the shades from open position. It starts to close the shade. 

Then I can end the learn procedure by clicking START. 

Once learn is complete it shows they are open despite being closed.

Basically everything is reversed.

Andrew-Joakimsen, 2023-06-03:
motor_direction is reversed. When the motor is installed on the left side of the blind it needs to be set to motor_direction = "right"

set_upper_limit is also reversed. I need to click "stop" to start the process and "start" when it reaches the open position (or rather a split second before)

@piwko28 This was the cause of all the issues and confusion with my use of this device & the custom converter above.

mateuszdrab, 2023-06-17:
I just got one of those devices today and can't figure out how to teach it the proper open and close position. 

@Andrew-Joakimsen  Would you be able to share the steps if you got yours working?

khoinguyen, 2023-06-21:
@piwko28 Thank you for the code!

I do some minor changes which it fix some bugs

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_cpbo62rn' },],
    model: 'LY-108',
    vendor: 'TuYa',
    description: 'Cover',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        exposes.text('work_state', ea.STATE),
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.battery(),
        exposes.enum('opening_mode', ea.STATE_SET, ['tilt', 'lift']).withDescription('Opening mode'),
        exposes.enum('motor_direction', ea.STATE_SET, ['left', 'right']).withDescription('Motor side'),
        exposes.enum('set_upper_limit', ea.STATE_SET, ['start', 'stop']).withDescription('Learning'),
        exposes.enum('factory_reset', ea.STATE_SET, ['SET']).withDescription('Remove limits'),
    ],
    whiteLabel: [
        tuya.whitelabel('Tuya', 'LY-108', 'Cover', ['_TZE200_cpbo62rn']),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(0), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(2)})],
            [2, 'position', tuya.valueConverter.coverPosition],
            [3, 'position', tuya.valueConverter.raw],
            [4, 'opening_mode', tuya.valueConverterBasic.lookup({'tilt': tuya.enum(0), 'lift': tuya.enum(1)})],
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'ready': tuya.enum(2)})],
            [13, 'battery', tuya.valueConverter.raw],
            [101, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(0), 'stop': tuya.enum(1)})],
            [107, 'factory_reset', tuya.valueConverter.setLimit],
        ],
    },
};

module.exports = definition;
```

Basically, I change 
- the `work_state` value `learning` to `ready` as after I successfully `set_upper_limit` it always shows `learning`, before that it is either `standby` or `success` which maybe something else I did not figure out yet
- Swap the enum of CLOSE and OPEN, it works for my case, not really sure if it is correct for everyone.

The step to train:
- factory_reset
- reload page to make sure every control in right state
- manually close the blind cover (bottom position)
- click Start of `set_upper_limit`, the device should run to open the blind, if wrong you will need to start over from factory_reset (this works for me, maybe some easier approach but I can't sure)
- Until the blind fully open, click Stop for `set_upper_limit`. Now the blind are fully learned and ready to go.
- Click Close to close the blind
- Click Open to open the blind
- to Set position, click Stop then set the position. Set position while Close/Open will not work, the motor just run a little bit then stop

mateuszdrab, 2023-06-29:
I still can't get it to work despite the instructions above.
I've been reading comments about this [item ](https://www.aliexpress.com/item/1005002954586484.html?spm=a2g0o.order_list.order_list_main.34.25771802eEeu0j) and users say that they needed to calibrate top upper and lower position, but I don't see that option. The unit I have is [this one](https://www.aliexpress.com/item/1005005574510278.html?spm=a2g0o.order_list.order_list_main.17.42d718024kPAdt)

Are they not the same? They seem to be.

Andrew-Joakimsen, 2023-06-29:
For me everything is opposite

 

To get it working correctly I set motor direction: left (when the motor is on the right side of the window)

 

Then I click set_upper_limit = stop to start the limit process. If it is working correctly you start from closed position, and it should begin to open. Right before it reaches your desired open point I click on set_upper_limit = start to end the procedure.

mateuszdrab, 2023-06-29:
I couldn't get it to work with the UI buttons, I tried so many times; however, with the dev console I got it working first time... Well, I don't mind a flimsy learning process as long as the states reflect properly (which currently do not). 

I think this will be fixed though by @khoinguyen

mateuszdrab, 2023-07-30:
Any chance we can merge the changes to resolve the everything is reversed issue?

drashish87, 2023-08-05:
Can anyone please advise who to increase the blind upper valve? I can get it to reset the upperlimit! 
I have tried above post but not sure what and where exactly do include that code!
Using Z2MQTT with sky connect

drashish87, 2023-08-05:
> I also received a Tuya gateway/hub this week and picked up a CC2351 USB device which I flashed to sniff the Zigbee traffic via Wireshark. After setting the device up with the Tuya gateway/hub and Smart Life app. and capturing the packets, I re-paired it with w/Z2M and using the dev. console was able to replicate the app. setup procedure.
> 
> NOTE: Because my motor is mounted on the left side of my blind, everything is reverse what the app. states, so when the app. says to fully close the blind before starting, I fully open.
> 
> Position the blind in the starting position manually, with the motor disconnected, then connect the motor and issue in this sequence:
> 
> RESET LIMITS, START STUDY, wait until the blind is as open/closed as you like, STOP STUDY
> 
> Dev. Console Values:
> 
> Cluster: 0xef00 Command: 0
> 
> ```ts
> RESET LIMITS
> {
>     "dpValues": [{
>             "dp": 107,
>             "datatype": 2,
>             "data": {
>                 "data":[0],
>                 "type":"Buffer"
>             }
>     }],
>     "seq":1
> }
> 
> START STUDY (Limit Setting)
> {
>     "dpValues": [{
>             "dp": 102,
>             "datatype": 2,
>             "data": {
>                 "data":[0],
>                 "type":"Buffer"
>             }
>     }],
>     "seq":1
> }
> 
> STOP STUDY (Limit Setting)
> {
>     "dpValues": [{
>             "dp": 102,
>             "datatype": 2,
>             "data": {
>                 "data":[1],
>                 "type":"Buffer"
>             }
>     }],
>     "seq":1
> }
> 
> OPEN
> {
>     "dpValues": [{
>             "dp": 1,
>             "datatype": 2,
>             "data": {
>                 "data":[0],
>                 "type":"Buffer"
>             }
>     }],
>     "seq":1
> }
> 
> STOP
> {
>     "dpValues": [{
>             "dp": 1,
>             "datatype": 2,
>             "data": {
>                 "data":[1],
>                 "type":"Buffer"
>             }
>     }],
>     "seq":1
> }
> 
> CLOSE
> {
>     "dpValues": [{
>             "dp": 1,
>             "datatype": 2,
>             "data": {
>                 "data":[2],
>                 "type":"Buffer"
>             }
>     }],
>     "seq":1
> }
> ```

Thank you for posting the code, searched for hours before finding this

mateuszdrab, 2023-10-07:
Any chance to get this looked at?

Koenkk, 2023-10-07:
This device should be supported in z2m 1.33.1 (`_TZE200_cpbo62rn`), what is not working yet? Setting the upper limit should be possible: https://www.zigbee2mqtt.io/devices/TS0601_cover_6.html#set-upper-limit-enum

mateuszdrab, 2023-10-07:
> This device should be supported in z2m 1.33.1 (`_TZE200_cpbo62rn`), what is not working yet? Setting the upper limit should be possible: https://www.zigbee2mqtt.io/devices/TS0601_cover_6.html#set-upper-limit-enum

My issue is with the fact that states are inverted and so weird things happen when you control it via home assistant. I think it is something to do with opened and closed state being mixed up.

I had no luck calibrating with z2m provided controls and has to use the manual ZigBee commands provided above

Andrew-Joakimsen, 2023-10-07:
Have you tried to set motor direction inverted?  To get it working correctly I set motor direction: left when the motor is on the right side of the window.

Koenkk, 2023-10-08:
@mateuszdrab inverted the states:

Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/8ab33577e8839581f91889c20dc1080a

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

mateuszdrab, 2023-10-17:
Just tried it. For me, it seems to make things worse. Blind is open when it says it is closed.
1 percent open is at the very end, where as 99% open is barely open.

mateuszdrab, 2023-10-17:
I'm not sure now if the external converer does anything. I recorded the interaction on the screen with the converter added and without it.
Starts as fully closed with the bar on the right. Clicking what I thought would be 10% (1/10th towards the left going from the right) results in the blind opening nearly fully. Then further steps to the left reduce it's opening until hitting the 0 on the very left results in the blind closing and the bar jumping to the right again.

https://github.com/Koenkk/zigbee2mqtt/assets/26556541/bf757eef-f3ef-4560-978d-2d7d7b139684

https://github.com/Koenkk/zigbee2mqtt/assets/26556541/66dd9bdd-d96f-44c8-b295-4005e8dc26ba

Koenkk, 2023-10-18:
- Can you check with https://gist.github.com/Koenkk/8ab33577e8839581f91889c20dc1080a
- If it doesn't change anything, can you confirm that the ext converter is loaded correctly? In the z2m frontend -> device ->  description should be `Cover motor CUSTOM`

mateuszdrab, 2023-10-19:
Thank you Koen

There was a little typo in your gist which prevented Z2M from starting so I fixed it.

So... with the code you provided, the percentages were inversed and that was a good step forward; however, it was still saying closed when open and vice versa so I reversed those enums and it now works fine. Right is closed, bar moves nicely to the left and percentage of opening increases until it goes to 100% where it says open.

My code is below:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const positionInverted = {
        to: async (v, meta) => {
            return 100 - v;
        },
        from: (v, meta, options) => {
            return 100 - v;
        },
    };

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_cpbo62rn'},
    ],
    model: 'TS0601_cover_6',
    vendor: 'TuYa',
    description: 'Cover motor CUSTOM',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.text('work_state', ea.STATE),
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.battery(),
        e.enum('opening_mode', ea.STATE_SET, ['tilt', 'lift']).withDescription('Opening mode'),
        e.enum('motor_direction', ea.STATE_SET, ['left', 'right']).withDescription('Motor side'),
        e.enum('set_upper_limit', ea.STATE_SET, ['start', 'stop']).withDescription('Learning'),
        e.enum('factory_reset', ea.STATE_SET, ['SET']).withDescription('Remove limits'),
    ],
    whiteLabel: [
        tuya.whitelabel('TuYa', 'LY-108', 'Cover', ['_TZE200_cpbo62rn']),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(2), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(0)})],
            [2, 'position', positionInverted],
            [3, 'position', positionInverted],
            [4, 'opening_mode', tuya.valueConverterBasic.lookup({'tilt': tuya.enum(0), 'lift': tuya.enum(1)})],
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'learning': tuya.enum(2)})],
            [13, 'battery', tuya.valueConverter.raw],
            [101, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],
            [107, 'factory_reset', tuya.valueConverter.setLimit],
        ],
    },
};

module.exports = definition;
```

With that said, for some reason the device description does not say CUSTOM, it's empty - but it definitely works.

[image]

drashish87, 2023-10-20:
I am running Zigbee2MQTT and MQTT on LXI proxmox and HA on proxmox too.

I am getting this error on saving the external_converter to configuration.yaml, any advise?
[image]

mateuszdrab, 2023-10-20:
> I am running Zigbee2MQTT and MQTT on LXI proxmox and HA on proxmox too.
> 
> I am getting this error on saving the external_converter to configuration.yaml, any advise?
> [image]
> 

This entry goes into the `configuration.yaml` of z2m which is located in `/config/zigbee2mqtt`

Koenkk, 2023-10-20:
@mateuszdrab great! Integrated the changes

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

> P.S: regarding the battery, it's always been saying 100% and I've had it for a few months now. I don't believe it - how can we check if this is an accurate reading? I have a weird feeling this is not reporting the battery correctly because I have 3 of them and they all say 100%.

100 is the value reported by the device, so I expect it to be correct. But to confirm someone has to check if datapoint 13 is indeed the battery one (https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)

mateuszdrab, 2023-10-20:
> @mateuszdrab great! Integrated the changes
>
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)
>
>
> 100 is the [...]

Awesome, thanks for that. I hope that it makes it better for other users and that I wasn't some edge case with it not working right.

Regarding the battery... I might have to look at it but I don't have the tuya gateway. 

I guess the right way to trial and error this would be to get all the data from the device daily and compare if anything changes over time 🤣

Koenkk, 2023-11-28:
@mateuszdrab I guess we are using the wrong data point or the firmware is bugged

- Confirm the correct battery % is reported with the TuYa hub
- If yes, find the [correct datapoint](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)

mateuszdrab, 2023-11-28:
> @mateuszdrab I guess we are using the wrong data point or the firmware is bugged
> 
> - Confirm the correct battery % is reported with the TuYa hub
> - If yes, find the [correct datapoint](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)

Ouch, I don't have a Tuya hub...

Koenkk, 2023-11-28:
For any non-mapped datapoints, you will see `Datapoint XX not defined for XX with value XX`

mateuszdrab, 2024-01-08:
I've been observing the datapoints over the past 3 weeks and none of them seem to indicate battery status.
One of my units died recently and was fully recharged but nothing seems to indicate that.

[image]

closed 2024-08-09 by github-actions[bot]

sammcj, 2025-01-30:
Has anyone managed to figure out how to read the battery status of these?