Koenkk/zigbee2mqtt #17438: TuYa TS0601_cover_1 (_TZE200_cowvfni3) motor_speed doesn't work anymore

ASNNetworks, 2023-04-23:
### What happened?

I reset my curtains due to some changes, but for some reason the motor_speed option in Z2M doesn't do anything anymore. I used to have it at its lowest speed and because of the reset it's much faster. Setting the speed to any other number, doesn't have any effect however. 

It does send the message in logs. but nothing changes. I also tried using a manual mqtt publish command, which also logs it as changed, but it has no effect.

### What did you expect to happen?

Allow the motor_speed change to work, like it used to do.

### How to reproduce it (minimal and precise)

I am not sure when this issue has been introduced. It worked for over 2 years, then I decided to reset the motor and now I the motor_speed settings doesn't do anything. I have two of these devices and both ignore the option (while the 2nd one was never reset).

### Zigbee2MQTT version

1.30.3 

### Adapter firmware version

20221226

### Adapter

Sonoff P

### Debug log

```
Debug 2023-04-23 20:08:16Received MQTT message on 'zigbee2mqtt/Slaapkamer Gordijnen/set' with data '{"options": {"motor_speed": 0}}'
Debug 2023-04-23 20:08:16Publishing 'set' 'options' to 'Slaapkamer Gordijnen'
```

Doesn't really tell much

ASNNetworks, 2023-05-24:
Still open

closed 2023-07-02 by github-actions[bot]

LaurentiuGrama, 2023-07-17:
I have the same issue with the device. The speed command doesn't work and the same i don't know how to reset/ start/end possitions