Koenkk/zigbee2mqtt #17439: Device support request - TuYa Smart Air Sensor _TZE200_mja3fuja

jmluxem, 2023-04-23:
### Link

https://a.aliexpress.com/_ms7pX0y

### Database entry

{“id”:105,”type” : ”Router”, “ieeeAddr”:”0xa4c138d4fe33f3b9”,”nwkAddr”:27178,”manufId”:4417,”manufName”:”_TZE200_mja3fuja”,

### Comments

Tried setting up this device and configuring an external converter but haven’t gotten anything working. 

### External converter

```shell
Nothing usable…

I was trying to setup a converter to make this report the same as another very similar device that I already have and that is supported: _TZE200_dwcarsat - the TuYa TS0601_smart_air_house_keeper
```

jmluxem, 2023-05-07:
This device seems to be identical to _TZE200_dwcarsat 

I have still been unsuccessful trying to get _TZE200_mja3fuja to mimic function of _TZE200_dwcarsat

Charliedean, 2023-05-20:
Also having this issue. @jmluxem did you mange to fix this?

oedo808, 2023-05-31:
Me too!  First zigbee device joined but no expected entities :-)

jmluxem, 2023-06-02:
I have not been able to fix this. Haven't had much free time to devote. Was hoping @Koenkk might be able to sprinkle some of his magic, since I don't believe any real datapoint-finding is required - I'd bet that a direct conversion from _TZE200_dwcarsat would allow this to function, but making that work properly escapes me. The unsupported device is _TZE200_mja3fuja, but I bought it thinking that it was identical to _TZE200_dwcarsat - they both have the same functions. I may be wrong but I don't exactly have the knowledge/time/hardware to test my belief.

Koenkk, 2023-06-02:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/002a6a797922f736db47fd4bb2f5f19c

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

jmluxem, 2023-06-03:
Heck yeah, that's why you're the boss :) 
That converter got temperature and humidity reporting correctly - confirmed by breathing into the sensor.
CO2, VOC, and Formaldehyde are reporting values, but tough for me to confirm validity/correctness since it seems the sensor updates those at a much slower rate, and it may not be very accurate to boot. PM25 not reporting any value currently.
Regardless, this is leagues better than anything I had gotten working. Thank you!

State values:
{
    "humidity": 43,
    "linkquality": 163,
    "temperature": 28.3,
    "co2": 13,
    "formaldehyd": 514,
    "pm25": null,
    "voc": 72
}

Please let me know if there are any other logs or items I can pull for your benefit.

Koenkk, 2023-06-03:
Can you try the following, in your `data/database.db` change the manufacturer name from `_TZE200_mja3fuja ` to `_TZE200_dwcarsat` and see if everything works now? (no external converter needed)

jmluxem, 2023-06-03:
I run HAOS with z2m as an addon. I am not seeing a 'data' directory. In my file structure, I have config/zigbee2mqtt/database.db

Within that database.db file, I changed the name to _TZE200_dwcarsat and restarted z2m, but it doesn't appear to be keeping the change - when i go back to the database.db file, the old _TZE200_mja3fuja name is back where I had just changed it.

Koenkk, 2023-06-04:
You need to stop z2m, make the change and start z2m.

jmluxem, 2023-06-05:
Thank you for that tip/correction. I updated the database.db file and now the device is working without the converter. Everything appears to be working

Koenkk, 2023-06-05:
Great, integrated the changes!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-05 by Koenkk