Koenkk/zigbee2mqtt #17450: Herdsman Hangs with CC2531

Tux12Fun, 2023-04-24:
### What happened?

`
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - stateChangeInd - {"state":8} +511ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,69,192,8,140] +511ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,69,192,8,140] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 2 - 5 - 192 - [8] - 140 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - stateChangeInd - {"state":8} +511ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
Zigbee2MQTT:error 2023-04-24 19:46:56: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-04-24 19:46:56: Failed to start zigbee
Zigbee2MQTT:error 2023-04-24 19:46:56: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-04-24 19:46:56: Exiting...
Zigbee2MQTT:error 2023-04-24 19:46:56: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/home/ui_zigbee/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.30.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.30.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ui_zigbee/.npm/_logs/2023-04-24T17_46_57_005Z-debug.log
`

### What did you expect to happen?

Zigbee2MQTT should start normay

### How to reproduce it (minimal and precise)

Updated Firmware of my CC2531 Adapter via Raspi GPIO

`
root@raspberrypi:/home/pi/zigbee/flash# ./cc_write -c4 -r3 -d5 -m90 3/CC2531ZNP-Prod.hex
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page 128/128.
 flash OK.
`

Started zigbee2mqtt with Debug Option for Handsman

### Zigbee2MQTT version

1.30.3 (commit #e1ecbd9)

### Adapter firmware version

20211115 (zstack12) used (Z-Stack-firmware/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip)

### Adapter

CC2531 ID=b524

### Debug log

Debug Log
`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.18
3 info using node@v14.21.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.30.3~prestart: zigbee2mqtt@1.30.3
[... 468 lines cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,69,192,8,140] +511ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,69,192,8,140] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 2 - 5 - 192 - [8] - 140 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - stateChangeInd - {"state":8} +511ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
Zigbee2MQTT:error 2023-04-24 19:46:56: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-04-24 19:46:56: Failed to start zigbee
Zigbee2MQTT:error 2023-04-24 19:46:56: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-04-24 19:46:56: Exiting...
Zigbee2MQTT:error 2023-04-24 19:46:56: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/home/ui_zigbee/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.30.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.30.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ui_zigbee/.npm/_logs/2023-04-24T17_46_57_005Z-debug.log
`

Tux12Fun, 2023-04-24:
For me it seams that it can talk to the adapter and the firmware should be correct as it responds with the correct Date but at the point of 

```
zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - stateChangeInd - {"state":8} +511ms
```

it seams that something is going wrong.

tonbor, 2023-04-26:
Same problem here any help?:
> zigbee2mqtt@1.30.3 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/node_modules/path-scurry/dist/cjs/index.js:352
    #resolveParts(dirParts) {
                 ^

SyntaxError: Unexpected token '('
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/rimraf/node_modules/glob/dist/cjs/glob.js:5:23)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.30.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.30.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2023-04-26T12_30_31_363Z-debug.log

tonbor, 2023-04-27:
Solved, update node version 12 to 16, made brand new simple configuration  yaml file and zigbee2mqtt is running again

Tux12Fun, 2023-05-02:
Solved this issue with updating coordinator

closed 2023-05-02 by Tux12Fun