Koenkk/zigbee2mqtt #17456: TS011F_plug_1 current/voltage not shown in home assistant

ict-one-nl, 2023-04-25:
### What happened?

[image]

Values are unavailable over MQTT though, autodiscovery issue?

### What did you expect to happen?

Values shown

### How to reproduce it (minimal and precise)

Paired the plug, was succesful

### Zigbee2MQTT version

 1.30.3 commit: 24c6b2e

### Adapter firmware version

20221226

### Adapter

1352

PattaFeuFeu, 2023-05-21:
It would still help to get a general default without any additional settings but [this Home Assistant forum entry](https://community.home-assistant.io/t/z2m-entities-disabled-by-integration-is-it-possible-to-enable-them-via-a-glob/548156/2) helped me.

In essence, add/update `device_options` with the following options to your ZigBee2Mqtt `configuration.yaml`:

```yaml
device_options:
  homeassistant:
    current:
      enabled_by_default: true
    power:
      enabled_by_default: true
    voltage:
      enabled_by_default: true
```

If you want the other values as well, add them in the same fashion but use `linkquality`, `update_state`, or `update_available` instead.

Not sure what exactly is necessary after saving the configuration file, but just for good measure, I restarted Mosquitto (my MQTT broker), Home Assistant, and ZigBee2Mqtt. It can still take a while until the values are reported. But after some minutes, I saw these values on Home Assistant.

ict-one-nl, 2023-05-23:
Hm that would solve it for now, but in general I would say that if the data is available and published to the mqtt bus, it should also be discovered by homeassistant

ict-one-nl, 2023-05-23:
Ah thanks!