Koenkk/zigbee2mqtt #17465: [New device support]: TS0601 _TZE204_lsanae15 DAC2161C

gresnyer, 2023-04-26:
### Link

https://www.aliexpress.com/item/1005004399475951.html?spm=a2g0o.order_list.order_list_main.9.61471802H7oMZf

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0xa4c138402b76e3dc","nwkAddr":64106,"manufId":4417,"manufName":"_TZE204_lsanae15","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"ނ�+i���+i��+i��+i���+f���+\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c16b28","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1682529333691,"defaultSendRequestWhen":"immediate"}

### Comments

I strongly believe that this device is exactly the same like TS0601 _TZE200_lsanae15 DAC2161C, only the "Zigbee Manufacturer" id changed to _TZE204_lsanae15, might this is a new revision. However I don't know how can I compare them.
I would like to ask for help to add official device support for this version. 

### External converter

```shell
I found external converter which working (not 100%, but working):

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const fzLocal = {
    tuya_xocadinrail_switch: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            for (const dpValue of msg.data.dpValues) {
                const value = tuya.getDataValue(dpValue);
                const dp = dpValue.dp;
                meta.logger.info(`RECEIVED DP #${dp} -- VALUE = ${value}`);
                
                switch (dp) {
                    case 16: // DPID that we added to common
                       return {   state: value ? 'ON' : 'OFF'};
                       
                      
                    case 1:
                        return {
                            energy: value / 100
                        };
                    case 6:   
                     return {
                            current: ((value[4] | value[3] << 8) / 1000), voltage: ((value[1] | value[0] << 8) / 10), power: ((value[7] | value[6] << 8))
                        };
                        
                    case 17:
                      const lookup = {
                       0: 'not set',
                       1: 'Over current threshold',
                       3: 'Over voltage threshold',
                                      };
                        return {
                            threshold_1: lookup[value[0]],
                            threshold1_value : (value[3] | value[2] << 8),
                            threshold_2: lookup[value[4]],
                            threshold2_value : (value[7] | value[6] << 8),                            
                    };    
                        
                      case 10:{
                          const lookup = {
                          0: 'clear',
                          1: 'over current threshold',
                          2: 'over power threshold',
                          4: 'over voltage threshold',
                          8: 'wrong frequency threshold',
                           };
                           return { alarm : lookup[value]} ;
                    }
                    case 18:{
                        meta.logger.warn(`zigbee-herdsman-converters:: NOT RECOGNIZED DP ` +
                            `#${dp} with data ${JSON.stringify(msg.data)} VALUE = ${value}`);
                            break;
                    }
                    default: {
                        meta.logger.warn(`zigbee-herdsman-converters:: NOT RECOGNIZED DP ` +
                            `#${dp} with data ${JSON.stringify(msg.data)} VALUE = ${value}`);
                            break;
                    }
                }
            }
        },
    },
};

const tzLocal = {
    state: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, 16, value === 'ON');
        },
    },
};

const definition = {
    fingerprint: [{
        modelID: 'TS0601',
        manufacturerName: '_TZE204_lsanae15'
    }],
    model: 'DAC2161C',
    vendor: 'Tuya',
    extend: extend.switch(),
    description: 'DIN Rail Smart Energy Meter',
    fromZigbee: [fzLocal.tuya_xocadinrail_switch, ],
    toZigbee: [tzLocal.state],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.switch().setAccess('state', ea.STATE_SET), 
              e.voltage(), 
              e.power(), 
              e.current(), 
              e.energy(),
              exposes.text('threshold_1', ea.STATE),
              exposes.text('threshold1_value', ea.STATE),
              exposes.text('threshold_2', ea.STATE),
              exposes.text('threshold2_value', ea.STATE),
              exposes.text('alarm', ea.STATE),

              ] 

};

module.exports = definition;
```

Koenkk, 2023-04-27:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/5305768c71b9b2b2f393cc7dc0c71420

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

gresnyer, 2023-04-27:
Thanks for the quick response. I replaced my converter with yours and all look good. I will be happy to see this included in your next public release and then get rid of external converter.

Thanks again... :)

Koenkk, 2023-04-27:
Done, thanks!

closed 2023-04-27 by Koenkk

MBRetrofit, 2023-11-21:
Let me thank you too,

The main problem for this device is the delay for reporting the power, is between 2 and 5 minute completly useless if is used in any automation. 

Also the Threshold 1 value is always 80A in HA.