Koenkk/zigbee2mqtt #17466: Zigbee2Mqtt fails with Exit Code 1 Connection not initialized after running for hours

artem-zinnatullin, 2023-04-26:
### What happened?

I'm noticing Zigbee2Mqtt `1.30.3` fails after hours of trouble-free run-time with `exit code 1` and this error in log:

```js
Zigbee2MQTT:info  2023-04-26 09:13:15: MQTT publish: topic 'zigbee2mqtt/Washing Machine', payload '{"ac_frequency":60,"current":0.04,"energy":1.78,"linkquality":148,"power":0.3,"power_factor":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":123.6}'
Zigbee2MQTT:info  2023-04-26 09:13:15: MQTT publish: topic 'zigbee2mqtt/Washing Machine', payload '{"ac_frequency":60,"current":0.04,"energy":1.78,"linkquality":152,"power":0.3,"power_factor":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":123.6}'
Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:540:19)
    at Driver.addTransientLinkKey (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:679:26)
    at Driver.preJoining (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:611:35)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:237:29)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:42)
    at /app/node_modules/zigbee-herdsman/src/utils/queue.ts:21:18
    at new Promise (<anonymous>)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:19:16)
    at EZSPAdapter.permitJoin (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:234:34)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:251:36)
```

On previous release `1.30.2` I was seeing crashes with exit code 255 and no useful log, but now it has stacktrace.

I'm running Zigbee2Mqtt in container with following key details:

```yaml
          resources:
            requests:
              cpu: "0.25"
              memory: "768Mi"
            limits:
              cpu: "4"
              memory: "768Mi"
              # Exposed through https://github.com/squat/generic-device-plugin
              squat.ai/zigbee: 1

            - --device
            - '{"name": "zigbee", "groups": [{"paths": [{"path": "/dev/ttyACM0"}]}]}'
```

### What did you expect to happen?

No crashes

### How to reproduce it (minimal and precise)

Just have it running in my case and it will fail eventually.

Happy to provide more info if I can!

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Stick (ZBDongle-E)

artem-zinnatullin, 2023-04-27:
Another log and exit code 1, happened after I've set some configuration options on Light Switch

```js
Zigbee2MQTT:error 2023-04-27 12:00:50: Publish 'set' 'state' to 'Office Main Light Switch' failed: 'Error: Command 0x847127fffe239beb/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":50}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Zigbee2MQTT:info  2023-04-27 12:00:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Office Main Light Switch' failed: 'Error: Command 0x847127fffe239beb/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":0,\"transtime\":50}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (sendZclFrameToEndpointInternal error)'","meta":{"friendly_name":"Office Main Light Switch"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2023-04-27 12:00:55: Publish 'set' 'state' to 'Living Room O-O-O Lamp' failed: 'Error: Command 0x282c02bfffeb2988/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Zigbee2MQTT:info  2023-04-27 12:00:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Living Room O-O-O Lamp' failed: 'Error: Command 0x282c02bfffeb2988/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (sendZclFrameToEndpointInternal error)'","meta":{"friendly_name":"Living Room O-O-O Lamp"},"type":"zigbee_publish_error"}'
Error: {"frameId":"stackStatusHandler","sequence":null} after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
```

flowcool, 2023-05-02:
Something similar here, multiple times a day:

```
2023-05-02T15:41:22.302693896Z Error: Reset error: Error: {"sequence":-1} after 10000ms
2023-05-02T15:41:22.341550833Z     at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:299:23
2023-05-02T15:41:22.341700932Z     at runNextTicks (node:internal/process/task_queues:60:5)
2023-05-02T15:41:22.341786523Z     at listOnTimeout (node:internal/timers:538:9)
2023-05-02T15:41:22.341838435Z     at processTimers (node:internal/timers:512:7)
2023-05-02T15:41:22.341914460Z     at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
2023-05-02T15:41:33.363486657Z Using '/app/data' as data directory
2023-05-02T15:42:23.279920674Z Error: {"frameId":"stackStatusHandler","sequence":null} after 10000ms
2023-05-02T15:42:23.280616050Z     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
2023-05-02T15:42:23.280744612Z     at listOnTimeout (node:internal/timers:569:17)
2023-05-02T15:42:23.280824683Z     at processTimers (node:internal/timers:512:7)
2023-05-02T15:42:29.803227996Z Using '/app/data' as data directory
```

Running into docker for a while without any issue until now.
Version de Zigbee2MQTT: [1.30.4]  commit: [b2dd21e]

thanasis00, 2023-05-02:
Almost same here, zigbee2mqtt was working perfectly, until suddenly I received a similar error:

```
Using '/app/data' as data directory
Zigbee2MQTT:info  2023-05-03 00:19:58: Logging to console and directory: '/app/data/log/2023-05-03.00-19-58' filename: log.txt
Zigbee2MQTT:info  2023-05-03 00:19:58: Starting Zigbee2MQTT version 1.30.2 (commit #cdf62ea)
Zigbee2MQTT:info  2023-05-03 00:19:58: Starting zigbee-herdsman (0.14.96)
Zigbee2MQTT:error 2023-05-03 00:20:19: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-05-03 00:20:19: Failed to start zigbee
Zigbee2MQTT:error 2023-05-03 00:20:19: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-05-03 00:20:19: Exiting...
Zigbee2MQTT:error 2023-05-03 00:20:19: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:540:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:338:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```
I'm using zbbridge with ezsp for adapter.

Everything has been working fine up until now.

artem-zinnatullin, 2023-05-03:
Update: 

1.30.4 is successfully running w/o restarts for 42 hours for me, no config or hardware changes on my end, didn't reboot the machine either.

```js
zigbee2mqtt-55d4ff9c66-frlwd      1/1     Running   0          42h
```

Closing for now!

artem-zinnatullin, 2023-05-09:
Have to reopen it, got 2 restarts in few days with same error

```js
Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Driver.addTransientLinkKey (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:679:26)
    at Driver.preJoining (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:611:35)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:237:29)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:42)
    at /app/node_modules/zigbee-herdsman/src/utils/queue.ts:21:18
    at new Promise (<anonymous>)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:19:16)
    at EZSPAdapter.permitJoin (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:234:34)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:251:36)
```

christiantroldmand, 2023-05-29:
Hi i am getting the same issue now after running without issues for months. Any fix to this?

Zigbee2MQTT:error 2023-05-29 00:17:12: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

Manu1312Woe, 2023-06-11:
I had the same problem for months and installed many different versions of zigbee2mqtt. But nothing fixed it.

Then I updated the zigbee-stick to 
"Coordinator-Typ EZSP v11"
"Coordinator-Version 7.2.2.0 build 190"
and it seems to fix the problem.

Since one week, there hasn't been a crash anymore.

Manu1312Woe, 2023-06-14:
This night it stopped working again with the same error..

Zigbee2MQTT:info  2023-06-14 08:26:02: Starting Zigbee2MQTT version 1.31.0 (commit #f440c86)
Zigbee2MQTT:info  2023-06-14 08:26:02: Starting zigbee-herdsman (0.14.117)
Zigbee2MQTT:error 2023-06-14 08:26:22: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-06-14 08:26:22: Failed to start zigbee
Zigbee2MQTT:error 2023-06-14 08:26:22: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-06-14 08:26:22: Exiting...
Zigbee2MQTT:error 2023-06-14 08:26:22: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
Using '/app/data' as data directory
[WARN  tini (7)] Tini is not running as PID 1 and isn't registered as a child subreaper.
Zombie processes will not be re-parented to Tini, so zombie reaping won't work.
To fix the problem, use the -s option or set the environment variable TINI_SUBREAPER to register Tini as a child subreaper, or run Tini as PID 1.

GitComfy, 2023-06-29:
Hoping to tag on to the end of this one...after an update of proxmox to version 8 my Sonoff USB30 dongle has stopped working with this error in the log...

Error: Reset error: Error: {"sequence":-1} after 10000ms
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:299:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

then if i go to the plug in i get 502: Bad Gateway

Ive been through a regconfig walk through with the same error. Wondered if anyone has seen this...

Hannes5612, 2023-06-29:
I switched to a deconz stick, now no problems at all for 2 weeks. I suppose the Sonoff stick just crashes after some time (god knows why it worked for ove a year)

GitComfy, 2023-06-29:
Ive got a Deconz stick running as well....the zigbee/sonoff setup was for devices that wouldnt connect to the DeConz....

Hannes5612, 2023-06-29:
Mh, im using the conbee II in combination with zigbee2mqtt. Only configuration I had to change was the stick model from `ezsp` to `deconz`, zigbee network continued to work like before

GitComfy, 2023-06-29:
Right - before i go any further im pretty rubbish at this...ive managed to get it setup and running (about 9 months ago) and kinda left it so now its not working again im almost kinda back to square one, as it were.

Im using ZHA for the DeConz config and was using the Sonoff for Zagbee2MQTT....as i say they were working together fine up to the point i upgraded to Proxmox 8....i get an 502:  Bad Gateway.

christiantroldmand, 2023-06-29:
Hi all. I managed to fix it. I had the USB mapped with a complete path Including device ID, after an HA update it does not read the value correctly on my RBP4.

Try and edit settings like below: 

[image]

That should fix it.

GitComfy, 2023-06-29:
ok - where do i find the value for "adapter"?

christiantroldmand, 2023-06-29:
> ok - where do i find the value for "adapter"?

Go to Settings -> Add Ons -> Zigbee2mqtt -> configuration

GitComfy, 2023-06-29:
err...yep - thats where i am but ive got no adapter in there...! so i need some sort of value to put, in there?

AlejandroGil, 2023-07-01:
> err...yep - thats where i am but ive got no adapter in there...! so i need some sort of value to put, in there? as in, you must have got the value for yours!

Settings > System > Hardware > All hardware > select your dongle and check path, mine is  `/dev/ttyACM0` as well

Manu1312Woe, 2023-07-12:
> I switched to a deconz stick, now no problems at all for 2 weeks. I suppose the Sonoff stick just crashes after some time (god knows why it worked for ove a year)

I decided to switch back to the Sonoff Dongle P and everything works fine since then. The Sonoff Donge E is only a repeater in my network, now.

SerinitySW, 2023-07-21:
I'm also seeing a similar error:

```
Zigbee2MQTT:info  2023-07-21 10:43:49: Starting zigbee-herdsman (0.16.0)
Zigbee2MQTT:error 2023-07-21 10:44:09: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-07-21 10:44:09: Failed to start zigbee
Zigbee2MQTT:error 2023-07-21 10:44:09: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-07-21 10:44:09: Exiting...
Zigbee2MQTT:error 2023-07-21 10:44:09: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

This happened out of the blue. I've tried updating my dongle's firmware to no avail. Made sure everything was up to date really. It just stopped working overnight.

christiantroldmand, 2023-07-21:
Did you try the fix suggested?

Settings > System > Hardware > All hardware > select your dongle and check path, mine is /dev/ttyACM0 as well

Mvh

Christian Klint

GitComfy, 2023-07-21:
Yes, mines on /USB though so doesnt really apply (to a certain extent!) - my conbee is on acm....

SerinitySW, 2023-07-21:
My dongle does not show up under all hardware

SerinitySW, 2023-07-22:
Solved my issue by unchecking and rechecking the device passthrough in unRAID.

peterdotjs, 2023-07-22:
Had the same issue and it ended up being related to docker. Running "service docker restart" resolved it. All other attempts of restarting/recreating the container didn't work.

GitComfy, 2023-07-22:
Im using proxmox v8. I have tried disconnecting and reconnecting the Sonoff - no dice.

gdschut, 2023-08-14:
I downgraded to 1.29.2 to solve this.

GitComfy, 2023-08-14:
Hi there. Well, i took it on the chin and rebuild the HA id say all working again...but its a little hit or miss....ive got a couple of sonoff dongles....one works one doesnt..!

closed 2023-09-22 by github-actions[bot]

jrgleason, 2023-09-24:
Similar issue so what was the answer? I see

```
Zigbee2MQTT:error 2023-09-24 05:02:15: Failed to start zigbee
Zigbee2MQTT:error 2023-09-24 05:02:15: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-09-24 05:02:15: Exiting...
Zigbee2MQTT:error 2023-09-24 05:02:15: Error: Connection not initialized
    at Ezsp.execCommand (/home/pi/Code/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Ezsp.version (/home/pi/Code/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
    at Driver.startup (/home/pi/Code/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/home/pi/Code/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/home/pi/Code/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/home/pi/Code/zigbee2mqtt/lib/zigbee.ts:60:27)
    at Controller.start (/home/pi/Code/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/home/pi/Code/zigbee2mqtt/index.js:107:5)
```
    
What does

"Settings > System > Hardware > All hardware > select your dongle and check path, mine is /dev/ttyACM0 as well"

Mean if I am on Raspberry pi? Also side note I have a Zwave controller running on /dev/ttyACM0. The Zigbee runs on /dev/ttyACM1 which is what I have in my config now.

jrgleason, 2023-09-24:
NM 

serial:
  port: /dev/ttyACM1
  adapter: ezsp
  
in data/configuration.yml worked

Manu1312Woe, 2023-09-24:
For me, the solution was to throw away the Sonoff-Dongle-E and use the Dongle-P.

Since months, there hasn't been any error anymore.

jrgleason, 2023-09-27:
Mine started having the same issue again, now as 

Error: Reset error: Error: {"sequence":-1} after 10000ms
    at /home/pi/Code/zigbee2mqtt-bkp/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:299:23
    at Queue.executeNext (/home/pi/Code/zigbee2mqtt-bkp/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

GitComfy, 2023-09-28:
Rebuilding the entire HA for me did the trick that being said it didnt work straight away.....

TimBroddin, 2023-11-30:
I got the same stacktrace as the post author. I'm thinking it has something to do with a device trying to join the network. 

I had permit join checked, so devices could always join. Now testing with that setting turned off.

torablien, 2023-12-16:
On the Sonoff ZBDongle-E, I am getting the same issue:

```
Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:140:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:172:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

My configuration.yaml has:
```
serial:
  port: /dev/ttyACM0
  adapter: ezsp 
```

Both zigbee2mqtt and HomeAssistant are running in Docker containers and I am able to use this dongle with ZHA.  I am also able to use z2m with by conbee/raspbee, so it seems like something between z2m and ZBDongle-E. Do people have z2m working with a ZBDongle-E?

GitComfy, 2023-12-16:
I couldnt find a way round this so ended up rebuilding mine from scratch.....

paulbaird87, 2024-01-12:
> Almost same here, zigbee2mqtt was working perfectly, until suddenly I received a similar error:
> 
> ```
> Using '/app/data' as data directory
> Zigbee2MQTT:info  2023-05-03 00:19:58: Logging to console and directory: '/app/data/log/2023-05-03.00-19-58' filename: log.txt
> Zigbee2MQTT:info  2023-05-03 00:19:58: Starting Zigbee2MQTT version 1.30.2 (commit #cdf62ea)
> Zigbee2MQTT:info  2023-05-03 00:19:58: Starting zigbee-herdsman (0.14.96)
> Zigbee2MQTT:error 2023-05-03 00:20:19: Error while starting zigbee-herdsman
> Zigbee2MQTT:error 2023-05-03 00:20:19: Failed to start zigbee
> Zigbee2MQTT:error 2023-05-03 00:20:19: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
> Zigbee2MQTT:error 2023-05-03 00:20:19: Exiting...
> Zigbee2MQTT:error 2023-05-03 00:20:19: Error: Connection not initialized
>     at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:540:19)
>     at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:338:35)
>     at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
>     at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
>     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
>     at Zigbee.start (/app/lib/zigbee.ts:58:27)
>     at Controller.start (/app/lib/controller.ts:101:27)
>     at start (/app/index.js:107:5)
> ```
> 
> I'm using zbbridge with ezsp for adapter.
> 
> Everything has been working fine up until now.
> 
> UPDATE: Figured out the problem with my device. For some reason, the tcp port 8888 was not exposed, so zigbee2mqtt could not connect to it. I started it manually and it worked. Need to figure out now why it does not expose it on every boot, as this was working perfectly fine up until now.

Do you mind explaining how you did this please?