Koenkk/zigbee2mqtt #17479: Danfos Ally eTRV0103 firmware update, wrong image type

Masoholik, 2023-04-27:
### What happened?

I bought a new Danfos Ally eTRV0103 and the overview is showing the Firmware Version 00.20.0008 00.20 with 20230324 as date.

[image]

i would like to update to the 1.20 firmware, but OTA Update says "no image avaiable for image type 288"
i saw here: https://github.com/Koenkk/zigbee-OTA/blob/master/index.json that there is already the new Firmware. But with image type 256?

When i use an local index.json file and change "imageType" to 288 and copy the ota file (https://www.danfoss.com/en/products/dhs/smart-heating/smart-heating/danfoss-ally/danfoss-ally-support/#tab-software)  to my server. then i get "image type mismatch".

Is there maybe a solution for this? And why is it showing me 00.20 as firmware version?

### What did you expect to happen?

do a firmware update

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

20230410

### Adapter

Zigstar Olizig

lux73, 2023-04-28:
not possbile -> the new Allys eTRV0103 got a new/different Microcontroller. There is actually no newer Firmware to Update

00.20 is the latest one - please search around the Forums

closed 2023-06-06 by github-actions[bot]

Mikkooooo, 2023-11-15:
I‘ve installed a few Popp smart zigbee thermostats.
Trying a FW update also shows :  No image avaiable for image type 288.

How to update with success ?

Mikkooooo