Koenkk/zigbee2mqtt #17483: TS0601 _TZE204_sxm7l9xa defined but shows as not supported

programmieramt, 2023-04-28:
### What happened?

I have the tuya human presence sensor but with another manufacturer id.

_TZE204_sxm7l9xa is the one I have

Zigbee2MQTT Version [1.30.3]

Same as in https://github.com/Koenkk/zigbee-herdsman-converters/issues/4940

### What did you expect to happen?

The device with the new manufacturer ID workes loke the others.

### How to reproduce it (minimal and precise)

Add the device

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

home assistant yellow

Koenkk, 2023-04-28:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/9295fc8afcc65f36027f9ab4d319ce64

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

patelyatin007, 2023-04-29:
2023-04-29 10:28:40fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,120]}}
2023-04-29 10:28:42fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,114]}}
2023-04-29 10:28:43fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,107]}}
2023-04-29 10:28:45fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,103]}}
2023-04-29 10:28:48fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
2023-04-29 10:28:50fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,96]}}
2023-04-29 10:28:51fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
2023-04-29 10:28:53fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,82]}}

patelyatin007, 2023-04-29:
these are the error that pops up

programmieramt, 2023-04-30:
Thanks for the quick answer:

I have now the lux working but get these errors:

`Warning 2023-04-30 14:12:24fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,226]}}
Warning 2023-04-30 14:12:26fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,60]}}
Warning 2023-04-30 14:12:28fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,67]}}
Warning 2023-04-30 14:12:29fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,182]}}
Warning 2023-04-30 14:12:31fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,108]}}
Warning 2023-04-30 14:12:32fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,103]}}`

Koenkk, 2023-04-30:
To figure out what these datpoints mean (e.g. 109) you need a TuYa bridge: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

programmieramt, 2023-04-30:
ok thank you. Very sad. Then I have to think about investing 20more euros into that :-)

dengi76, 2023-05-02:
I join The same problem!

programmieramt, 2023-05-02:
> I join The same problem!

Just ordered the tuya box in china. Perhabs we find a solution as soon as we find the datapoints

nicolacavallo84, 2023-05-02:
same problem for me too

mephistolein, 2023-05-03:
same problem here, i have no tuya bridge

RamonSmit, 2023-05-03:
I've got a bridge and today I've got four of those sensors. I'm going to try to check it out.

RamonSmit, 2023-05-03:
This is what I've found:

```
104 - Lux
105 - Presence
106 - Motion sensitivity 
107 - Max detection distance
108 - Min detection distance
109 - Target distance
110 - No delay time?
111 - Someone set the time? 
```

programmieramt, 2023-05-03:
> This is what I've found:
> 
> ```
> 104 - Lux
> 105 - Presence
> 106 - Motion sensitivity 
> 107 - Max detection distance
> 108 - Min detection distance
> 109 - Target distance
> 110 - No delay time?
> 111 - Someone set the time? 
> ```

I found this doku: https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_3-understanding-tuya-datapoints

But not sure what to add to the external converter(https://gist.github.com/Koenkk/9295fc8afcc65f36027f9ab4d319ce64)

RamonSmit, 2023-05-03:
I'll think there need to be a meta block like:
```js
 meta: {
        tuyaDatapoints: [
            [104, 'illuminance', tuya.valueConverter.raw],
            [105, 'presence', tuya.valueConverter.boolean],
            ... 
        ],
    },
```

programmieramt, 2023-05-03:
> I'll think there need to be a meta block something like this, I'm going to test it locally :-) :
> 
> ```js
>  meta: {
>         tuyaDatapoints: [
>             [104, 'illuminance', tuya.valueConverter.raw],
>             [105, 'presence', tuya.valueConverter.boolean],
>             [106, 'radar_sensitivity', tuya.valueConverter.raw],
>             [107, 'maximum_range', tuya.valueConverter.raw],
>             [108, 'minimum_range', tuya.valueConverter.raw],
>             [109, 'target_distance', tuya.valueConverter.raw],
>             [110, 'detection_delay', tuya.valueConverter.raw],
>             [111, 'fading_time', tuya.valueConverter.raw],
>         ],
>     },
> ```
> 
> I'm not really familiar with the systems yet.

hey thank you, I thought the same. But at the moment I still get the same errors as before.

RamonSmit, 2023-05-03:
I've got it some kind of working with https://gist.github.com/RamonSmit/157f08633b8a05feb8e845d34872999f, need to finetune it obviously a bit more :-)

programmieramt, 2023-05-03:
> I've got it some kind of working with https://gist.github.com/RamonSmit/157f08633b8a05feb8e845d34872999f, need to finetune it obviously a bit more :-). It's a nice way of learning hehe.

I found some examples where it was not `boolean` but `trueFalse1` but don't know

RamonSmit, 2023-05-03:
Feel free to collaborate, I'm new in here 😄

ecenshu, 2023-05-04:
I've managed to get mine working with the following:

```
meta: {
        tuyaDatapoints: [
            [104, 'illuminance_lux', tuya.valueConverter.raw],
            [105, 'presence', tuya.valueConverter.trueFalse1],
            [106, 'radar_sensitivity', tuya.valueConverter.raw],
            [107, 'maximum_range', tuya.valueConverter.divideBy100],
            [108, 'minimum_range', tuya.valueConverter.divideBy100],
            [109, 'target_distance', tuya.valueConverter.divideBy100],
            [110, 'detection_delay', tuya.valueConverter.raw],
            [111, 'fading_time', tuya.valueConverter.raw]
        ],
    },
```

ecenshu, 2023-05-04:
I am getting rare occurrences of the following in the log:

`No converter available for 'detection_delay' (6.1)`

programmieramt, 2023-05-04:
> illuminance_lux and presence seems to be working however any attempts at setting the state for radar_sensitivity, maximum_range, minimum_range, detection_delay or fading_time gives an error log entry like the one below:
> 
> `No converter available for 'detection_delay' (6.1)`

Thanks for the input. I have the same problem changing values

anhnt3, 2023-05-05:
Here working ext_converter.js:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        { modelID: 'TS0601', manufacturerName: '_TZE204_sxm7l9xa' }
    ],
    model: 'TS0601_smart_human_presence_sensor',
    vendor: 'TuYa',
    description: 'Smart Human presence sensor',
    fromZigbee: [{
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'tuya_smart_human_presense_sensor');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
            let result = null;
            switch (dp) {
                case 105://tuya.dataPoints.tshpsPresenceState:
                    result = { presence: { 0: false, 1: true }[value] };
                    break;
                case 106://tuya.dataPoints.tshpscSensitivity:
                    result = { radar_sensitivity: value };
                    break;
                case 108://tuya.dataPoints.tshpsMinimumRange:
                    result = { minimum_range: value / 100 };
                    break;
                case 107://tuya.dataPoints.tshpsMaximumRange:
                    result = { maximum_range: value / 100 };
                    break;
                case 109://tuya.dataPoints.tshpsTargetDistance:
                    result = { target_distance: value / 100 };
                    break;
                case 110://tuya.dataPoints.tshpsDetectionDelay:
                    result = { detection_delay: value / 10 };
                    break;
                case 111://tuya.dataPoints.tshpsFadingTime:
                    result = { fading_time: value / 10 };
                    break;
                case 104://tuya.dataPoints.tshpsIlluminanceLux:
                    result = { illuminance_lux: value };
                    break;
                case tuya.dataPoints.tshpsCLI: // not recognize
                    result = { cli: value };
                    break;
                case tuya.dataPoints.tshpsSelfTest:
                    result = { self_test: tuya.tuyaHPSCheckingResult[value] };
                    break;
                default:
                    meta.logger
                        .warn(`fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP ${dp} with data ${JSON.stringify(dpValue)}`);
            }
            return result;
        },
    }],
    toZigbee: [{
        key: ['radar_sensitivity', 'minimum_range', 'maximum_range', 'detection_delay', 'fading_time'],
[... 35 lines cut ...]
            .withDescription('Maximum range').withUnit('m'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
        // exposes.text('cli', ea.STATE).withDescription('not recognize'),
        exposes.enum('self_test', ea.STATE, Object.values(tuya.tuyaHPSCheckingResult))
            .withDescription('Self_test, possible resuts: checking, check_success, check_failure, others, comm_fault, radar_fault.'),
    ],
    // meta: {
    //     tuyaDatapoints: [
    //         [104, 'illuminance', tuya.valueConverter.raw],
    //         [105, 'presence', tuya.valueConverter.trueFalse1],
    //         [106, 'radar_sensitivity', tuya.valueConverter.raw],
    //         [107, 'maximum_range', tuya.valueConverter.raw],
    //         [108, 'minimum_range', tuya.valueConverter.raw],
    //         [109, 'target_distance', tuya.valueConverter.raw],
    //         [110, 'detection_delay', tuya.valueConverter.raw],
    //         [111, 'fading_time', tuya.valueConverter.raw]
    //     ],
    // }
};

module.exports = definition;
```

tomiss288, 2023-05-05:
It's work. Many thx. Your great.

kvsergy, 2023-05-05:
Thank you very much !! great !!!

dengi76, 2023-05-06:
It's strange, but it didn't work for me?!? Could you post the "ext_converter.js" file. Thank you

dengi76, 2023-05-06:
It's strange, but it didn't work for me?!? Could you post the file "ext_converter.js". Thank you

dengi76, 2023-05-06:
THANKS A LOT !!! Everything is working !!!

andysaab, 2023-05-06:
> THANKS A LOT !!! Everything is working !!! But ..., the illumination and the self_test do not work. Is everyone like that?

For me too - the illumination and the self_test do not work :(

dengi76, 2023-05-06:
> БОЛЬШОЕ СПАСИБО !!! Все работает!!! Но..., подсветка и самопроверка не работают. Все такие?

Moreover, in the "state" of the device, all data arrives, but in the properties - no!

andysaab, 2023-05-06:
To correct the illumination, you need to correct this line like this^
exposes: [
        e.illuminance**_lux**(), e.presence(),

dengi76, 2023-05-06:
> Чтобы исправить в Австралии, вам нужно исправить эту резолюцию по рекомендации: выставляет: [ e.illuminance_lux(), e.presence(),

Thanks - it worked!!!

Roman-RV78, 2023-05-06:
Hello. Sensitivity of the radar is always 7. The value always returns to 7. Could anyone check?

andysaab, 2023-05-06:
> Hello. Sensitivity of the radar is always 7. The value always returns to 7. Could anyone check?
Сonfirm exactly the same.

ecenshu, 2023-05-06:
The sensitivity setting should be the following: 

                    await tuya.sendDataPointValue(entity, **106**, value);

anhnt3, 2023-05-06:
> 

My typo mistake,thanks

andysaab, 2023-05-06:
Sensitivity still does not change after editing :(

Roman-RV78, 2023-05-06:
Restart Z2M? Everything works for me

Roman-RV78, 2023-05-06:
I also noticed that detection delay cannot be lower than 0.5. On another sensor, you can set 0

andysaab, 2023-05-06:
> Restart Z2M? Everything works for me
Yep, it working after clearning browser cache.

phoenixperson-c137, 2023-05-06:
Thanks!! it is working :)

I'm trying to play with the settings though and things seems a bit weird, is there any chance `fading_time` and `detection_delay` are swapped?

hvoss-techfak, 2023-05-06:
I think that might be the case, because you also can't lower the detection delay below 0.5 seconds, which would make sense if they are switched.

andysaab, 2023-05-06:
> I think that might be the case, because you also can't lower the detection delay below 0.5 seconds, which would make sense if they are switched.

Confirm. It's set to zero but after that jump to 0.5 again immediately.

Koenkk, 2023-05-07:
Great work @anhnt3 , I've refactored your converter to the new format. Can you check if everything works? Then I can add out-of-the-box support for this device: https://gist.github.com/Koenkk/9295fc8afcc65f36027f9ab4d319ce64

andysaab, 2023-05-07:
> Great work @anhnt3 , I've refactored your converter to the new format. Can you check if everything works? Then I can add out-of-the-box support for this device: https://gist.github.com/Koenkk/9295fc8afcc65f36027f9ab4d319ce64
Hi @Koenkk!
Tnx for your job, I try it - nothing change on my case.
1. The Detection can set to zero but after that jump to 0.5 again immediately.
2. SelfTest doesn't work like before.

anhnt3, 2023-05-07:
> 

The Detection can set to zero but after that jump to 0.5 again immediately.
-> Yes, the device report back 0.5, I think it is device capbility.
SelfTest doesn't work like before.
-> No idea the tuyapoint dp for self test. May be someone, check it with tuya zigbee hub.

anhnt3, 2023-05-07:
> 1. The Detection can set to zero but after that jump to 0.5 again immediately.
-> Yes, the device report back 0.5, I think it is device capbility.

> 2. SelfTest doesn't work like before.

The Detection can set to zero but after that jump to 0.5 again immediately.
-> Yes, the device report back 0.5, I think it is device capbility.
SelfTest doesn't work like before.
-> No idea the tuyapoint dp for self test. May be someone, check it with tuya zigbee hub.

RamonSmit, 2023-05-07:
> SelfTest doesn't work like before.

I've tried to find it, but i couldn't find it in the developer portal :(

Koenkk, 2023-05-07:
Updated https://gist.github.com/Koenkk/9295fc8afcc65f36027f9ab4d319ce64, removed self test and fixed min detection value. Please let me know if everything is ok

anhnt3, 2023-05-07:
> Updated https://gist.github.com/Koenkk/9295fc8afcc65f36027f9ab4d319ce64, removed self test and fixed min detection value. Please let me know if everything is ok

confirmed. thanks.

programmieramt, 2023-05-07:
> Updated https://gist.github.com/Koenkk/9295fc8afcc65f36027f9ab4d319ce64, removed self test and fixed min detection value. Please let me know if everything is ok

With the external converter the devices are not really responsiv and very unreliable. Did I made something wrong?

What is the workflow after changing the converter?

Roman-RV78, 2023-05-07:
It looks like detection_delay and fading_time are mixed up in places,
detection_delay 111 ----- with Value Min(0)
fading_time 110 ----- with Value Min(0.5)

soulblade7, 2023-05-08:
> It looks like detection_delay and fading_time are mixed up in places, detection_delay 111 ----- with Value Min(0) fading_time 110 ----- with Value Min(0.5)

Just confirming that this fixed things for me. Prior to this the 2 sensors I have behaved erratically.

anhnt3, 2023-05-08:
> It looks like detection_delay and fading_time are mixed up in places, detection_delay 111 ----- with Value Min(0) fading_time 110 ----- with Value Min(0.5)

Tested and confirmed. Thanks.

programmieramt, 2023-05-08:
> It looks like detection_delay and fading_time are mixed up in places, detection_delay 111 ----- with Value Min(0) fading_time 110 ----- with Value Min(0.5)

WIth the fix of these two mixed lines everzthing works fine so far. Thank you.

programmieramt, 2023-05-08:
> 

I think if everythink is running fine it will be merged into the z2m code and with some of the next releases it will be fully functional out of the box

savaloi, 2023-05-09:
Seems to work fine, however the image is wrong for the wall mount version of the device - it shows up as the ceiling image in Z2M - is there a way to fix this? Image as listed in Issue #17583

Koenkk, 2023-05-09:
Added the device.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@savaloi to confirm,[this](https://github.com/Burki24/Z2M_Pictures/blob/main/devices/radar-wall-1.jpg) should be the image?

savaloi, 2023-05-10:
> Added the device.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)
> 
> @savaloi to confirm,[this](https://github.com/Burki24/Z2M_Pictures/blob/main/devices/radar-wall-1.jpg) should be the image?

Yes that’s the one. Thanks

NoProsNoNoobs, 2023-05-14:
Hi,

I am not sure if this is the right place for this. I have this exact same manufacturer id and have been using this converter for about an week now, but sometimes it doesn't change presence from True to False when no one is in the room.

It updates the presence when i change something in it's settings like radar sensitivity or something else. I thought maybe it has something to do with the converter, so that's why i am asking here.

ShadowRep, 2023-05-15:
The pictures in the edge version are now wrong.

Version 1 should look either like this: [V1 Ceiling Version Aliexpress](https://ae01.alicdn.com/kf/S1f8b735da4b14a128ea4c38ebbe5d228K/Smart-Life-Zigbee-Human-Presence-Detector-Tuya-Wifi-MmWave-Radar-Pir-Montion-Sensor-With-Luminance-Detection.jpg_640x640.jpg)
or like this: [V1 Wall Version Aliexpress](https://github.com/Burki24/Z2M_Pictures/blob/main/devices/radar-wall-1.jpg?raw=true).

Instead it now shows a different Tuya presence sensor: [Tuya PS-HPS Human Presence Sensor](https://www.zigbee2mqtt.io/images/devices/TS0601_smart_human_presence_sensor_1.jpg)

The picture it was showing before I switched to edge ([old picture non edge version](https://www.zigbee2mqtt.io/images/devices/TS0601_smart_human_presence_sensor.jpg)) was also wrong but way closer in appearance to the real one.

Can this please be changed?

[68 more replies]

closed 2023-05-24 by Koenkk