Koenkk/zigbee2mqtt #17489: Device turning on and then quickly and unexpectedly turning off

rccoleman, 2023-04-28:
### What happened?

I've been seeing some strange behavior recently involving single devices and groups that I ask to turn on via Home Assistant and sometimes quickly turn on and back off again.  Sometimes it's a single device and sometimes it's one device in a group of 2 or more, but it's a relatively recent phenomenon.

### What did you expect to happen?

I expect a device that I command to turn on stays on and doesn't quickly turn off.

### How to reproduce it (minimal and precise)

I can't reproduce this at will, but simply turning a device or group on and off a few times (maybe 5-10) in Home Assistant is enough to get it to happen.

### Zigbee2MQTT version

1.30.3-dev commit: 13c7348

### Adapter firmware version

20221226

### Adapter

Tubeszb USB adapter

### Debug log

Here's the debug log for a successful attempt to turn on a Zigbee group with 2 plug-in modules by turning on the associated switch in Home Assistant:

```
debug 2023-04-28 14:29:25: Received MQTT message on 'zigbee2mqttp/MBR Lamp Group/set' with data 'ON'
debug 2023-04-28 14:29:25: Publishing 'set' 'state' to 'MBR Lamp Group'
info  2023-04-28 14:29:26: MQTT publish: topic 'zigbee2mqttp/MBR Corner Lamp', payload '{"linkquality":105,"power_on_behavior":"previous","state":"ON","update":{"installed_version":268513324,"latest_version":268513324,"state":"idle"},"update_available":null}'
info  2023-04-28 14:29:26: MQTT publish: topic 'zigbee2mqttp/MBR Lamp Group', payload '{"state":"ON"}'
info  2023-04-28 14:29:26: MQTT publish: topic 'zigbee2mqttp/MBR Outside Lamp', payload '{"linkquality":134,"power_on_behavior":"previous","state":"ON","update":{"installed_version":268513324,"latest_version":268513324,"state":"idle"},"update_available":null}'
```

Note that both devices transition to the "ON" state.

Here's an example where I do the exact same thing in Home Assistant:

```
info  2023-04-28 14:29:27: MQTT publish: topic 'zigbee2mqttp/MBR Lamp Group', payload '{"state":"ON"}'
info  2023-04-28 14:29:27: MQTT publish: topic 'zigbee2mqttp/MBR Outside Lamp', payload '{"linkquality":142,"power_on_behavior":"previous","state":"ON","update":{"installed_version":268513324,"latest_version":268513324,"state":"idle"},"update_available":null}'
info  2023-04-28 14:29:27: MQTT publish: topic 'zigbee2mqttp/MBR Outside Lamp', payload '{"linkquality":142,"power_on_behavior":"previous","state":"ON","update":{"installed_version":268513324,"latest_version":268513324,"state":"idle"},"update_available":null}'
info  2023-04-28 14:29:29: MQTT publish: topic 'zigbee2mqttp/MBR Corner Lamp', payload '{"linkquality":109,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":268513324,"latest_version":268513324,"state":"idle"},"update_available":null}'
```

See that the same command to turn on the group is sent, but in this case we get two separate status reports for "MBR Outside Lamp" turning on and one for "MBR Corner Lamp" turning *OFF*.  In reality, both modules turn on, and then "MBR Corner Lamp" quickly turns off.

I've made a few changes to my configuration that could have had an effect, but I can't narrow down to one thing.

- I switched from a Sonoff-P adapter to the Tubeszb adapter
- I updated the firmware in the plug-in modules that I'm using (https://www.amazon.com/gp/product/B09KNDM4VV/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) to v1.00.43
- I've updated to the latest Zigbee2MQTT `latest-dev` container tag a few times

I'll turn on herdsman debug to see if I see anything else fishy, but wanted to start the report with what I have.

rccoleman, 2023-04-28:
Same as this behavior, which points to the firmware: https://www.reddit.com/r/homeassistant/comments/12ovc28/thirdreality_zigbee_smart_outlets_latest_firmware/

jmluxem, 2023-04-28:
+1 
I’m seeing this same behavior on my ThirdReality plugs

jmluxem, 2023-05-01:
One thing to note, I've only seen this occur when the device receives an On command. I haven't seen the devices malfunction when turning off. I also have video of the situation happening, but I have to assume the logs posted above are more helpful. 

Any help or advice on this issue would be tremendously appreciated. It's driving me crazy

rccoleman, 2023-05-05:
I've been communicating with Third Reality and have what I think is a workaround. They suggested the following:

Enter the Third Reality Plug device page in Zigbee2MQTT.
Select “Dev console” tab.
Select “1” under the Endpoint dropdown.
Select “OnOff” under the Cluster dropdown.
Select “startUpOnOff” under the Attribute dropdown.
Enter “255” into the startUpOnOff field and click "Write”.

This essentially tells the plug to turn on following a power cycle, rather than the default of 'off'. After doing that, one of my plugs turned on, quickly turned off, then turned back on with a couple of seconds. I suspect that the plug is resetting or crashing in response to the 'on' command sometimes, causing it to turn off and then return to the state specified by the attribute. It's a hack for sure, but leaves the plug in the expected state.

jmluxem, 2023-05-05:
Testing your fix now. I'll take anything at this point. Thank you very much for the input

jmluxem, 2023-05-05:
I've seen one device show the error since applying the Dev Console change - so that doesn't appear to be a full fix.

I did notice that ThirdReality has a new firmware release available now for my plugs, v1.00.55. Running it on one now and updating a couple others to test that. I'm not able to find any release notes related to the update

rccoleman, 2023-05-05:
Good to hear there's a new firmware release, I'll try it too.  I only saw the new on/off/on behavior on one of the plugs where I modified the attribute, so my sample size is too small to make any declarations.

jmluxem, 2023-05-06:
I reached out to ThirdReality about firmware release notes or info for v.1.00.55, and below is their response. I'm hopeful this is a permanent fix. 

""
About the release information on v1.00.55 of the smart plug, it's for keeping SatrUpOnOff states after OTA, and adding tags for different plug type.
""

rccoleman, 2023-05-10:
I haven't seen this issue on any plugs after updating them to v1.00.55, so I'm going to close this.  It's clearly not a Zigbee2MQTT issue.