Koenkk/zigbee2mqtt #17490: MOES gu10 spots offline ZLD-RCW

Dutchy-79, 2023-04-29:
### What happened?

I bought 5 MOES gu10 spots and installed 2 few months ago. Thursday I finally installed the other 3 and before going outside they were already connected to Z2M (sonoff zigbee 3.0 dongle plus)

They all turned on in the evening and off in the morning. Next day the 3 new didn't turn on and showed offline, tried pairing again and none of them connected (permit join was on)

This morning the other 2 I installed months ago didn't turn off and showed offline, also won't pair anymore. 

### What did you expect to happen?

The lights not going offline 

### How to reproduce it (minimal and precise)

Not sure because it happened on my 5 lights 3 first next morning other 2 while nothing was changed. Didn't add or remove nodes or installed updates. Didn't even reboot the server. 

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

20221102

### Adapter

Sonoff zigbee 3.0 dongle plus 

### Debug log

Log will follow, can't download on phone

closed 2023-06-06 by github-actions[bot]

jonny190, 2023-12-20:
@Dutchy-79 did you manage to get them to enter pairing mode and if so how. Ive got 2 out of 4 that connected and unknown and ive not been able to get any of the 4 to enter pairing mode

Dutchy-79, 2023-12-20:
It took me a lot of time to get them to pair, eventually they started blinking. I think it's 5 times on off

jonny190, 2023-12-20:
Where you staring from on or off?
https://photos.app.goo.gl/pjhoVzx1grdWqc7x8

Dutchy-79, 2023-12-20:
Started with off, though some lights took me several attempts

jonny190, 2023-12-20:
Hahaha I'm on day 2 of trying to reset them, I'd understand if it was just
the 2 none paired ones that wouldn't reset, but all 4?

It feels like I'm doing something wrong

Wickedy123, 2024-01-03:
The instructions say to power cycle three times: on-off, on-off, on-off, ON >> [up to 5 seconds] >>> starts flashing.

You need to leave a decent gap between off and on, so the power drains out. Having paired 30+ of these bulbs, some need 3+ seconds before they think they're 'off', but for most 1.5 will do it.

Wickedy123, 2024-01-03:
The problem I've had with ALL these bulbs (38 I think), is that they ALL unpaired themselves when I had a powercut. I'm using them with ZHA (because my z2mqtt is overloaded). 

If anyone has successfully powered off for 1 hour+ and then had all their MOES GU10 bulbs join again, I would be interested to know. 
I'm testing them for the same when paired with the official MOES (Tuya) hub. 

The firmware version stated in the Tuya app is v1.2.4, there don't seem to be any firmware updates available.