Koenkk/zigbee2mqtt #17492: Conbee II: Add install code is not supported

rouvenst, 2023-04-29:
### What happened?

I want to add my new Bosch Thermostat II 230 V (BTH-RM230Z) with Home Assistant and Conbee II via install code. 
Therefore I scanned the qr code from the back of the device and entered it into the "Add install code" tool.
Afterwards the error `Add install code is not supported` is shown.

Logs: 

```
Zigbee2MQTT:error 2023-04-29 12:06:14: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'Add install code is not supported'
Zigbee2MQTT:info  2023-04-29 12:06:14: MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{},"error":"Add install code is not supported","status":"error","transaction":"ookk5-1"}'
```

### What did you expect to happen?

It should add the device to z2m.

### How to reproduce it (minimal and precise)

I'm using Conbee 2 from dresden-elektronik.
Used the function "Add install code" from Settings > Tools.

Related to #14510. I guess Conbee 2 did not support install codes yet. With v2.22.0-beta there should be support. Maybe we can leverage this here (https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6791)

### Zigbee2MQTT version

 1.30.3

### Adapter firmware version

0x26780700

### Adapter

ConBee2

Hedda, 2023-05-09:
FYI, it first needs to be supported in the firmware for ConBee/RaspBee as well as documented for their deconz-serial-protocol:

https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20

If it is already supported for it in their firmware then it needs to be implemented in the deconz adapter code for zigbee-herdsman:

https://github.com/Koenkk/zigbee-herdsman/tree/master/src/adapter/deconz

zigbee-herdsman is the Zigbee framework that Zigbee2MQTT and IoBroker uses:

https://github.com/Koenkk/zigbee-herdsman/

Suggest posting a new feature for it under issues in the zigbee-herdsman repository for developer tracking:

https://github.com/Koenkk/zigbee-herdsman/issues

PS: Similar to request related to deconz radio library for zigpy (which is used by Home Assistant's ZHA, Domoticz, and Jeedom):

https://github.com/zigpy/zigpy-deconz/issues/214

Ping @ChrisHae and @manup

rouvenst, 2023-06-12:
no stale

rouvenst, 2023-07-14:
no stale

lerra, 2023-08-15:
Still relavent

SnDsound, 2023-09-12:
Any news from battlefield?

Hedda, 2023-09-12:
> Any news from battlefield?

The problem is upstream as Dresden-Elektronik's own developers (as in the makers of ConBee/RaspBee) need to further document their deCONZ serial protocol for third-parties who use it -> https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20

ChrisHae, 2023-09-12:
It is not yet implemented in the serial protocol. When it is, it can be added to z2mqtt deconz adapter.

lerra, 2023-10-19:
Would be nice to understand the estimated timeline so I know if I should
move away from conbee II or not

mjaensch, 2023-11-23:
I would also very much appreciate any news on this problem.
Winter is here and my Bosch Thermostats are lying around uselessly.

Meilu, 2023-12-12:
I have just received my Bosch thermostat but receiving the same error with my Conbee stick. News on this issue very much appreciated.

Hedda, 2023-12-13:
Still no news, again the issue is with upstream, see -> https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20

> The problem is upstream as Dresden-Elektronik's own developers (as in the company that makes ConBee and RaspBee) need to further document their deCONZ serial protocol to help third-parties who use it -> [dresden-elektronik/deconz-serial-protocol#20](https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20)

ChrisHae, 2023-12-13:
There is no official support of install codes for ConBee or RaspBee.

Hedda, 2023-12-13:
@ChrisHae is it still on your roadmap? You wrote this in your other issue:

>  The Conbee2 fw does not support it yet. But it will come soon. And then it will also be implemented in zigbee herdsman deconz adapter.

_Originally posted by @ChrisHae in https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20#issuecomment-1431599456_

ChrisHae, 2023-12-13:
I think it is. But I'm not firmware developer. Better ask in dresden elektronik github pages or manup directly.

cromelex, 2024-02-04:
Is there are news on this? Bought a Bosch Thermostat when I saw it as supported on zigbee2mqtt's list, only to have the issue trying to pair with the conbeeII.

Is there a list somewhere of the Adapters that actually support this? Might as well upgrade from the conbee2 if this wil ltake too long.

Thanks

Hedda, 2024-02-04:
> Is there a list somewhere of the Adapters that actually support this?

Only supported by Texas Instruments and Silicon Labs based adapters so far as not documented for deconz -> https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20#issuecomment-1431599456

gr3k, 2024-09-28:
You can check this:

6.2.3 Zigbee Install Codes
The Serial Protocol and firmware doesn’t support Zigbee Install Codes directly. In order to join a
device via Install Code, a device specific Link Key needs to be derived from the Install Code by
the host application.
This 16-byte Link Key together with the device MAC address can than be configured by a Write Parameter Request (0×0B) for the Link Key parameter (0×19).
Once configured, during joining process, the device will receive the Network Key encrypted with
the Link Key instead of the global Truster Center Link Key.
A C++ OpenSSL reference implementation for the key derivation can be found at:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/crypto/mmohash.cpp

source: http://deconz.dresden-elektronik.de/raspbian/deCONZ-Serial-Protocol_en_1.22.pdf

Hedda, 2024-09-29:
See development discussion in this feature request issue for tracking in the zigbee-herdsman repository:

- https://github.com/Koenkk/zigbee-herdsman/issues/1184

zigbee-herdsman is the Zigbee framework library that Zigbee2MQTT depends on so maybe better to diacuss this there.

antoine101009, 2024-12-24:
Hello do you have any news for add install code in a combee 2 ? 
I have ordre a bosch siren but install code is not supported :(

vjonas, 2024-12-30:
> Hello do you have any news for add install code in a combee 2 ? I have ordre a bosch siren but install code is not supported :(

I'm afraid it will never get added, it's been so long now. Did you try updating the firmware of the conbee?

Hedda, 2024-12-30:
I believe that Nerivec addes code for it in zigbee-herdsman few weeks ago (so that code should ve available in Zigbee2MQTT dev branch) but he could/would not test it himself as he does not own or use a ConBee adapter himself. See discussion here:

- https://github.com/Koenkk/zigbee-herdsman/issues/1184

MamboBulgur, 2025-02-01:
any update? I never came up with the idea to check if it’s compatible with my conbee II :(
I followed the thread mentioned above but it I don’t understand what’s going on there

Hedda, 2025-02-01:
> I believe that Nerivec addes code for it in zigbee-herdsman few weeks ago (so that code should ve available in Zigbee2MQTT dev branch) but he could/would not test it himself as he does not own or use a ConBee adapter himself. See discussion here:
> 
> * [deconz driver/adapter (for ConBee/RaspBee Zigbee Coordinator) does not support joining/pairing via install code (or qr code) zigbee-herdsman#1184](https://github.com/Koenkk/zigbee-herdsman/issues/1184)
> 
> Note that you will need to first update to latest firmware on the ConBee regardless as otherwise it will never work.

Comment and link above from december 2024 still apply so follow that link and read all the posts there for latest news.