Koenkk/zigbee2mqtt #17495: [New device support]: Aubess Tuya Door, Temperature and Humidity Sensor

CamFlyerCH, 2023-04-29:
### Link

https://www.aliexpress.com/item/1005005194831629.html

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0xe0798dfffe254160","nwkAddr":35581,"manufId":4098,"manufName":"_TZE200_nvups4nh","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0012��*\u0017\u0012��*\u0013\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_nvups4nh","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1682794951397,"defaultSendRequestWhen":"immediate"}

### Comments

The same device as in the staled and closed https://github.com/Koenkk/zigbee2mqtt/issues/15009
The device should transmit 4 values: Reed contact open or closed, Temperature, Humidity and Battery Level.

I captured some debug data:
`2023-04-29 21:09:26Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1792}' from endpoint 1 with groupID 0
2023-04-29 21:09:27Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,33],"type":"Buffer"},"datatype":2,"dp":7}],"seq":1792}' from endpoint 1 with groupID 0
2023-04-29 21:09:27Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,62],"type":"Buffer"},"datatype":2,"dp":8}],"seq":1792}' from endpoint 1 with groupID 0
2023-04-29 21:09:27Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1792}' from endpoint 1 with groupID 0
2023-04-29 21:09:27Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":2048}' from endpoint 1 with groupID 0
2023-04-29 21:09:28Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,35],"type":"Buffer"},"datatype":2,"dp":7}],"seq":2048}' from endpoint 1 with groupID 0
2023-04-29 21:09:28Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,61],"type":"Buffer"},"datatype":2,"dp":8}],"seq":2048}' from endpoint 1 with groupID 0
2023-04-29 21:09:28Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2}],"seq":2048}' from endpoint 1 with groupID 0
2023-04-29 21:09:30Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":2304}' from endpoint 1 with groupID 0
2023-04-29 21:09:30Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,32],"type":"Buffer"},"datatype":2,"dp":7}],"seq":2304}' from endpoint 1 with groupID 0
2023-04-29 21:09:30Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,61],"type":"Buffer"},"datatype":2,"dp":8}],"seq":2304}' from endpoint 1 with groupID 0
2023-04-29 21:09:30Received Zigbee message from '0xe0798dfffe254160', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2}],"seq":2304}' from endpoint 1 with groupID 0`

I think 1 is the reed contact (1 if door is open and 0 if it's closed, 2 ist battery state, 7 is temperature and 8 is humidity. In the data above Humidity of around 61 seams ok, since it rises to 98 if I breath into the unit. Temperature instead seams to need some conversion. Values 2, 7 and 8 are transmitted in intervals of 120 seconds, Value 1 only while operating the reed switch / door.

The unit is cheap and also cheep in the inside and the housing. It uses two AAA batteries instead of a button cell. I like that.
I can also upload some pictures from the inside.

CamFlyerCH, 2023-04-29:
Inside:
[image]

[image]

CamFlyerCH, 2023-04-29:
I managed to create an external converter. Did I do that the right way ?

https://gist.github.com/CamFlyerCH/26c37bc97fe2a3be3b0a043dc1708883

Koenkk, 2023-04-30:
Looks good, do you mind making a PR? (click edit icon [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js), if not I can also integrate it for you

CamFlyerCH, 2023-04-30:
> Looks good, do you mind making a PR? (click edit icon [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js), if not I can also integrate it for you

I am no very familiar with structure and naming "policies" and also how to test the code then. I think it would be way more efficient if you can do it. Also after 12h in use the data looks good on the humidity and temperature sensor and the door open events are reliable and fast.

Koenkk, 2023-04-30:
Added! I've changed the `door` to a `contact` sensor (to make it similar to other contact sensors)

Koenkk, 2023-04-30:
TS0601_contact_temperature_humidity_sensor, @CamFlyerCH

closed 2023-04-30 by Koenkk

Striiter, 2023-12-07:
This still don't work. I've just added it. Logs show this:

2023-12-07 22:16:43.821 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0601' manuf='_TZE200_nvups4nh' nwk=0x33C6 ieee=e0:79:8d:ff:fe:68:4c:9c is_initialized=True>
2023-12-07 22:16:43.823 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_nvups4nh TS0601 (e0:79:8d:ff:fe:68:4c:9c)
2023-12-07 22:16:43.823 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2023-12-07 22:16:43.823 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}

2023-12-07 22:45:27.819 DEBUG (Thread-13) [aiosqlite] executing functools.partial(<built-in method executemany of sqlite3.Connection object at 0x7f8eb10698a0>, 'INSERT INTO attributes_cache_v12 VALUES (?, ?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET value=excluded.value, last_updated=excluded.last_updated', [(e0:79:8d:ff:fe:68:4c:9c, 1, 0, 4, '_TZE200_nvups4nh', 1701985308.27506), (e0:79:8d:ff:fe:68:4c:9c, 1, 0, 5, 'TS0601', 1701985308.275083), (e0:79:8d:ff:fe:68:4c:9c, 1, 0, 1, 68, 1701985336.749626), (e0:79:8d:ff:fe:68:4c:9c, 1, 0, 65506, 31, 1701985336.749654), (e0:79:8d:ff:fe:68:4c:9c, 1, 0, 65508, 0, 1701985336.749665), (e0:79:8d:ff:fe:68:4c:9c, 1, 0, 65503, ' \x06\x05-\x12 \x06\x05-\x12', 1701985337.320405)])
2023-12-07 22:45:27.820 DEBUG (Thread-13) [aiosqlite] operation functools.partial(<built-in method executemany of sqlite3.Connection object at 0x7f8eb10698a0>, 'INSERT INTO attributes_cache_v12 VALUES (?, ?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET value=excluded.value, last_updated=excluded.last_updated', [(e0:79:8d:ff:fe:68:4c:9c, 1, 0, 4, '_TZE200_nvups4nh', 1701985308.27506), (e0:79:8d:ff:fe:68:4c:9c, 1, 0, 5, 'TS0601', 1701985308.275083), (e0:79:8d:ff:fe:68:4c:9c, 1, 0, 1, 68, 1701985336.749626), (e0:79:8d:ff:fe:68:4c:9c, 1, 0, 65506, 31, 1701985336.749654), (e0:79:8d:ff:fe:68:4c:9c, 1, 0, 65508, 0, 1701985336.749665), (e0:79:8d:ff:fe:68:4c:9c, 1, 0, 65503, ' \x06\x05-\x12 \x06\x05-\x12', 1701985337.320405)]) completed

Koenkk, 2023-12-08:
You are not using zigbee2mqtt, but ZHA, create an issue there