Koenkk/zigbee2mqtt #17499: Z2M -Yet again, SRSP - AF - dataRequest after 6000ms))

cloudbr34k84, 2023-04-29:
### What happened?

Woke to lots of devices offline.  Between going to bed and waking up z2m decided to play up.  No. Idea why. 

I'm sick of this happening.  I don't think it can handle large networks tbh. 

Partial log

Zigbee2MQTT:warn  2023-04-30 07:15:48: Failed to ping 'Front Patio Pendant Bulb 3' (attempt 1/2, Read 0xa4c138e2163723b7/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))

### What did you expect to happen?

My devices were fine.  Im

### Zigbee2MQTT version

 1.30.2 commit: unknown

### Adapter firmware version

20221226

### Adapter

Sonoff Coordinator P

cloudbr34k84, 2023-04-29:
@Koenkk mate need some help pls. This is the 2nd time within 2 weeks I'm having these issues. I went to bed last night, everything was perfect, woke up to like 80 devices offline and the coordinator I guess not working as well. I have never seen this error before - 0x70ac08fffe916c19/1 manuSpecificTuya.mcuGatewayConnectionStatus({"payloadSize":1,"payload":1. 
Ideas?

I'm starting to think the firmware has trouble with large networks still

cloudbr34k84, 2023-04-30:
Update:
So i changed to channel 26 and started re-adding devices. Everything going great, but as soon as I hit around 70 devices I think , I start seeing the errors again. and now devices are going offline. Im at 82 devices with another 40 to go. 
and now my stick has decided to restart and now I'm getting an error Error: AREQ - ZDO - activeEpRsp after 10000ms
```
info  2023-04-30 12:55:12: Logging to console and directory: '/config/zigbee2mqtt/log/2023-04-30.12-55-12' filename: log.txt
info  2023-04-30 12:55:12: Starting Zigbee2MQTT version 1.30.3 (commit #unknown)
info  2023-04-30 12:55:12: Starting zigbee-herdsman (0.14.103)
error 2023-04-30 12:55:35: Error while starting zigbee-herdsman
error 2023-04-30 12:55:35: Failed to start zigbee
error 2023-04-30 12:55:35: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-04-30 12:55:35: Exiting...
error 2023-04-30 12:55:35: Error: AREQ - ZDO - activeEpRsp after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
```

Koenkk, 2023-04-30:
Could you try the `20230410` firmware? This might provide some stability improvements in your case: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

cloudbr34k84, 2023-05-01:
> Could you try the `20230410` firmware? This might provide some stability improvements in your case: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

ill try now as its happened again. I have been chilling with my baby all morning and I decided to try and pair this device, and then all of a sudden I'm getting errors and devices just dropping off again.. Is there a better stick than the Sonoff P??

cloudbr34k84, 2023-05-01:
> ill try now as its happened again. I have been chilling with my baby all morning and I decided to try and pair this device, and then all of a sudden I'm getting errors and devices just dropping off [...]

So I'm trying to flash this and for some reason, it won't let work, but it will work for launchpad_coordinator_20221226.hex
```
>Initiate access to target: COM4 using 2-pin cJTAG.
>Reading file: C:/Users/cloud/Documents/Zigbee/firmware/CC1352P2_CC2652P_launchpad_coordinator_20230410.hex/CC1352P2_CC2652P_launchpad_coordinator_20230410.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.
>Initiate access to target: COM4 using 2-pin cJTAG.
>No response from device. Device may not be in bootloader mode. Reset device and try again.
If problem persists, check connection and baud rate.
>Connecting over serial bootloader failed: No response from device. Device may not be in bootloader mode. Reset device and try again.
If problem persists, check connection and baud rate.
>Failed to create device object.
```

cloudbr34k84, 2023-05-01:
@Koenkk i really don't know what is happening. None of the Sonoff sticks would flash so I had to resort to trying out my Zigstar POE adapter. I have been stuck on this issue" Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby"

How do I just start again??

Koenkk, 2023-05-01:
what command are you using to flash? command should have `--bootloader-sonoff-usb`

cloudbr34k84, 2023-05-01:
Im using Smart RF Flash Programmer 2.
How is it i can flash the previous hex file and not this one?

cloudbr34k84, 2023-05-02:
closing this down to start a new issue.. defiantly something wrong with Z2m on larger networks

closed 2023-05-02 by cloudbr34k84