Koenkk/zigbee2mqtt #17500: TS0601_soil (_TZE200_myd45weu) - Battery drain (reporting way too much)

ASNNetworks, 2023-04-29:
### What happened?

I have this plant moisture soil sensor and the battery drains way too quickly. I have it under 6 weeks and already battery has went from 100% to 20%. I notice it reports way too much than necessary for something like this. Basically it reports every 40 seconds, non-stop. Perhaps this is the reason it drains so fast?

I tried changing the reporting settings, but nothing sticks and it keeps giving me a timeout no matter the attributes I try to add (reporting list is empty).

### What did you expect to happen?

I expect it to not drain batteries every 6 weeks.

### Zigbee2MQTT version

 1.30.3

### Adapter firmware version

20221226

### Adapter

Sonoff P

### Debug log

```
Info 2023-04-30 01:47:48MQTT publish: topic 'zigbee2mqtt/Woonkamer Plant Eettafel', payload '{"battery":20,"battery_state":"low","last_seen":"2023-04-30T01:47:43+02:00","linkquality":116,"soil_moisture":79,"temperature":21,"temperature_unit":"celsius"}'
Info 2023-04-30 01:48:29MQTT publish: topic 'zigbee2mqtt/Woonkamer Plant Eettafel', payload '{"battery":20,"battery_state":"low","last_seen":"2023-04-30T01:48:24+02:00","linkquality":116,"soil_moisture":79,"temperature":21,"temperature_unit":"celsius"}'
Info 2023-04-30 01:49:10MQTT publish: topic 'zigbee2mqtt/Woonkamer Plant Eettafel', payload '{"battery":20,"battery_state":"low","last_seen":"2023-04-30T01:49:05+02:00","linkquality":120,"soil_moisture":79,"temperature":21,"temperature_unit":"celsius"}'
Info 2023-04-30 01:49:51MQTT publish: topic 'zigbee2mqtt/Woonkamer Plant Eettafel', payload '{"battery":20,"battery_state":"low","last_seen":"2023-04-30T01:49:47+02:00","linkquality":120,"soil_moisture":79,"temperature":21,"temperature_unit":"celsius"}'
```

cloudbr34k84, 2023-04-30:
yeah same, i cant change the reporting as well

TheLordVader, 2023-04-30:
I concur, same problem here with all 3 see soil sensors.

krikk, 2023-05-19:
i have 4 of these.. same problem... complete waste of money, i think this has to be fixed in the device firmware... and i am not sure if it is even updateable...

stpuk, 2023-05-21:
same problem here, battery drain of 40% in 24 hours due to level of reporting and no matter what I try I cannot set the reporting interval higher without receiving the timeout error.

jonathanathe, 2023-05-25:
The same thing here reports more than it should... I hope it can be solved if not, it will be a waste of money.

Koenkk, 2023-05-26:
- Does the draining also happen with the TuYa hub?
- Could somebody sniff the traffic when pairing this device with the TuYa hub and with z2m? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

mattbruman, 2023-05-30:
Having the issue too, I am looking to replace the batteries with solar so it only updates during the day I don't need it updating overnight just during the day so I know to go water.

TheLordVader, 2023-05-30:
How do you mean replace the battery with solar? Interesting, please explain.

mattbruman, 2023-05-30:
> How do you mean replace the battery with solar? Interesting, please explain.
> […](#)

By taking a solar panel, lipo battery, and a step up convertor and wire it to the soil sensor directly. Will need to seal it really well after.

krikk, 2023-06-11:
> * Does the draining also happen with the TuYa hub?
i think it will, because i just sniffed the traffic when joined to the tuya hub and i also get data packages ever 41 seconds (like with zigbee2mqtt

> * If yes, could somebody sniff the traffic when pairing this device with the TuYa hub and with z2m? 

i have a sniff when doing a join of one of the sensors with my test tuya hub, attached the capture file...
[tuya_soil_sensor_tuya_join.zip](https://github.com/Koenkk/zigbee2mqtt/files/11715745/tuya_soil_sensor_tuya_join.zip)
the joined device has the adress: 0xa4c138f6b848930f

Koenkk, 2023-06-12:
If the draining also happens with the original TuYa hub, I cannot do anything here, seems the device sucks.

closed 2023-07-21 by github-actions[bot]

memehammad, 2023-09-17:
Any updates on this? I'd like to change the reporting interval for mine, i've got the new version that doesnt report every few seconds but this causes it to not report enough data. I'd like to change the max report interval for soild_moisture to 10 minutes but i'm getting this error:

`Error` 2023-09-17 12:39:29Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0xa4c138acb1c97854/1 msSoilMoisture([{"attribute":"measuredValue","minimumReportInterval":60,"maximumReportInterval":600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 15482 - 1 - 83 - 1032 - 7 after 10000ms)'
`Info` 2023-09-17 12:39:29MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{},"error":"ConfigureReporting 0xa4c138acb1c97854/1 msSoilMoisture([{\"attribute\":\"measuredValue\",\"minimumReportInterval\":60,\"maximumReportInterval\":600,\"reportableChange\":0}], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 15482 - 1 - 83 - 1032 - 7 after 10000ms)","status":"error","transaction":"b6djq-4"}'

ASNNetworks, 2023-09-17:
> Any updates on this? I'd like to change the reporting interval for mine, i've got the new version that doesnt report every few seconds but this causes it to not report enough data. I'd like to [...]

Read the last comments. This device also behaves this way with the original gateway as well. So nothing z2m can do about this. It's up to Xiaomi to fix this piece of this expensive sensor.

ReX1983, 2024-08-17:
Is there any update or you gave up using TS0601_soil? I thought it was supported, so I bought 5 of them and they behave very randomly. One reports every hour, others never since paired.

TheLordVader, 2024-08-19:
I gave on on all of my 3 devices. They are in a box in the garage...

ReX1983, 2024-08-19:
> Mine works flawlessly, it just needs a stable connection.

Which repeaters do you use?

memehammad, 2024-08-19:
> > Mine works flawlessly, it just needs a stable connection.
> 
> Which repeaters do you use?

https://a.aliexpress.com/_EuTk42Z
I just use these really cheap smart plugs. They even have power monitoring!

krikk, 2024-08-19:
i also gave up on this sensor, it's crap in hardware so not really fixable...