Koenkk/zigbee2mqtt #17502: [New device support]: The device(_TZE204_sxm7l9xa) does not work.

kookss1, 2023-04-30:
### Link

https://ko.aliexpress.com/item/1005004724689720.html?spm=a2g0o.order_list.order_list_main.33.2762140fOyQAIR&gatewayAdapt=glo2kor

### Database entry

{"id":42,"type":"Router","ieeeAddr":"0xa4c138454eab7c15","nwkAddr":12575,"manufId":4417,"manufName":"_TZE204_sxm7l9xa","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���+i","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_sxm7l9xa","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1682818399569,"defaultSendRequestWhen":"immediate"}

### Comments

hello.

The Tuya presence sensor could not be connected to the HA, so I connected it using an external converter.

However, the sensor value cannot be used normally.
I doubt whether it is in a different format from "TS0601_smart_human_presence_sensor". In other communities, I saw a post about a problem with a device with the same manufacturer id (_TZE204_sxm7l9xa).

Please help me to connect the new presence sensor.
thank you

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ikvncluo'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_lyetpprm'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_jva8ink8'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_holel4dk'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_xpq2rzhq'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_wukb7rhc'},
        {modelID: 'TS0601', manufacturerName: '_TZE204_ztc6ggyl'},
        {modelID: 'TS0601', manufacturerName: '_TZE204_sxm7l9xa'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ztc6ggyl'}],
    model: 'TS0601_smart_human_presence_sensor',
    vendor: 'TuYa',
    description: 'Smart Human presence sensor',
    fromZigbee: [fz.tuya_smart_human_presense_sensor],
    toZigbee: [tz.tuya_smart_human_presense_sensor],
    exposes: [
        e.illuminance_lux(), e.presence(),
        exposes.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Minimum range').withUnit('m'),
        exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Maximum range').withUnit('m'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
        // exposes.text('cli', ea.STATE).withDescription('not recognize'),
        exposes.enum('self_test', ea.STATE, Object.values(tuya.tuyaHPSCheckingResult))
            .withDescription('Self_test, possible resuts: checking, check_success, check_failure, others, comm_fault, radar_fault.'),
    ],
};

module.exports = definition;
```

amasuke55, 2023-04-30:
That's interesting, unfortunately I have the same problem. I don't know if this information helps, but I bought the wall mount version from the exact same shop on aliexpress "haozee".

kookss1, 2023-05-01:
>

kookss1, 2023-05-01:
> That's interesting, unfortunately I have the same problem. I don't know if this information helps, but I bought the wall mount version from the exact same shop on aliexpress "haozee". I bought three of the same thing before, but from other shops on aliexpress and this is the fourth one and first that looks a little different at the USB port. It seems like they changed something about this device.

I also bought a wall-mounted presence sensor.
I checked in other communities and the Datapoint format seems to be a bit different. Here's what I've currently checked:

[104, 'illuminance_lux', tuya.valueConverter.raw],
[105, 'presence', tuya.valueConverter.trueFalse1],
[106, 'sensitivity', tuya.valueConverter.raw],
[107, 'far_detection', tuya.valueConverter.raw],
[108, 'fading_time', tuya.valueConverter.raw], 
[109, 'target_distance', tuya.valueConverter.raw],
[110, 'near_detection', tuya.valueConverter.raw],
[111, 'detection_delay', tuya.valueConverter.raw],

phoenixperson-c137, 2023-05-01:
Same here :(

Bought 2 from Yingfan store on Ali, it took almost a month to deliver but they worked fine, so I ordered 3 more, this time from haozee and it appears as unsupported.

Any chance it can be added?

cicciociccio333, 2023-05-02:
same problem for me 👎 I just received my presense sensor but zigbee2mqtt flood the log with the same messages and the sensor doesn't work.

hvoss-techfak, 2023-05-02:
Same problem here. I bought the sensor from the "wenzhi Factory Store"

Sch8er, 2023-05-02:
Same problem here. Shop: Quality House Life Store

`2023-05-02 20:32:51Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_sxm7l9xa'
2023-05-02 20:32:51Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html`

nicolacavallo84, 2023-05-02:
same problem for me too

phoenixperson-c137, 2023-05-05:
Tried again with zigbee2mqtt edge - it is now recognized as cover.
Tried fixing with an external converter but then I'm getting the same DP errors as before.

Is there any hope it can be fixed? or should I just give up and look for alternative sensors?

Koenkk, 2023-05-05:
@kookss1 could you post your external converter here so that others can test it?

phoenixperson-c137, 2023-05-05:
Sure

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ikvncluo'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_lyetpprm'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_jva8ink8'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_holel4dk'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_xpq2rzhq'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_wukb7rhc'},
        {modelID: 'TS0601', manufacturerName: '_TZE204_ztc6ggyl'},
        {modelID: 'TS0601', manufacturerName: '_TZE204_sxm7l9xa'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ztc6ggyl'}],
    model: 'TS0601_smart_human_presence_sensor',
    vendor: 'TuYa',
    description: 'Smart Human presence sensor',
    fromZigbee: [fz.tuya_smart_human_presense_sensor],
    toZigbee: [tz.tuya_smart_human_presense_sensor],
    exposes: [
        e.illuminance_lux(), e.presence(),
        exposes.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Minimum range').withUnit('m'),
        exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Maximum range').withUnit('m'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
        // exposes.text('cli', ea.STATE).withDescription('not recognize'),
        exposes.enum('self_test', ea.STATE, Object.values(tuya.tuyaHPSCheckingResult))
            .withDescription('Self_test, possible resuts: checking, check_success, check_failure, others, comm_fault, radar_fault.'),
    ],
    meta: {
        tuyaDatapoints: [
            [104, 'illuminance', tuya.valueConverter.raw],
            [105, 'presence', tuya.valueConverter.boolean],
            [106, 'radar_sensitivity', tuya.valueConverter.raw],
            [107, 'maximum_range', tuya.valueConverter.raw],
            [108, 'minimum_range', tuya.valueConverter.raw],
            [109, 'target_distance', tuya.valueConverter.raw],
            [110, 'detection_delay', tuya.valueConverter.raw],
            [111, 'fading_time', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

hvoss-techfak, 2023-05-05:
For some reason there are two issues for the same device. In the other Issue someone posted a working external converter.
https://github.com/Koenkk/zigbee2mqtt/issues/17483#issuecomment-1536638558

Martijndegeus, 2023-05-05:
With the converter @phoenixperson-c137 posted the device is recognized as human presence sensor, but I still get error messages about an unknown DP 109. With the one @hvoss-techfak referenced it does work (although it's partly commented out).

Sch8er, 2023-05-08:
#17483

closed 2023-05-08 by kookss1

Martijndegeus, 2023-05-08:
Using the external converter mentioned in #17483 seems to work.