Koenkk/zigbee2mqtt #17506: [New device support]: Tuya curtain motor LS82 (TZE200_xu4a5rhj)

dpbarinov, 2023-04-30:
### Link

Смотри, что есть на AliExpress! Умные шторы.Электрический занавес Умный дом Tuya Wi-Fi Zigbee Alexa Google SmartThings IFTTT за 7 142,40 ₽ - уже со скидкой -23% https://sl.aliexpress.ru/p?key=bhP0wmZ

### Database entry

—

### Comments

Device pairing but not supported

dpbarinov, 2023-04-30:
Works good with following converter

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_xu4a5rhj',
        },
    ],
    model: 'TS0601_LS82',
    vendor: 'TuYa',
    description: 'Tuya curtain motor',
    fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
    toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.composite('options', 'options', ea.STATE_SET)
                .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                    .withValueMin(0).withValueMax(255).withDescription('Motor speed'))
                .withFeature(exposes.binary('reverse_direction', ea.STATE_SET, true, false)
                    .withDescription('Reverse the motor direction'))],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
        ],
    },
};

module.exports = definition;

Koenkk, 2023-04-30:
Added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-04-30 by Koenkk