Koenkk/zigbee2mqtt #17512: Cannot change settings

MarcPot, 2023-04-30:
### What happened?

When trying to change an mqtt setting z2m errors out and doesn't change the setting.

```Debug 2023-04-30 21:06:57Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"mqtt":{"base_topic":"zigbee2mqtt","client_id":"''","force_disable_retain":false,"include_device_information":false,"keepalive":60,"reject_unauthorized":true,"server":"mqtt://10.22.10.103","user":"''","version":5}},"transaction":"l8fpi-4"}'
Info 2023-04-30 21:06:57Succesfully changed options
Info 2023-04-30 21:06:57MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":true},"status":"ok","transaction":"l8fpi-4"}'
Debug 2023-04-30 21:07:02Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"mqtt":{"base_topic":"zigbee2mqtt","client_id":"''","force_disable_retain":false,"include_device_information":false,"keepalive":60,"reject_unauthorized":true,"server":"mqtt://10.22.10.169","user":"''","version":5}},"transaction":"l8fpi-5"}'
Error 2023-04-30 21:07:02Request 'zigbee2mqtt/bridge/request/options' failed with error: 'Cannot read properties of null (reading 'length')'
Debug 2023-04-30 21:07:02TypeError: Cannot read properties of null (reading 'length') at Object.apply (/usr/lib/zigbee2mqtt/lib/util/settings.ts:472:16) at Bridge.bridgeOptions (/usr/lib/zigbee2mqtt/lib/extension/bridge.ts:180:42) at Bridge.onMQTTMessage (/usr/lib/zigbee2mqtt/lib/extension/bridge.ts:123:63) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/usr/lib/zigbee2mqtt/lib/eventBus.ts:109:22) at MQTT.onMessage (/usr/lib/zigbee2mqtt/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/usr/lib/zigbee2mqtt/lib/extension/frontend.ts:94:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/usr/lib/zigbee2mqtt/node_modules/ws/lib/websocket.js:1178:20) at Receiver.emit (node:events:513:28)
Info 2023-04-30 21:07:02MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{},"error":"Cannot read properties of null (reading 'length')","status":"error","transaction":"l8fpi-5"}' ```

### Zigbee2MQTT version

1.28.2-dev

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11362129/log.txt)

Calimerorulez, 2023-05-07:
Same problem here :(

LutraVulgaris, 2023-05-20:
I got it too

closed 2023-06-27 by github-actions[bot]