Koenkk/zigbee2mqtt #17514: [New device support]: Philips Hue portable table lamp

robinbohnen, 2023-05-01:
### Link

https://www.philips-hue.com/nl-nl/p/hue-white-and-color-ambiance-hue-go-draagbare-tafellamp/8719514404595

### Database entry

{"id":124,"type":"Router","ieeeAddr":"0x001788010cb452e6","nwkAddr":8341,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"929003128501","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"929003128501","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210305","swBuildId":"1.82.10"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210305","swBuildId":"1.82.10","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1682917013737,"defaultSendRequestWhen":"immediate"}

### Comments

Please support this new device, thanks! 

### Supported color modes

color temperature, color (rgb)

### Color temperature range

Doesnt work, not sure how to retrieve this in another way

Koenkk, 2023-05-01:
Added! Groeten uit Brabant 😄 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-01 by Koenkk

robinbohnen, 2023-05-02:
Thanks! Groeten uit Brabant terug!